micropython/micropython #8647: Module uasyncio is lack of support for SSL/TLS support.

RuofengX, 2022-05-10:
## TL;DR 

How to open a SSLor TLS connection using uasyncio?

------

I want to port a CPython project onto Micropython.   
When I am trying to use `uasyncio.open_connection(..., ssl=True)`, it raises an exception saying that there's no ssl argument. I confirm it on the doc page [here](https://docs.micropython.org/en/latest/library/uasyncio.html#uasyncio.open_connection).

Then I try to use ssl module to warp the socket in `StreamWriter.get_extra_info("socket")`, but I cannot get the "raw" socket of this connection, because function `uasyncio.Stream.get_extra_info` only accept "peername" as a param, no support for "socket". We can check it [here](https://docs.micropython.org/en/latest/library/uasyncio.html#uasyncio.Stream.get_extra_info).

If there's a method to open an SSL/TLS connection by using uasyncio? **I really want to use asyncio because I considered that is the best part of python language.**

RuofengX, 2022-05-10:
Here is the code https://github.com/RuofengX/uPyxy/blob/c3772838be208056282e01ed0d3f8da0c8fa4de8/main/lib/client.py#L103

peterhinch, 2022-05-10:
You might like to look at [mqtt_as](https://github.com/peterhinch/micropython-mqtt) which uses nonblocking sockets and works over TLS.

RuofengX, 2022-05-10:
> I implemented a workaround for my project [here](https://github.com/hyx0329/micropython-uicefox/blob/b9af3acfe80587b765d39eb9803605f33561138b/uicefox.py#L7). It's basically a custom connection maker based on the original one.

That's awesome! I will check it right now😍

stephanelsmith, 2022-05-18:
I ran into this last week and also implemented my own asyncio friendly implementation.  My version is optimized for my application, but asyncio ssl open_connection is surely something that would be apprecaited by future travelers.

I will also agree asyncio is my favorite part of python language!  Game changer for micro-controller dev.

dpgeorge, 2023-12-14:
Asyncio SSL server and client are now supported.

PR #11897 (2023-12-14): extmod/asyncio: Add ssl support with SSLContext.
This adds `asyncio` ssl support ~~and fix async stream read methods in unix port.
The reason is stream ssl sockets seem to behave different from plain stream sockets. The main difference
is doing partial reads on ssl sockets will clear any write/read/close "event" flag even if there is still
pending data to be read. So registering the ssl socket again in the poller will wait there until a new event
or timeout.
The fix is check first if the ssl socket has pending data and read/return it instead of registering in the poller to wait
for an event.~~ [... cut]
closed 2023-12-14 by dpgeorge