micropython/micropython #8648: Differences between class definitions and instances of classes (objects)

arizvisa, 2022-05-10:
Apologies if this question is answered in the documentation as I think it might be related to the already noted constraints on the method-resolution order, but I wasn't clear about it..or whom to ask since I'm not on Slack and the forum seems to be related to mostly porting issues.

Essentially I'm trying to differentiate a method/callable that's been assigned to a class from a method/callable that's been assigned to an instance. CPython splits this up by providing an instancemethod which requires one to access the `im_func` or nowadays the `__func__` attribute in order to compare them. This is irrelevant in Micropython, however, so this is fine.

However, one thing I noticed was that Micropython doesn't seem to differentiate the layouts of the definition of a class from each of the classes that inherit from it..and any modification of that layout will also affect an object __after__ it has already been instantiated. It seems as if the reference to the namespace of a class is actually "flattened" between all the classes that inherit from it. I'll try to explain with some examples.

```
class base(object):
    pass
class X(base):
    pass
class Y(base):
    pass

x, y = X(), Y()

assert(not hasattr(X, 'nothere') and not(hasattr(Y, 'nothere')))
assert(not hasattr(x, 'nothere') and not(hasattr(y, 'nothere')))

base.nothere = "i am here"

# now "X", and "Y" and their corresponding instances "x", and "y" now have this attribute.
assert(not hasattr(X, 'nothere') and not(hasattr(Y, 'nothere')))
assert(not hasattr(x, 'nothere') and not(hasattr(y, 'nothere')))
```

This is another example of the above, but it shows a value being passed between two already created instances without it being explict. The instance of the parent class simply adds an attribute to its class, which any child classes will immediately inherit.
```
class parent(object):
    @classmethod
    def flash(self):
        self.hand = 'a cookie'

class child(parent):
    def flash(self):
        return self.hand if hasattr(self, 'hand') else 'nothing' 

person1, person2 = parent(), child()
print(person2.flash(), 'in the hand')
person1.flash()
print('poof,', person2.flash())
```

Is this intended behaviour due to the constraints on the method resolution order as laid out in the documentation? Or is it just a side effect of the implementation? If it's the latter, I imagine that to fix it there'll be an unfortunate trade that would need to be made: If the namespace isn't just copied from the base class everytime a subclass is instantiated (or derived), then every instance of said subclass would need its own reference to the class' namespace so that when a base class is modified, it doesn't affect everything all of its references...something like that, anyways...

dpgeorge, 2022-05-27:
Testing the above to examples on both CPython and MicroPython, CPython and MicroPython behaviour agree for both examples.

So, with those examples at least, MicroPython has the correct behaviour.

arizvisa, 2022-05-27:
Ah, my mistake. You're 100% right. I was trying to avoid just throwing my random own unit-tests at you guys and screwed that up pretty well while trying to explain what I think is going on. Sorry about that. :)

Give me a moment while I test this on Micropython (since it seems to work on CPython and PyPy) and try and isolate the exact symptom that I seem to be encountering.

arizvisa, 2022-05-27:
Okay, to be clear. I did intend this as being a question which was why I was initially looking for places where I could ask it, but I went into "bugtracker-mode" on GH's issue tracker and went on a tangent trying to clarify this as a symptom which is pretty lol. Anyways, I revisited my assumptions and figured out what I was trying to ask when I created this.

Essentially my intent is to compare the equality of the callable that is promoted into a method which makes this a question about introspection and thus it likely appears as more of a feature-need rather than a bug/issue in the tracker.

The following is an example of the same callable being applied to two-related classes. However, micropython does not have a way to access the callable's reference from a method that it is based on. (without explicitly capturing it during the construction of the class).
```
same_callable = lambda: 1

class ParentClass(object):
    method = same_callable

class ChildClass(ParentClass):
    method_different_name = same_callable
    
Parent, Child = ParentClass(), ChildClass()

if hasattr(Parent.method, '__func__'):
    print('cpy|pypy', Parent.method.__func__ is Child.method_different_name.__func__)

# insert bogus comparisons here
```

Another example which is the same callable being applied to two different classes.
```
same_callable = lambda: 1

class Class1(object):
    method_1 = same_callable

class Class2(object):
    method_2 = same_callable

Instance1 = Class1()
Instance2 = Class2()

if hasattr(Instance1.method_1, '__func__'):
    print('cpy|pypy', Instance1.method_1.__func__ is Instance2.method_2.__func__)

# insert bogus comparisons here
```

During my flailing, I had made the assumption that if I access the __dict__ I could then get at the callable that is converted to a method and then use that to compare. However, this was actually using the logic that I created the issue with and hence it was not actually doing what I had intended it to do.

As described, one solution to this seems to be to capture the callable during the definition of the class, however, if a user assigns a callable to an instance of the class (implying that a method and a function are within the same namespace)..there's no way to compare their similarity due to methods not having a reference to the callable that composes them.

The following example differentiates between a function and a method in order to determine the equality of the callable that composes them.
```
same_callable = lambda: 1
class Class(object):
    method1 = same_callable

Instance = Class()
Instance.method2 = same_callable

m1, m2 = Instance.method1, Instance.method2
if hasattr(m1, '__func__'):
    print('cpy|pypy', m1.__func__ is (m2 if isinstance(m2, type(same_callable)) else m2.__func__))
```

Sorry for mis-stating this as a bug with a symptom when it appears as more of a request. I get that this seems to be a very niche need as I believe most people who access these attributes are doing very particular things other than determining equality.

dpgeorge, 2022-05-31:
It sounds like you just need `__func__` implemented on bound-method instances.

This patch will do that (and also add `__self__`):
```diff
--- a/py/objboundmeth.c
+++ b/py/objboundmeth.c
@@ -85,12 +85,20 @@ STATIC mp_obj_t bound_meth_call(mp_obj_t self_in, size_t n_args, size_t n_kw, co
 
 #if MICROPY_PY_FUNCTION_ATTRS
 STATIC void bound_meth_attr(mp_obj_t self_in, qstr attr, mp_obj_t *dest) {
+    mp_obj_bound_meth_t *self = MP_OBJ_TO_PTR(self_in);
     if (dest[0] != MP_OBJ_NULL) {
         // not load attribute
         return;
     }
+    if (attr == MP_QSTR___func__) {
+        dest[0] = self->meth;
+        return;
+    }
+    if (attr == MP_QSTR___self__) {
+        dest[0] = self->self;
+        return;
+    }
     // Delegate the load to the method object
-    mp_obj_bound_meth_t *self = MP_OBJ_TO_PTR(self_in);
     mp_load_method_maybe(self->meth, attr, dest);
 }
 #endif
```

Then the above code snippets will work the same in MicroPython as CPython.

BTW, if you do `repr(Parent.method)` in MicroPython then it'll have enough info in it to distinguish the `__func__` value, eg:
```
<bound_method 7f3626061860 <ParentClass object at 7f3626061300>.<function <lambda> at 0x7f3626061280>>
<bound_method 7f3626061880 <ChildClass object at 7f3626061320>.<function <lambda> at 0x7f3626061280>>
```

arizvisa, 2022-05-31:
That is flipping perfect. Thank you so much for looking into it...it sounds exactly like what I need to check the symmetry between the callable types. Checking `repr` is also a great idea, I definitely never considered parsing its output to compare it to the `id` of something. If it ends up being decided that these other attributes are not added to Micropython then this will still allow me resolve my issue.

I'll apply this patch, and switch over to its additions.. instead of trying to extract it via the namespace and let you know so we can close this issue.

I'm sure you get this all the time, but your work on this project is super awesome. :-)