micropython/micropython #8655: External C modules on ESP8266

w4rum, 2022-05-11:
The [documentation](https://docs.micropython.org/en/latest/develop/cmodules.html) describes how one can include external C modules when building Micropython. However, I can't get this to work with the ESP8266 port. The `USER_C_MODULES` flag doesn't appear to have any effect. In fact, the produced firmware blob has the same md5sum either way:

```fish
# (note that this is fish syntax, so I'm using (id -u) instead of $UID)

# without USER_C_MODULES
> docker run --rm -v $HOME:$HOME -u (id -u) -w $PWD larsks/esp-open-sdk make clean
[...]
> docker run --rm -v $HOME:$HOME -u (id -u) -w $PWD larsks/esp-open-sdk make
[...]
> md5sum build-GENERIC/firmware-combined.bin 
a34e0952c8da9a27949b9261b9129a1e  build-GENERIC/firmware-combined.bin

# with USER_C_MODULES
> docker run --rm -v $HOME:$HOME -u (id -u) -w $PWD larsks/esp-open-sdk make clean
[...]
> docker run --rm -v $HOME:$HOME -u (id -u) -w $PWD larsks/esp-open-sdk make USER_C_MODULES=../../../st7789_mpy/st7789/micropython.cmake
[...]
> md5sum build-GENERIC/firmware-combined.bin
a34e0952c8da9a27949b9261b9129a1e  build-GENERIC/firmware-combined.bin
```

When looking over the [Makefile](https://github.com/micropython/micropython/blob/master/ports/esp8266/Makefile) in `ports/esp8266`, I noticed that neither `USER_C_MODULES` nor `cmake` is referenced at all. I also couldn't find anything else on how to include external modules in an ESP8266 build.

This prevents me from using drivers written in C, e.g., [devbis/st7789_mpy](https://github.com/devbis/st7789_mpy), with my ESP8266, even though the README on the driver itself claims to have once worked with the ESP8266. So I assume this is an issue with the build system, not the driver itself.

Is there any way to include external C modules with the ESP8266 build? If there is, could that be added to the wiki and/or the [ESP8266 README](https://github.com/micropython/micropython/blob/master/ports/esp8266/README.md)?

jimmo, 2022-05-11:
> `docker run --rm -v $HOME:$HOME -u (id -u) -w $PWD larsks/esp-open-sdk make USER_C_MODULES=../../../st7789_mpy/st7789/micropython.cmake`

The esp8266 port doesn't use cmake, so wouldn't expect to see a .cmake file here. I suspect it's just ignoring it.

The instructions at that repo say to use
`make USER_C_MODULES=../../../st7789_mpy/`
which sounds correct for a user c module.

jimmo, 2022-05-11:
> Yeah, that's what I think as well. But is there any other sensible way to include external modules?

Does it work when you do 

```
docker prefix ... make USER_C_MODULES=../../../st7789_mpy/
```

w4rum, 2022-05-13:
Nope, that still produces the same md5sum as without the flag.

dpgeorge, 2022-05-19:
If I run the following from the `ports/esp8266` directory then it works, it includes the example user C module from this repo:
```
$ make USER_C_MODULES=../../examples/usercmodule
```
(You have to remove the C++ example because esp8266 won't link with C++.)

I suspect you need to use one directory lower, eg:
```
$ docker prefix ... make USER_C_MODULES=../../../
```

w4rum, 2022-05-22:
@dpgeorge Thank you, that pointed me in the right direction!

I still don't quite understand where in that build chain `USER_C_MODULES` is read because a string-search on the `ports/esp8266` returns nothing:
```sh
~/Downloads/micropython/ports/esp8266
> rg USER_C_MODULES
[...no output...]
```
But anyway, I figured out my problem. And it appears that what I'm doing actually *is* documented correctly in the micropython docs. I was just overreading it:
> Set the build-time flag USER_C_MODULES to point to the modules you want to include. **For ports that use Make this variable should be a directory which is searched automatically for modules.** For ports that use CMake this variable should be a file which includes the modules to build.

Note that it says "a directory which is searched for modules", not "a module directory".

Linking to the module directory or linking the cmake file directly, as I've done in my OP, doesn't seem to work. Neither
* `USER_C_MODULES=../../examples/usercmodule/cexample/` nor
* `USER_C_MODULES=../../../st7789_mpy/st7789/`
work.

It only works when I linked to a directory one step higher such as `../../examples/usercmodule/` as @dpgeorge suggested or `../../../st7789_mpy/` as @jimmo suggested.

Having a working example with `../../examples/usercmodules` allowed me to find out that my working copy of `st7789_mpy` was broken. Hard resetting it restored the module directory `st7789_mpy/st7789` to a good state and caused it be included in the build process. So this is entirely my fuckup, sorry for the inconvenience.

I suppose it would be nice if the build process had a verbose setting that hinted at possible problems with module directories, such as logging which directories are checked for modules and whether they are ignored or not. But I guess proper RTFM would have fixed this as well.

*(The build process still doesn't work because the module doesn't build but that's not an issue for this repo I guess.)*

closed 2022-05-22 by w4rum

jimmo, 2022-05-23:
> I still don't quite understand where in that build chain USER_C_MODULES is read because a string-search on the ports/esp8266 returns nothing:

@w4rum Because this is common to all ports, it's handled in `py.mk` (see https://github.com/micropython/micropython/blob/master/py/py.mk#L32) which is included from `esp8266/Makefile`.

> I suppose it would be nice if the build process had a verbose setting that hinted at possible problems with module directories

I think you're right, would be good to address this (I have fallen for this before, and apologies for not being more clear in my initial message about why that path "looks correct"). The limitation here is that we only get to set USER_C_MODULES once, so it has to be a single directory that contains all the modules you want. (We could definitely do something cleverer here though with a path separator etc)

dpgeorge, 2022-05-23:
> The limitation here is that we only get to set USER_C_MODULES once

IMO this variable should point to a `.mk` file that is included in the build process and then can include subsequent modules.  Then you can construct a specific `.mk` file for your board/project that includes various C modules from arbitrary locations.  This is how it works for cmake builds.

w4rum, 2022-05-23:
@jimmo 
> Because this is common to all ports, it's handled in py.mk

Ah, I see. Thank you!

abuvanth, 2023-01-05:
fixed my issue