micropython/micropython #8656: Build from source failed for rp2

mattmeye, 2022-05-11:
Hello,

i can't build the project from source for rp2 and ARDUINO_NANO_RP2040_CONNECT.

steps to reproduce:
```
git clone https://github.com/micropython/micropython.git
git clone https://github.com/micropython/micropython-lib.git
cd micropython
make -C mpy-cross
make -C ports/rp2 BOARD=ARDUINO_NANO_RP2040_CONNECT
```

build error:
```
[ 95%] Linking CXX executable firmware.elf
   text    data     bss     dec     hex filename
 408968     156  220184  629308   99a3c /mnt/x/tools/rp2040/micropython/ports/rp2/build-ARDUINO_NANO_RP2040_CONNECT/firmware.elf
make[3]: Leaving directory '/mnt/x/tools/rp2040/micropython/ports/rp2/build-ARDUINO_NANO_RP2040_CONNECT'
[ 96%] Built target firmware
make[3]: Entering directory '/mnt/x/tools/rp2040/micropython/ports/rp2/build-ARDUINO_NANO_RP2040_CONNECT'
Scanning dependencies of target micropy_extmod_btree
make[3]: Leaving directory '/mnt/x/tools/rp2040/micropython/ports/rp2/build-ARDUINO_NANO_RP2040_CONNECT'
make[3]: Entering directory '/mnt/x/tools/rp2040/micropython/ports/rp2/build-ARDUINO_NANO_RP2040_CONNECT'
[ 97%] Building C object CMakeFiles/micropy_extmod_btree.dir/mnt/x/tools/rp2040/micropython/lib/berkeley-db-1.xx/btree/bt_close.c.obj
In file included from /mnt/x/tools/rp2040/micropython/lib/berkeley-db-1.xx/btree/bt_close.c:48:
/mnt/x/tools/rp2040/micropython/lib/berkeley-db-1.xx/PORT/include/db.h:1:1: error: expected identifier or '(' before '.' token
    1 | ../../include/db.h
      | ^
In file included from /mnt/x/tools/rp2040/micropython/lib/berkeley-db-1.xx/btree/bt_close.c:49:
/mnt/x/tools/rp2040/micropython/lib/berkeley-db-1.xx/btree/btree.h:89:3: warning: data definition has no type or storage class
   89 | } PAGE;
      |   ^~~~
/mnt/x/tools/rp2040/micropython/lib/berkeley-db-1.xx/btree/btree.h:89:3: warning: type defaults to 'int' in declaration of 'PAGE' [-Wimplicit-int]
/mnt/x/tools/rp2040/micropython/lib/berkeley-db-1.xx/btree/btree.h:127:2: error: unknown type name 'pgno_t'
  127 |  pgno_t pgno;   /* page number stored on */
      |  ^~~~~~
/mnt/x/tools/rp2040/micropython/lib/berkeley-db-1.xx/btree/btree.h:157:2: error: unknown type name 'recno_t'
  157 |  recno_t nrecs;   /* number of records */
      |  ^~~~~~~
# removed lines
      |              ^~~~~~~~~~~~
/mnt/x/tools/rp2040/micropython/lib/berkeley-db-1.xx/btree/bt_close.c:171:13: error: 'BTREE' {aka 'struct _btree'} has no member named 'bt_psize'
  171 |  m.psize = t->bt_psize;
      |             ^~
/mnt/x/tools/rp2040/micropython/lib/berkeley-db-1.xx/btree/bt_close.c:172:12: error: 'BTREE' {aka 'struct _btree'} has no member named 'bt_free'
  172 |  m.free = t->bt_free;
      |            ^~
/mnt/x/tools/rp2040/micropython/lib/berkeley-db-1.xx/btree/bt_close.c:173:13: error: 'BTREE' {aka 'struct _btree'} has no member named 'bt_nrecs'
  173 |  m.nrecs = t->bt_nrecs;
      |             ^~
In file included from /mnt/x/tools/rp2040/micropython/lib/berkeley-db-1.xx/btree/bt_close.c:49:
/mnt/x/tools/rp2040/micropython/lib/berkeley-db-1.xx/btree/btree.h:42:27: error: 'BTREE' {aka 'struct _btree'} has no member named 'flags'
   42 | #define F_ISSET(p, f) ((p)->flags & (f))
      |                           ^~
/mnt/x/tools/rp2040/micropython/lib/berkeley-db-1.xx/btree/bt_close.c:174:12: note: in expansion of macro 'F_ISSET'
  174 |  m.flags = F_ISSET(t, SAVEMETA);
      |            ^~~~~~~
/mnt/x/tools/rp2040/micropython/lib/berkeley-db-1.xx/btree/bt_close.c:177:25: error: 'MPOOL_DIRTY' undeclared (first use in this function)
  177 |  mpool_put(t->bt_mp, p, MPOOL_DIRTY);
      |                         ^~~~~~~~~~~
/mnt/x/tools/rp2040/micropython/lib/berkeley-db-1.xx/btree/bt_close.c:178:10: error: 'RET_SUCCESS' undeclared (first use in this function); did you mean 'EXIT_SUCCESS'?
  178 |  return (RET_SUCCESS);
      |          ^~~~~~~~~~~
      |          EXIT_SUCCESS
make[3]: *** [CMakeFiles/micropy_extmod_btree.dir/build.make:63: CMakeFiles/micropy_extmod_btree.dir/mnt/x/tools/rp2040/micropython/lib/berkeley-db-1.xx/btree/bt_close.c.obj] Error 1
make[3]: Leaving directory '/mnt/x/tools/rp2040/micropython/ports/rp2/build-ARDUINO_NANO_RP2040_CONNECT'
make[2]: *** [CMakeFiles/Makefile2:1588: CMakeFiles/micropy_extmod_btree.dir/all] Error 2
make[2]: Leaving directory '/mnt/x/tools/rp2040/micropython/ports/rp2/build-ARDUINO_NANO_RP2040_CONNECT'
make[1]: *** [Makefile:84: all] Error 2
make[1]: Leaving directory '/mnt/x/tools/rp2040/micropython/ports/rp2/build-ARDUINO_NANO_RP2040_CONNECT'
make: *** [Makefile:23: all] Error 2
make: Leaving directory '/mnt/x/tools/rp2040/micropython/ports/rp2'
```

Am I calling the build incorrectly?

jimmo, 2022-05-11:
Try doing a `make -C ports/rp2 submodules` (See https://github.com/micropython/micropython/tree/master/ports/rp2#building)

Is there different instructions that we need to update somewhere?

czietz, 2022-05-11:
What's the file system on `/mnt/x/...`? Does it support symbolic links? The build process complains about a syntax error in `/mnt/x/tools/rp2040/micropython/lib/berkeley-db-1.xx/PORT/include/db.h` which is actually a symbolic link.

mattmeye, 2022-05-11:
it is an ubuntu 20.x in windows (wsl).

czietz, 2022-05-11:
Well ... and what's the file system on `/mnt/x/...`? Does it support symbolic links? What does...
```
ls -l /mnt/x/tools/rp2040/micropython/lib/berkeley-db-1.xx/PORT/include/
```
show?

mattmeye, 2022-05-11:
-rwxrwxrwx 1 matt matt    7 May 11 11:06 bsd-queue.h
-rwxrwxrwx 1 matt matt 4694 May 11 11:06 cdefs.h
-rwxrwxrwx 1 matt matt 7304 May 11 11:06 compat.h
-rwxrwxrwx 1 matt matt   18 May 11 11:06 db.h
-rwxrwxrwx 1 matt matt   26 May 11 11:06 filevtable.h
-rwxrwxrwx 1 matt matt   21 May 11 11:06 mpool.h
-rwxrwxrwx 1 matt matt 2819 May 11 11:06 ndbm.h
-rwxrwxrwx 1 matt matt 8612 May 11 11:06 queue.h

czietz, 2022-05-11:
Okay, so the links are broken / missing. It should show up as...
```
lrwxrwxrwx 1 pi pi    7 May 11 21:15 bsd-queue.h -> queue.h
-rw-r--r-- 1 pi pi 4694 May 11 21:15 cdefs.h
-rw-r--r-- 1 pi pi 7304 May 11 21:15 compat.h
lrwxrwxrwx 1 pi pi   18 May 11 21:15 db.h -> ../../include/db.h
lrwxrwxrwx 1 pi pi   26 May 11 21:15 filevtable.h -> ../../include/filevtable.h
lrwxrwxrwx 1 pi pi   21 May 11 21:15 mpool.h -> ../../include/mpool.h
-rw-r--r-- 1 pi pi 2819 May 11 21:15 ndbm.h
-rw-r--r-- 1 pi pi 8612 May 11 21:15 queue.h
```

So, it could be that your X: drive (that shows up as /mnt/x) does not support symbolic links. (I asked you this question already.)

Another possibility: Did you perhaps clone the repository with a _Windows_ version of git? I'm not really sure how that handles links, in particular if combined with WSL. Try installing git under Ubuntu (running in WSL) and clone the repo again; not to `/mnt/x/` but somewhere in your Ubuntu home directory.

mattmeye, 2022-05-11:
Hi @czietz ,
i think you're absolutely right, the /mnt/x does not support symlinks. Normally this is no problem in wsl. In the home profile folder the build is very fast and without errors. Thank you a lot for your help!

closed 2022-05-11 by mattmeye