micropython/micropython #8660: SOS! question.

QiaoTuCodes, 2022-05-13:
hey guys . 

we know when pyboard start , the program always from boot.py & main.py first .  
so how can i change main.py name or boot.py name and default load path to auto runing?  before i want packaging main.py& boot.py codes inside to firmware.  i'll need main.py or boot.py always runing form firmware. not using sd card or flash space. 

thank you !

jimmo, 2022-05-13:
This question might be better asked on the forum -- https://forum.micropython.org/

On STM32 ports (e.g. pyboard) from boot.py you can configure what file to use as the "main" (otherwise it defaults to main.py like you say).
https://docs.micropython.org/en/latest/library/pyb.html#pyb.main

I'm not quite sure what you're asking though... is this something to do with freezing boot.py & main.py ?

QiaoTuCodes, 2022-05-13:
> This question might be better asked on the forum -- https://forum.micropython.org/
>
> On STM32 ports (e.g. pyboard) from boot.py you can configure what file to use as the "main" (otherwise it defaults [...]

yes! you right. i mean how can i freezing boot.py & main.py codes.

jimmo, 2022-05-13:
> yes! you right. i mean how can i freezing boot.py & main.py codes.

You can freeze them the same way as freezing any other Python code. The easiest way is to write a manifest script. https://docs.micropython.org/en/latest/reference/manifest.html

QiaoTuCodes, 2022-05-13:
> > yes! you right. i mean how can i freezing boot.py & main.py codes.
> 
> You can freeze them the same way as freezing any other Python code. The easiest way is to write a manifest script. https://docs.micropython.org/en/latest/reference/manifest.html

i got it. but if freezd . how to automatic start this codes script? like : 

boot.py :
=============
imprt os
os.listdir('/')
=============

now freezd the boot.py file in firmware.hex . 
so how can i just push on power. codes will be automatic runing.?

QiaoTuCodes, 2022-05-13:
i mean : need pyboard automatic runing something freezd pyb files.

QiaoTuCodes, 2022-05-13:
how can i do that?

QiaoTuCodes, 2022-05-13:
main.py & boot.py always can auto created in power on. i just need always can't auto created the 2 files in power on. i want change the 2files name & local path. it's all.

jimmo, 2022-05-13:
I'm sorry I don't quite understand...

> main.py & boot.py always can auto created in power on.

That's not supposed to happen. main.py and boot.py will only be created automatically if there is no filesystem, but once the filesystem is created, if you delete them they will stay deleted.

> i mean : need pyboard automatic runing something freezd pyb files.

boot.py (and main.py) will be run automatically, and the will be found using sys.path. So if they're frozen, and ".frozen" is in sys.path after "" (which is the default), then the ones on the filesystem will be used preferentially if they exist. If you want to force it to use the frozen ones, then either change sys.path or remove them from the filesystem.

QiaoTuCodes, 2022-05-13:
> 对不起，我不太明白...
> 
> > main.py 和 boot.py 总是可以在开机时自动创建。
> 
> 这不应该发生。main.py 和 boot.py 只会在没有文件系统的情况下自动创建，但是一旦创建了文件系统，如果你删除它们，它们将保持被删除状态。
> 
> > 我的意思是：需要 pyboard 自动运行一些 freezd pyb 文件。
> 
> boot.py（和 main.py）将自动运行，并且可以使用 sys.path 找到。因此，如果它们被冻结，并且“.frozen”在“”之后的 sys.path 中（这是默认设置），那么文件系统上的那些将被优先使用（如果它们存在）。如果你想强制它使用冻结的，那么要么改变 sys.path 要么从文件系统中删除它们。

> I'm sorry I don't quite understand...
>
>
> That's not supposed to happen. main.py and boot.py will only be created automatically if there is no filesystem, but once the filesystem is created, if you [...]

great! so how can i change sys.path ?  if i'll need change run automatic files as 'self_run.py ' and i freezd self_run.py in frozen folder . the files will be make inside firmwave.hex.

jimmo, 2022-05-13:
> so how can i change sys.path ?

The best option is to set MICROPY_BOARD_FROZEN_BOOT_FILE in your board defintion's mpconfigboard.h
See https://github.com/micropython/micropython/blob/master/ports/stm32/main.c#L573 for how this is used. It's run before boot.py (and main.py).

Set this to a frozen file (e.g. preboot.py) that modifies sys.path.

Then boot.py (and subsequently main.py) will be loaded.

jimmo, 2022-05-13:
The other option, if you don't want to make a mpconfigboard.h is just to have boot.py live on the filesystem, and then it can do "import frozenboot" to get your frozen boot code. Then you don't have to worry about sys.path at all.

If your problem is the auto-creation of boot.py and main.py though, you can just disable this (or modify it to generate anything you like in boot.py and main.py) see https://github.com/micropython/micropython/blob/master/ports/stm32/factoryreset.c

Or, even better, make a firmware image that includes the filesystem too.

QiaoTuCodes, 2022-05-13:
okey ! you answer's my need infomations. thanks you so much!

closed 2022-05-13 by jimmo