micropython/micropython #8661: righ IDF /commit hash to avoid compilation issues for ESP32

kindmartin, 2022-05-13:
Im trying to compile for a ESP32 port this 1,9.4 release and doing so getting some IDF error. ref as https://github.com/micropython/micropython/issues/8607.

What would be the right IDF version to use to compile this ESP32 port? how do I get the right hash?

Im cloning 4.4.1 but getting errors.

git clone -b v4.4.1 --recursive https://github.com/espressif/esp-idf.git

martin@ubuntu:~/esp/micropython/ports/esp32$ make submodules
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
** WARNING **
The git hash of ESP IDF does not match the supported version
The build may complete and the firmware may work but it is not guaranteed
ESP IDF path: /home/martin/esp/esp-idf
Current git hash: 1329b19fe494500aeb79d19b27cfd99b40c37aec
Supported git hash: 3ede9f011b50999b0560683f9419538c066dd09e
make: *** No rule to make target 'submodules'. Stop.

thanks in advance

jimmo, 2022-05-13:
> port this 1,9.4 release

What does this mean? Are you trying to compile MicroPython v1.9.4 (that's four years old!).

> What would be the right IDF version to use to compile this ESP32 port? how do I get the right hash?

The error message tells you the supported git hash (see the "supported git hash" line).

FWIW, if you want to compile the latest micropython, the automatic builds use the v4.0.2 and v4.4 tags.

Note the readme says "Currently MicroPython supports v4.0.2, v4.1.1, v4.2.2, v4.3.2 and v4.4, although other IDF v4 versions may also work."  I don't know if 4.4.1 specifically is supported (but seems likely).

kindmartin, 2022-05-13:
thanks @jimmo . I was meaning 1.18 !( Lastest at https://github.com/micropython/micropython/releases)
so the question is how do I specifically use v4.3.2 for instance?

martin@ubuntu:~/esp$ git clone -b v4.3.2--recursive https://github.com/espressif/esp-idf.git
cd ~/esp/esp-idf
./install.sh all
do I need to issue this?
git checkout "hash" where hah is some defined value ?  

then I do
$HOME/esp/esp-idf/export.sh

git clone --recursive https://github.com/micropython/micropython.git
cd micropython
make -C mpy-cross
cd ports/esp32
make submodules

make j=16 BOARD=GENERIC_SPIRAM

Generated /home/martin/esp/micropython/ports/esp32/build-GENERIC_SPIRAM/micropython.bin

Project build complete. To flash, run this command:
/home/martin/.espressif/python_env/idf4.3_py3.9_env/bin/python ../../../esp-idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build-GENERIC_SPIRAM/bootloader/bootloader.bin 0x8000 build-GENERIC_SPIRAM/partition_table/partition-table.bin 0x10000 build-GENERIC_SPIRAM/micropython.bin
or run 'idf.py -p (PORT) flash'
bootloader  @0x001000    22512  (    6160 remaining)
partitions  @0x008000     3072  (    1024 remaining)
application @0x010000  1417552  (  614064 remaining)
total                  1483088

All worked. ! thanks again.

;M

closed 2022-05-13 by kindmartin