micropython/micropython #8664: WebREPL https://micropython.org/webrepl/ do not works - possibly a bug?

beyonlo, 2022-05-13:
Hi all.

In the docs https://docs.micropython.org/en/latest/esp8266/tutorial/repl.html?highlight=webrepl say this:

"For your convenience, WebREPL client is hosted at http://micropython.org/webrepl " 

Unfortunately using that URL do not works, but if I copy the webrepl.html from https://github.com/micropython/webrepl to my desktop and open it locally, the WebREPL works fine. I tried to put those files (from github) in my webserver (HTTP Server), but not works too. 

Please, could you confirm if that is a BUG? I would like so much to access WebREPL over a WebServer, not locally (using the webrepl.html).

I'm doing tests with WebREPL to learning it, and to use as WebSocket Server for my html pages that will use the webrepl.html (websocket client) as a start point for my WebSocket web apps.

Any idea how so solve that?

Thank you in advance.

jimmo, 2022-05-14:
@beyonlo

Yes, there's a red banner at the top of https://micropython.org/webrepl that explains this. See https://github.com/micropython/micropython/issues/5266/ for more info.

At the time that the banner was written, it was possible to explicitly tell the browser to go to the http version, but it seems that this is no longer possible (it looks like we're enabling HSTS, which is why your browser isn't allowing you to access the http version).

The workaround for now is to host webrepl.html locally.

closed 2022-05-14 by jimmo

beyonlo, 2022-05-14:
@jimmo Hello.

**Short answer: Opera works.**

Sorry, I see the banner, but it was not clear that after change to HTTP,  browser back to the HTTPS, and I did not noticed that.

This problem (back to https when use http) do not happen on the Opera browser - works fine.

My goal is to run the webrepl.html inside the MicroPython on the ESP32 using the [microdot](https://github.com/miguelgrinberg/microdot) webserver. The Microdot was able to host the webrepl.html and show complete terminal, and when I tried to connect the WebREPL (using http for sure - like as opera), it do not connect. BUT, if I stop (CTRL+C) the Microdot webserver, and page webrepl.html still opened (hosted on microdot), the websocket open and WebREPL works. If I start microdot again, WebREPL stop. **Any idea what is this problem?**

This is my code to run WebREPL client on the MicroPython using Microdot:

**rep.py:**
```
from microdot import Microdot, redirect, send_file

app = Microdot()

@app.route('/term')
def term(request):
    response = send_file('term.js')
    response.headers = {'Content-Type': 'text/js'}
    return response

@app.route('/')
def rep(request):
    response = send_file('webrepl.html')
    response.headers = {'Content-Type': 'text/html'}
    return response

app.run(debug=True)
```
In the https://github.com/micropython/webrepl/blob/master/webrepl.html I just change:

Line 42:
from this: `<script src="term.js"></script>`
to this: `<script src="/term"></script>`

Line 43: Remove it - for test I do not need save file for now.

Just put the `webrepl.html`  and the `term.js` into Micropython and run the code above (`rep.py`)

beyonlo, 2022-05-14:
@jimmo hey, works (in parts) now on the microdot. I just used microdot in asyncio mode. I was using in thread mode :)

It shows:

```
Welcome to MicroPython!                                                                                                                               
Password:                                                                                                                                             
WebREPL connected                                                                                                                                     
>>> 
```
So, I five my password, and connect, but stop after that point (above).

In the real terminal (Linux) show this:

```
import rep
Starting async server on 0.0.0.0:80...
WebREPL connection from: ('192.168.1.6', 38044)
Task exception wasn't retrieved
future: <Task> coro= <generator object 'serve' at 3fcc0d80>
Traceback (most recent call last):
  File "uasyncio/core.py", line 1, in run_until_complete
  File "microdot_asyncio.py", line 204, in serve
  File "microdot_asyncio.py", line 317, in dispatch_request
  File "microdot_asyncio.py", line 131, in write
  File "uasyncio/stream.py", line 1, in stream_awrite
  File "uasyncio/stream.py", line 1, in drain
OSError: [Errno 104] ECONNRESET
GET / 200
dupterm: Exception in read() method, deactivating: OSError: [Errno 104] ECONNRESET
GET /term
WebREPL connection from: ('192.168.1.6', 38046)
dupterm: EOF received, deactivating

```
Any idea?

Thank you in advance!