micropython/micropython #8668: i'm trying  compiling  the javascript port and i got those errors in linking stage of firmware.js

tektecher, 2022-05-15:
wasm-ld: error: build/py/qstr.o: undefined symbol: mp_qstr_frozen_const_pool
wasm-ld: error: build/py/qstr.o: undefined symbol: mp_qstr_frozen_const_pool
wasm-ld: error: build/py/qstr.o: undefined symbol: mp_qstr_frozen_const_pool
wasm-ld: error: build/py/qstr.o: undefined symbol: mp_qstr_frozen_const_pool
wasm-ld: error: build/py/qstr.o: undefined symbol: mp_qstr_frozen_const_pool
wasm-ld: error: build/py/objmodule.o: undefined symbol: uos_module
wasm-ld: error: build/py/objmodule.o: undefined symbol: mp_module_machine
wasm-ld: error: build/py/builtinhelp.o: undefined symbol: mp_frozen_mpy_names
wasm-ld: error: build/py/builtinhelp.o: undefined symbol: mp_frozen_mpy_names
wasm-ld: error: build/py/builtinhelp.o: undefined symbol: mp_frozen_mpy_names
wasm-ld: error: build/py/builtinhelp.o: undefined symbol: mp_frozen_mpy_names
wasm-ld: error: build/py/builtinhelp.o: undefined symbol: mp_frozen_mpy_names
wasm-ld: error: build/py/builtinhelp.o: undefined symbol: mp_frozen_mpy_names
wasm-ld: error: build/py/frozenmod.o: undefined symbol: mp_frozen_mpy_content
emcc: error:

jimmo, 2022-05-17:
@tektecher Could you please provide more information, e.g. OS, compiler, command line, micropython git hash, etc. Have you done a `make clean` first?

I am able to successfully compile the javascript port with emcc 3.1.9 on linux, and the continuous integration is currently passing too.

tektecher, 2022-05-17:
thanks for your reply. I'm using Ubuntu and micropython version V1.9.0.  emcc 3.1.10 and yes i did make clean

jimmo, 2022-05-17:
>micropython version V1.9.0

@tektecher The v1.9 tag (there is no v1.9.0) is from 2017 (which predates the javascript port).

tektecher, 2022-05-17:
i forget to mention i have added my board folder (inside javascript port folder) which is based on esp32. and modified the mpconfigport.h. 

/*
 * This file is part of the MicroPython project, http://micropython.org/
 *
 * The MIT License (MIT)
 *
 * Copyright (c) 2013, 2014 Damien P. George and 2017, 2018 Rami Ali
 *
 * Permission is hereby granted, free of charge, to any person obtaining a copy
 * of this software and associated documentation files (the "Software"), to deal
 * in the Software without restriction, including without limitation the rights
 * to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
 * copies of the Software, and to permit persons to whom the Software is
 * furnished to do so, subject to the following conditions:
 *
 * The above copyright notice and this permission notice shall be included in
 * all copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 * FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
 * AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 * LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
 * OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
 * THE SOFTWARE.
 */

// Board-specific definitions
#include "mpconfigboard.h"

#include <stdint.h>

// options to control how MicroPython is built

// You can disable the built-in MicroPython compiler by setting the following
// config option to 0.  If you do this then you won't get a REPL prompt, but you
// will still be able to execute pre-compiled scripts, compiled with mpy-cross.
#define MICROPY_ENABLE_COMPILER     (1)

// Python internal features
/*#define MICROPY_READER_VFS                  (1)
#define MICROPY_ENABLE_FINALISER            (1)
#define MICROPY_STACK_CHECK                 (1)*/
#define MICROPY_ENABLE_EMERGENCY_EXCEPTION_BUF (1)
#define MICROPY_REPL_EMACS_KEYS             (1)
#define MICROPY_ERROR_REPORTING             (MICROPY_ERROR_REPORTING_NORMAL)
#define MICROPY_MODULE_FROZEN_STR           (0)
#define MICROPY_MODULE_FROZEN_MPY           (1)
#define MICROPY_QSTR_EXTRA_POOL             (mp_qstr_frozen_const_pool)
/*#define MICROPY_VFS                         (1)*/

#define MICROPY_QSTR_BYTES_IN_HASH  (2)
#define MICROPY_ALLOC_PATH_MAX      (256)
#define MICROPY_ALLOC_PARSE_CHUNK_INIT (16)
#define MICROPY_EMIT_X64            (0) // BROKEN
#define MICROPY_EMIT_THUMB          (0) // BROKEN
#define MICROPY_EMIT_INLINE_THUMB   (0)
#define MICROPY_COMP_MODULE_CONST   (0)
#define MICROPY_COMP_CONST          (1)
#define MICROPY_COMP_DOUBLE_TUPLE_ASSIGN (1)
#define MICROPY_COMP_TRIPLE_TUPLE_ASSIGN (0)
#define MICROPY_MEM_STATS           (0) // BROKEN
#define MICROPY_DEBUG_PRINTERS      (0)
#define MICROPY_ENABLE_GC           (1)
#define MICROPY_GC_ALLOC_THRESHOLD  (1)
#define MICROPY_GC_USES_ALLOCATED_SIZE (1)
#define MICROPY_REPL_EVENT_DRIVEN   (1)
#define MICROPY_HELPER_REPL         (1)
#define MICROPY_HELPER_LEXER_UNIX   (0)
[... 126 lines cut ...]
// different targets may be defined in different ways - either as int
// or as long. This requires different printf formatting specifiers
// to print such value. So, we avoid int32_t and use int directly.
#define UINT_FMT "%u"
#define INT_FMT "%d"
typedef int mp_int_t; // must be pointer size
typedef unsigned mp_uint_t; // must be pointer size
typedef long mp_off_t;

// extra built in names to add to the global namespace
#define MICROPY_PORT_BUILTINS \
    { MP_ROM_QSTR(MP_QSTR_open), MP_ROM_PTR(&mp_builtin_open_obj) },

// We need to provide a declaration/definition of alloca()
#include <alloca.h>

#ifndef MICROPY_HW_BOARD_NAME
    #define MICROPY_HW_BOARD_NAME "JS"
    #define MICROPY_HW_MCU_NAME "Emscripten"
#endif

#define MP_STATE_PORT MP_STATE_VM

#define MICROPY_PORT_ROOT_POINTERS \
    const char *readline_hist[8];

jimmo, 2022-05-17:
> i forget to mention i have added my board folder (inside javascript port folder) which is based on esp32. and modified the mpconfigport.h.

@tektecher This is an important detail!!!

You've enabled a bunch of things that aren't supported on the javascript port. The diff is

```diff
26a27,29
> // Board-specific definitions
> #include "mpconfigboard.h"
> 
35a39,50
> // Python internal features
> /*#define MICROPY_READER_VFS                  (1)
> #define MICROPY_ENABLE_FINALISER            (1)
> #define MICROPY_STACK_CHECK                 (1)*/
> #define MICROPY_ENABLE_EMERGENCY_EXCEPTION_BUF (1)
> #define MICROPY_REPL_EMACS_KEYS             (1)
> #define MICROPY_ERROR_REPORTING             (MICROPY_ERROR_REPORTING_NORMAL)
> #define MICROPY_MODULE_FROZEN_STR           (0)
> #define MICROPY_MODULE_FROZEN_MPY           (1)
> #define MICROPY_QSTR_EXTRA_POOL             (mp_qstr_frozen_const_pool)
> /*#define MICROPY_VFS                         (1)*/
> 
87c102
< #define MICROPY_ENABLE_PYSTACK      (1)
---
> // #define MICROPY_ENABLE_PYSTACK      (1)
125c140,142
< #define MICROPY_SCHEDULER_DEPTH     (1)
---
> #define MICROPY_SCHEDULER_DEPTH     (16)
> 
> #define MICROPY_PY_MACHINE                  (1)
129a147,148
> extern const struct _mp_obj_module_t uos_module;
> extern const struct _mp_obj_module_t mp_module_machine;
132a152,153
>     { MP_ROM_QSTR(MP_QSTR_uos), MP_ROM_PTR(&uos_module) }, \
>     { MP_OBJ_NEW_QSTR(MP_QSTR_machine), (mp_obj_t)&mp_module_machine }, \
190,191c211,214
< #define MICROPY_HW_BOARD_NAME "JS"
< #define MICROPY_HW_MCU_NAME "Emscripten"
---
> #ifndef MICROPY_HW_BOARD_NAME
>     #define MICROPY_HW_BOARD_NAME "JS"
>     #define MICROPY_HW_MCU_NAME "Emscripten"
> #endif
```

You're adding uos and machine, which aren't defined on the javascript port (hence the linker error for uos_module and mp_module_machine) and support for frozen mpy (hence the linker errors for mp_qstr_frozen_const_pool and mp_frozen_mpy_names).

tektecher, 2022-05-17:
so what is your advice for me. it's my first time to deal with python and i tried to do my port as in this repo https://github.com/microBlock-IDE/micropython/tree/V1.8.0/ports/javascript

dlech, 2022-05-17:
> i forget to mention i have added my board folder (inside javascript port folder) which is based on esp32.

If you have an ESP32 board, why are you using the Javascript port instead of the the ESP32 port?

tektecher, 2022-05-17:
@dlech  because i want to make a simulator for the board

dpgeorge, 2022-05-17:
It's great that you want to make a simulator, but this is not the place to get help doing that.  If you have a bug report then you can report that here.  Otherwise https://forum.micropython.org is the place to discuss development.

closed 2022-05-17 by dpgeorge