micropython/micropython #8671: ESP32-C3 BLE Long Range (Coded PHY) support

Solmis, 2022-05-16:
Hello, I would like to use BLE Long Range in communication between the other device and my micropython application. I'm using ESP32-C3, which supports this BLE feature. As far as I know this is not supported by micropython, as well as the extended advertising, am I right? Are you planning to add Coded PHY support to micropython? I saw that micropython port to esp32 uses esp-idf, which itself supports both extended advertising and coded PHY. I've started changing the micropython (port) source code by myself, but so far without success. Could you give me some tips on how to achieve extended advertising & coded phy support in BLE in micropython? In fact I need to scan and connect with other device (which advertsises in extended advertising mode and coded PHY (125kbps)).

stephanelsmith, 2022-05-17:
This feature would also be applicable to ESP32-S3 with BLE 5 support.

radziq1302, 2022-06-06:
Is there any progress on this issue?

FrankTT3D, 2023-06-14:
Long Range PHY coded opens a wealth of IoT applications. I was trying to find out accessible modes from the current µPython docu, but it seems we cannot alter the BT transmission mode whatsoever?
Could anyone give a statement on plans for implementation for the Pi Pico W or the ESP32?

jonnor, 2024-09-08:
Hi. MicroPython is a community ran open-source project, driven by its contributors. So to enable new features such as this one, the project relies on people willing to research, implement a proof-of-concept, create a merge request, and then work together with maintainers to get the MR into an acceptable state.

Josverl, 2026-08-24:
wrt to an possible API extension; 
it should be possible to add a `phy` parameter to the relevant methods and default that to the current behaviour, and to improve ergonomics add a way to set a global default.

- If `phy` is omitted → default 1M (current behavior)
- If unsupported → raise `OSError(EINVAL)`
- If extended advertising is required → automatically enable it

It would require some code to detect setting not supported by the underlying hardware or stack - so if that guard is needed (+1) there would be some size impact to boards that have no benefit from this.

``` py 
import bluetooth
from bluetooth import (
    PHY_1M,
    PHY_2M,
    PHY_CODED_S2,
    PHY_CODED_S8,
)

ble = bluetooth.BLE()
ble.active(True)

# Set default PHY globally (New Method) 
ble.set_default_phys(tx_phy=PHY_CODED_S8,
                     rx_phy=PHY_CODED_S8)

# explicit scan for long-range devices
ble.gap_scan(5000, phy=PHY_CODED_S8)

# Explicit advertisement using coded PHY
ble.gap_advertise(100000, adv_data=b"LR", phy=PHY_CODED_S8)

# Controller attempts connection using the specified PHY
# Optionally - If peer does not support it, fall back to 1M automatically
ble.gap_connect(
    addr_type,
    addr,
    timeout_ms=5000,
    phy=PHY_CODED_S8,
)
```
perhaps also add a way to get the PHYs used by a specific connection ?
`tx, rx = ble.connection_phys(conn_handle); print(tx, rx)`

agatti, 2026-08-24:
Thanks for the proposed API, it looks quite reasonable.

For the implementation, maybe this is better to begin as a PR that just adds optional 2M PHY support first.  This would still require adding almost every user-facing change proposed here and give the opportunity for a discussion on the new API as well.  In theory this also forces having to look into each port's way to know how to detect PHY capabilities too (if such a thing is possible).

Once that is sorted out, adding coded PHY should be almost transparent for the user besides a few new constants, and thus hopefully easier to reason with from a development/reviewer perspective.

agatti, 2026-08-24:
There's an interesting situation: whilst adding 2M (and therefore coded phy, just change the constants) to NimBLE with a slight modification to the proposed API [1] took ~15 minutes, adding that to BTStack is slightly more complex.

Apparently BTStack API allows per-connection PHY changes, and they have to occur from inside an event callback - but said callback does not have any obvious place to fit the requested phy mode for the connection.  I can store that as a global variable but then the whole API needs to be rethought.

Either that or just leave BTStack behind, who cares about Raspberry Pi Picos anyway... :)

[1] Setting the phy is done via `bluetooth.BLE.config(tx_phy=..., rx_phy=...)` instead of with a dedicated function.

Josverl, 2026-08-24:
> Either that or just leave BTStack behind, who cares about Raspberry Pi Picos anyway... :)

well, it would be rather nice ....

I had been testing with the Pico2_W with similar results. Perhaps the binary in `cyw43_btfw_43439.h` limits it to BLE 4.2 ?
_CYW43439 capability:_
  With the C3 as central (known-good implementation) connected to the Pico 2 W,
  requesting a PHY change gives:
      ble.gap_set_phy(h, tx_phy=PHY_2M,    rx_phy=PHY_2M)    -> OSError -538
      ble.gap_set_phy(h, tx_phy=PHY_CODED, rx_phy=PHY_CODED) -> OSError -538
  -538 decodes as BLE_HS_ERR_HCI_BASE (0x200) + 0x1A =
  BLE_ERR_UNSUPP_REM_FEATURE, "Unsupported Remote Feature". 

>  bluetooth.BLE.config(tx_phy=..., rx_phy=...) instead of with a dedicated function.

Much nicer