micropython/micropython #8673: ESP_ERR_INVALID_ARG on write to RGB LED on ESP32-S3-DevKitC-1

RetiredWizard, 2022-05-16:
I've flashed the "Generic ESP32-S3 (SPIRAM) Espressif" version v1.18 on to an ESP32-S3-DevKitC-1-N8R2 and when I try to turn on the RGB LED using the following code:

    import machine,neopixel
    pixels = neopixel.NeoPixel(machine.Pin(48),1)
    pixels.fill( (100,100,100) )
    pixels.write()

The pixels.write() command returns the following traceback:

    Traceback (most recent call last):
      File "<stdin>", line 1, in <module>
      File "neopixel.py", line 1, in write
    OSError: (-258, 'ESP_ERR_INVALID_ARG')

Looking at the neopixel library on github the error appears to be coming from the machine.bitstream function.

I can't tell if there's a problem with the arguments the neopixel class is passing to bitstream or if the bitstream function is just not ready for this board yet.

I posted this as a question to the Micropython forum and a suggestion was made to create a github issue with the comment that "ESP_ERR shouldn't bubble up like that".

dpgeorge, 2022-05-17:
Can you please try firmware without SPIRAM, just the generic S3 from here: https://micropython.org/download/GENERIC_S3/

Also please try the latest nightly build (both SPIRAM and non-SPIRAM).

jbecknyc, 2022-05-31:
The issue is RMT.bitstream_channel defaulting to 7.  While as ESP32 has 8 RMT Tx channels, the ESP32S3 has only 4, so the default should be less than 4 for the S3 variant.  In the meantime, the neopixel driver works fine  if you just set the bitstream_channel yourself, i.e.:
```
from esp32 import RMT
RMT.bitstream_channel(3)
```

jonnor, 2024-09-08:
A fix is provided above.

closed 2024-09-08 by jonnor