micropython/micropython #8676: How to package the python program into a UF2 file with MicroPython

MarchAwake, 2022-05-17:
How to package the python program into a UF2 file with MicroPython?

MarchAwake, 2022-05-17:
Thanks a lot @jimmo

MarchAwake, 2022-05-17:
freezing can create main.py in the filesystem

jimmo, 2022-05-17:
> freezing can create main.py in the filesystem

Not exactly, frozen files are not in the filesystem. However, you can freeze main.py and use it instead of the filesystem version. This was discussed recently in #8660

MarchAwake, 2022-05-18:
I start the service in boot.py. What negative impact does this have on MicroPython?

robert-hh, 2022-05-18:
None. boot.py and main.py are executed one after the other and run in the same context.

jimmo, 2022-05-18:
> None. boot.py and main.py are executed one after the other and run in the same context.

On ESP32 this is true (there isn't really a lot of point to boot.py on ESP32). However, on STM32, boot.py runs before network and USB init. boot.py should technically only be used for configuring usb, network, (and filesystems to find main.py if necessary).

mattytrentini, 2022-06-13:
If this is clear now @MarchAwake can you please close this ticket? Thanks!

jonnor, 2024-09-08:
Clarifications have been provided and no response indicating this did not resolve the issue in 2 years.

closed 2024-09-08 by jonnor