micropython/micropython #8681: Test suite: Problem with inet tests and the MIMXRT port.

robert-hh, 2022-05-19:
Setup: MIMXRT board with the latest firmware, connected by Ethernet to a PC.
Test case: Trying to run `multi_net/tcp_accept_recv.py` or others from that set.

Observation:

a) The test runs fine when instance0 is micropython and instance1 is the board.
b) It fails when instance0 is the board and instance1 is micropython. The log is below.

When the test fails, there is no activity traced by wireshark. When the tests succeds, the expected activity is visible with wireshark.
Running the statements of the script manually at the two instances reveals an error no. 128 instead of 107 for not connected.
So what should be changed?
Log:
```
micropython/tests$ ./run-multitests.py -i pyb:a0 multi_net/tcp_accept_recv.py
multi_net/tcp_accept_recv.py on ttyACM0|micropython: FAIL
### TEST ###
--- instance0 ---
TIMEOUT
--- instance1 ---
Traceback (most recent call last):
  File "<stdin>", line 80, in <module>
  File "<stdin>", line 30, in instance1
OSError: [Errno 111] ECONNREFUSED
### TRUTH ###
--- instance0 ---
107
--- instance1 ---

### DIFF ###
--- /tmp/tmpqz4d9lma	2022-05-19 17:05:34.130534000 +0200
+++ /tmp/tmpjcts2wz_	2022-05-19 17:05:34.130534000 +0200
@@ -1,4 +1,7 @@
 --- instance0 ---
-107
+TIMEOUT
 --- instance1 ---
-
+Traceback (most recent call last):
+  File "<stdin>", line 80, in <module>
+  File "<stdin>", line 30, in instance1
+OSError: [Errno 111] ECONNREFUSED
1 tests performed
0 tests passed
1 tests failed: multi_net/tcp_accept_recv.py
```

robert-hh, 2022-05-19:
So I found the answers:
a) in run_multitest.py getting the IP address works only for WLAN. It fails on LAN, and then the HOST_IP is used, which is wrong. Using code which gets the LAN IP works.
b) in  uasyncio/stream.py and in ssl_data.py there are statements as `import usocket as socket`. When using LWIP, there is no usocket module, causing uasyncio_tcp_server_client.py to fail and ssl_data.py to be skipped. Using just `import socket` solves that.
What would be the best way to fix it? For the usocket/socket problem, one could try to import usocket first, and the socket. But since the is the default mechanism for importing the 'u' modules, `import socket` should be fine.
For the WLAN/LAN case, the code section could be replaced by:
```
        try:
            import network
            if hasattr(network, "WLAN"):
                ip = network.WLAN().ifconfig()[0]
            else: 
                ip = network.LAN().ifconfig()[0]
        except:
            ip = HOST_IP
```

robert-hh, 2022-05-20:
The missing usocket module is back with #8684.

robert-hh, 2022-05-21:
The set of https://github.com/micropython/micropython/pull/8678, https://github.com/micropython/micropython/issues/8684 and https://github.com/micropython/micropython/pull/8686 fixes the problem. At least two of them could go into a single PR.

dpgeorge, 2022-05-24:
All issues should be addressed by 75efb3267c7ec54f7131906d6ab1a3ab5185a185, beeb250d58b70b60a01dae3e2fac2741f56381f5 and dd35f76db330851bdf937585219d5cae21222fa3

closed 2022-05-24 by dpgeorge