micropython/micropython #8683: esp32s3 USB JTAG Serial interface should be enabled to access the mpy terminal

wuyuanyi135, 2022-05-20:
# Problem Description
I want to use the built-in USB serial functionality comes with the ESP32-S3 SoC in addition to the UART interface (not simultaneously). However, I found that the firmwares (both [v1.18 (2022-01-17) .bin](https://micropython.org/resources/firmware/GENERIC_S3_SPIRAM-20220117-v1.18.bin) and [v1.18-473-g7883ae413 (2022-05-19) .bin](https://micropython.org/resources/firmware/GENERIC_S3_SPIRAM-20220519-unstable-v1.18-473-g7883ae413.bin) do not support the serial over USB JTAG Serial interface. The sympton was that the device got reset as soon as I opened the USB serial port.

I compile my own firmware using the master branch
(7883ae413ddfa6181d784533b236658658383d0c) with
```
make -C mpy-cross 
make  -C ports/esp32 SER_C_MODULES=../../../examples/usercmodule/micropython.cmake  FROZEN_MANIFEST=$(pwd)/ports/esp32/boards/manifest_test.py BOARD=GENERIC_S3_SPIRAM
make -C ports/esp32 submodules BOARD=GENERIC_S3_SPIRAM
make -C ports/esp32  BOARD=GENERIC_S3_SPIRAM
```

and flashed the firmware with
```
python -m esptool --chip esp32s3 -b 460800 --before default_reset --after no_reset --chip esp32s3  write_flash  0x0 bui
ld-GENERIC_S3_SPIRAM/bootloader/bootloader.bin 0x8000 build-GENERIC_S3_SPIRAM/partition_table/partition-table.bin 0x10000 build-GENERIC_S3_SPIRAM/micropython.bin
```

The USB serial no longer reset the board. However, the USB Serial terminal has no feedback: I can see from the UART terminal that the input was accepted (e.g., more `>>>` after hit the enter key) but nothing showed up in the USB Terminal.

# Solution
I have added one line to `micropython/ports/esp32/boards/GENERIC_S3_SPIRAM/sdkconfig.board` file:

```
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y
```

And the USB terminal works just fine. It is shared by the UART one but both side will see the same feedback consistently. 

In addition, I have tried to comment out the line in`micropython/ports/esp32/boards/GENERIC_S3_SPIRAM/mpconfigboard.cmake`:
```
set(IDF_TARGET esp32s3)

set(SDKCONFIG_DEFAULTS
    boards/sdkconfig.base
#    boards/sdkconfig.usb
    boards/sdkconfig.spiram_sx
    boards/GENERIC_S3_SPIRAM/sdkconfig.board
)
```
And the USB terminal still works. 

# Recommendations
Add `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y` to the esp32s3 boards so that the USB serial will be enabled.

# Possible related issues
#8524

jonnor, 2024-09-08:
This might have been fixed by https://github.com/micropython/micropython/commit/fe9eaf12f391232f195ac052005ac2b852409dd6

Can you please test and verify on a recent MicroPython? Version 1.23.0 or later

projectgus, 2025-02-04:
The `ESP32_GENERIC_S3` firmware now works over the USB port using the native USB interface (not actually Serial/JTAG, the native USB OTG, but for serial access the functionality is the same).

If you specifically want to use Serial/JTAG peripheral instead, on recent MicroPython you should be able to make a custom build config like this:

```
#define MICROPY_HW_USB_CDC                  (0)
#define MICROPY_HW_ESP_USB_SERIAL_JTAG (1) 
```

closed 2025-02-04 by projectgus