micropython/micropython #8685: Error "array subscript 0 is outside array bounds" on arm-none-eabi-gcc 12.1

jimmo, 2022-05-20:
FYI for anyone who runs into this. I'm seeing this error with the arm-none-eabi-gcc 12.1 Arch Linux package. Appears to affect at least rp2 and cc3200. In the case of rp2 it also affects their pico-examples projects too (although they don't compile with `-Werror` by default). For now the solution is to downgrade to 11.x.

See https://gcc.gnu.org/bugzilla/show_bug.cgi?id=99578 for the details. It was reverted in 11.3 (https://github.com/gcc-mirror/gcc/commit/91f7d7e1bb6827bf8e0b7ba7eb949953a5b1bd18) but it's not clear that the same fix was applied to 12.x? I don't see it anywhere in the history of the 12.1 tag? Not quite sure what's going, will update this bug with more info.

sosi-deadeye, 2022-05-23:
Temporary fix in micropython/ports/rp2/CMakeLists.txt:
```
target_compile_options(${MICROPY_TARGET} PRIVATE
    -Wall
    # -Werror
)
```

Don't use this for production. Yesterday I compiled a version for RP Pico. It worked.

projectgus, 2022-06-20:
A slightly lighter workaround for rp2 is to run

```
make -j CFLAGS=-Wno-array-bounds
```
(or can change the commented line in the ports/rp2/CMakeLists.txt file to `-Werror -Wno-array-bounds`)

jonnor, 2024-09-08:
Is this still an issue ?

Note that arm-none-eabi-gcc in Arch Linux is now at version 14.1.0

dpgeorge, 2024-09-08:
I think we can close this.  I'm on Arch Linux and not run into this lately.

closed 2024-09-08 by dpgeorge