micropython/micropython #8688: ESP8266 v1.18 Unreliable UDP Receiving From Broadcast Address (ex 192.168.1.255)

PsuFan, 2022-05-21:
Anyone else having issues receiving UDP packets? I can send 10-20 packets and only receive one packet. I send them with another esp or with my computer and can see the packets in Wireshark but hardly any are received with the 8266. It seems to be substantially worse when I send to the broadcast (ex 192.168.1.255) vs directly to the esp IP address. Should I be using a multicast address instead? Not sure what the differences are.

```
socket = socket(AF_INET, SOCK_DGRAM)
socket.setsockopt(SOL_SOCKET, SO_REUSEADDR, 1)
socket.bind(('', 1011))
socket.setblocking(False)

while True:
	try:
		data, addr = self.socket.recvfrom(1024)
		print('Recv Data From %s' % str(addr))
	except OSError as err:
		if err.args[0] != 11:
			print(err)

jonnor, 2024-09-08:
Is this issue still present on latest MicroPython, version 1.23 or later?

I am assuming this is on WiFi. Congestion, signal quality and routing issues have serious impacts on UDP packet drop rate.
So key questions are: What is the package drop rate when compared to two computers on the same WiFi network? At similar locations.
What is the signal strength of the ESP like? Does it happen on all WiFi networks?

PsuFan, 2024-09-08:
The real question you should be asking is why is there a difference between direct and broadcast. Does esp drop boadcasts if it’s busy or something? Yes it’s WiFi, I was able to reliably by send the UDP directly to the desired IP.

closed 2024-09-08 by PsuFan