micropython/micropython #8693: Motor  doesn't work under idf 4.2.2 + micropython 1.18

JackieLs, 2022-05-24:
I build the esp32-micropython which is  based v1.18 with the idf-4.2.2, It's worked well for led, except the motor.
But when I use the micropython master branch with the idf-4.4.1.All is right.
1、Is there any konwn reason about this?
2、Is the master branch is a statable branch ? May I use this branch?

dpgeorge, 2022-05-24:
You will need to provide more information about your hardware (which board, which motor) and the MicroPython code that you are running, and also what you mean by "motor doesn't work".

JackieLs, 2022-05-29:
[image]

we found here is the reason. and we correct it.

dpgeorge, 2022-05-30:
It's not clear what the problem is here.  It sounds like:
- MicroPython v1.18 with IDF v4.2.2 has problems with PWM driving a motor
- MicroPython master with IDF v4.1.1 does not have this problem
- it's related to the `oen_inv` argument to `gpio_matrix_out`

Can you give more details?

FanPower, 2022-05-31:
[image]

I have a similar problem，when I use MicroPython v1.18 with IDF v4.2.1. I have problems with PWM driving a motor.

steps:

1.use GPIO_A  output PWM_A;
2.PWM_A.STOP;
3.use GPIO_B  output PWM_B，At this time，GPIO_A also output PWM_B，Until I follow the image.