micropython/micropython #8696: `stm32/port` build all failed under ubuntu18.04(take stm32f769-discovery as example)

r00t1900, 2022-05-24:
## environment
- compiler version: GNU Tools for Arm Embedded Processors 7-2018-q3-update
- os version: ubuntu 18.04 amd64
- python version: 3.5 or 3.6
- build target: STM32F769-DISCOVERY
- build command: `make BOARD=STM32F769DISC`

## question

Every STM32 port can not build successfully, always end with:
```
../../py/mkrules.mk:104: recipe for target 'build-STM32F769DISC/genhdr/qstr.i.last' failed
make: *** [build-STM32F769DISC/genhdr/qstr.i.last] Error 1
make: *** Deleting file 'build-STM32F769DISC/genhdr/qstr.i.last'
```
> I've built `mpy-cross` by `make -C mpy-cross`.

I also notice that in `tools/c.sh` there are compiling command for `stm32f769-discovery`. But why it only invoke the `mboot`? I can build `mboot` but it appearently not all the main firmware cause it is just a bootloader.

**So why can't I build micropython for stm32f769?**

## build output

```
# make BOARD=STM32F769DISC
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build-STM32F769DISC/genhdr/qstr.i.last
In file included from ../../py/mpconfig.h:62:0,
                 from ../../py/mpstate.h:31,
                 from ../../lib/lv_bindings/lvgl/src/hal/../misc/lv_gc.h:72,
                 from ../../lib/lv_bindings/lvgl/src/hal/lv_hal_disp.c:15:
./mpconfigport.h:336:0: error: "LV_ROOTS" redefined [-Werror]
 #define LV_ROOTS
 
In file included from ../../lib/lv_bindings/lvgl/src/hal/lv_hal_disp.c:15:0:
../../lib/lv_bindings/lvgl/src/hal/../misc/lv_gc.h:66:0: note: this is the location of the previous definition
 #define LV_ROOTS LV_ITERATE_ROOTS(LV_DEFINE_ROOT)
 
cc1: all warnings being treated as errors
In file included from ../../py/mpconfig.h:62:0,
                 from ../../py/mpstate.h:31,
                 from ../../lib/lv_bindings/lvgl/src/hal/../misc/lv_gc.h:72,
                 from ../../lib/lv_bindings/lvgl/src/hal/lv_hal_indev.c:15:
./mpconfigport.h:336:0: error: "LV_ROOTS" redefined [-Werror]
 #define LV_ROOTS
 
In file included from ../../lib/lv_bindings/lvgl/src/hal/lv_hal_indev.c:15:0:
../../lib/lv_bindings/lvgl/src/hal/../misc/lv_gc.h:66:0: note: this is the location of the previous definition
 #define LV_ROOTS LV_ITERATE_ROOTS(LV_DEFINE_ROOT)
 
cc1: all warnings being treated as errors
In file included from ../../py/mpconfig.h:62:0,
                 from ../../py/mpstate.h:31,
                 from ../../lib/lv_bindings/lvgl/src/misc/lv_gc.h:72,
                 from ../../lib/lv_bindings/lvgl/src/misc/lv_mem.c:12:
./mpconfigport.h:336:0: error: "LV_ROOTS" redefined [-Werror]
 #define LV_ROOTS
 
In file included from ../../lib/lv_bindings/lvgl/src/misc/lv_mem.c:12:0:
../../lib/lv_bindings/lvgl/src/misc/lv_gc.h:66:0: note: this is the location of the previous definition
 #define LV_ROOTS LV_ITERATE_ROOTS(LV_DEFINE_ROOT)
 
cc1: all warnings being treated as errors
In file included from ../../py/mpconfig.h:62:0,
                 from ../../py/mpstate.h:31,
                 from ../../lib/lv_bindings/lvgl/src/misc/../misc/lv_gc.h:72,
                 from ../../lib/lv_bindings/lvgl/src/misc/lv_style.c:10:
./mpconfigport.h:336:0: error: "LV_ROOTS" redefined [-Werror]
 #define LV_ROOTS
 
[... 216 lines cut ...]
In file included from ../../lib/lv_bindings/lvgl/src/draw/sw/lv_draw_sw_gradient.c:10:0:
../../lib/lv_bindings/lvgl/src/draw/sw/../../misc/lv_gc.h:66:0: note: this is the location of the previous definition
 #define LV_ROOTS LV_ITERATE_ROOTS(LV_DEFINE_ROOT)
 
cc1: all warnings being treated as errors
Traceback (most recent call last):
  File "../../py/makeqstrdefs.py", line 188, in <module>
    preprocess()
  File "../../py/makeqstrdefs.py", line 62, in preprocess
    for output in p.imap(pp(flags), chunks):
  File "/usr/lib/python3.5/multiprocessing/pool.py", line 695, in next
    raise value
  File "/usr/lib/python3.5/multiprocessing/pool.py", line 119, in worker
    result = (True, func(*args, **kwds))
  File "../../py/makeqstrdefs.py", line 46, in run
    return subprocess.check_output(args.pp + flags + files)
  File "/usr/lib/python3.5/subprocess.py", line 626, in check_output
    **kwargs).stdout
  File "/usr/lib/python3.5/subprocess.py", line 708, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['arm-none-eabi-gcc', '-E', '-DUSE_QSPI_XIP=0', '-I../../lib/lv_bindings', '-I../../lib/lv_bindings/driver/png/lodepng', '-I../../lib/lwip/src/include', '-I.', '-I../..', '-Ibuild-STM32F769DISC', '-I../../lib/cmsis/inc', '-I../../lib/stm32lib/CMSIS/STM32F7xx/Include/', '-I../../lib/stm32lib/STM32F7xx_HAL_Driver/Inc', '-Iusbdev/core/inc', '-Iusbdev/class/inc', '-Ilwip_inc', '-Wall', '-Wpointer-arith', '-Wno-error=format', '-Werror', '-Wdouble-promotion', '-Wfloat-conversion', '-std=gnu99', '-nostdlib', '-Wno-unused-function', '-DLODEPNG_NO_COMPILE_ENCODER', '-DLODEPNG_NO_COMPILE_DISK', '-DLODEPNG_NO_COMPILE_ALLOCATORS', '-DFFCONF_H="lib/oofatfs/ffconf.h"', '-DMICROPY_VFS_FAT=1', '-DMICROPY_PY_USSL=1', '-DMICROPY_SSL_MBEDTLS=1', '-I../../lib/mbedtls/include', '-DMICROPY_PY_LWIP=1', '-DMBEDTLS_CONFIG_FILE="mbedtls/mbedtls_config.h"', '-DSTM32F769xx', '-DUSE_FULL_LL_DRIVER', '-mthumb', '-mfpu=fpv5-d16', '-mfloat-abi=hard', '-mtune=cortex-m7', '-mcpu=cortex-m7', '-Os', '-DNDEBUG', '-Iboards/STM32F769DISC', '-DSTM32_HAL_H=<stm32f7xx_hal.h>', '-DMBOOT_VTOR=0x08000000', '-DMICROPY_HW_VTOR=0x08000000', '-DMICROPY_FLOAT_IMPL=MICROPY_FLOAT_IMPL_DOUBLE', '-fdata-sections', '-ffunction-sections', '-DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool', '-DMICROPY_MODULE_FROZEN_MPY', '-DMICROPY_MODULE_FROZEN_STR', '-DMICROPY_ROM_TEXT_COMPRESSION=1', '-DNO_QSTR', '../../lib/lv_bindings/lvgl/src/widgets/lv_checkbox.c', '../../lib/lv_bindings/lvgl/src/widgets/lv_dropdown.c', '../../lib/lv_bindings/lvgl/src/widgets/lv_label.c', '../../lib/lv_bindings/lvgl/src/widgets/lv_arc.c', '../../lib/lv_bindings/lvgl/src/widgets/lv_textarea.c', '../../lib/lv_bindings/lvgl/src/widgets/lv_slider.c', '../../lib/lv_bindings/lvgl/src/widgets/lv_canvas.c', '../../lib/lv_bindings/lvgl/src/widgets/lv_btnmatrix.c', '../../lib/lv_bindings/lvgl/src/widgets/lv_roller.c', '../../lib/lv_bindings/lvgl/src/widgets/lv_table.c', '../../lib/lv_bindings/lvgl/src/widgets/lv_objx_templ.c', '../../lib/lv_bindings/lvgl/src/widgets/lv_switch.c', '../../lib/lv_bindings/lvgl/src/widgets/lv_line.c', '../../lib/lv_bindings/lvgl/src/widgets/lv_img.c', '../../lib/lv_bindings/lvgl/src/widgets/lv_btn.c', '../../lib/lv_bindings/lvgl/src/widgets/lv_bar.c', '../../lib/lv_bindings/lvgl/src/hal/lv_hal_disp.c', '../../lib/lv_bindings/lvgl/src/hal/lv_hal_tick.c', '../../lib/lv_bindings/lvgl/src/hal/lv_hal_indev.c', '../../lib/lv_bindings/lvgl/src/misc/lv_ll.c', '../../lib/lv_bindings/lvgl/src/misc/lv_lru.c', '../../lib/lv_bindings/lvgl/src/misc/lv_txt_ap.c', '../../lib/lv_bindings/lvgl/src/misc/lv_mem.c', '../../lib/lv_bindings/lvgl/src/misc/lv_style_gen.c', '../../lib/lv_bindings/lvgl/src/misc/lv_style.c', '../../lib/lv_bindings/lvgl/src/misc/lv_color.c', '../../lib/lv_bindings/lvgl/src/misc/lv_utils.c', '../../lib/lv_bindings/lvgl/src/misc/lv_bidi.c', '../../lib/lv_bindings/lvgl/src/misc/lv_anim_timeline.c', '../../lib/lv_bindings/lvgl/src/misc/lv_txt.c', '../../lib/lv_bindings/lvgl/src/misc/lv_gc.c', '../../lib/lv_bindings/lvgl/src/misc/lv_area.c', '../../lib/lv_bindings/lvgl/src/misc/lv_printf.c', '../../lib/lv_bindings/lvgl/src/misc/lv_log.c', '../../lib/lv_bindings/lvgl/src/misc/lv_tlsf.c', '../../lib/lv_bindings/lvgl/src/misc/lv_templ.c', '../../lib/lv_bindings/lvgl/src/misc/lv_async.c', '../../lib/lv_bindings/lvgl/src/misc/lv_fs.c', '../../lib/lv_bindings/lvgl/src/misc/lv_anim.c', '../../lib/lv_bindings/lvgl/src/misc/lv_timer.c', '../../lib/lv_bindings/lvgl/src/misc/lv_math.c', '../../lib/lv_bindings/lvgl/src/font/lv_font_montserrat_10.c', '../../lib/lv_bindings/lvgl/src/font/lv_font_montserrat_12_subpx.c', '../../lib/lv_bindings/lvgl/src/font/lv_font_montserrat_8.c', '../../lib/lv_bindings/lvgl/src/font/lv_font_dejavu_16_persian_hebrew.c', '../../lib/lv_bindings/lvgl/src/font/lv_font_montserrat_14.c', '../../lib/lv_bindings/lvgl/src/font/lv_font_simsun_16_cjk.c', '../../lib/lv_bindings/lvgl/src/font/lv_font_montserrat_44.c', '../../lib/lv_bindings/lvgl/src/font/lv_font_montserrat_16.c', '../../lib/lv_bindings/lvgl/src/font/lv_font_montserrat_26.c', '../../lib/lv_bindings/lvgl/src/font/lv_font_unscii_16.c', '../../lib/lv_bindings/lvgl/src/font/lv_font_montserrat_34.c', '../../lib/lv_bindings/lvgl/src/font/lv_font_loader.c', '../../lib/lv_bindi [... cut ...] irq.c', '../../shared/runtime/pyexec.c', '../../shared/runtime/stdout_helpers.c', '../../shared/runtime/sys_stdio_mphal.c', '../../shared/timeutils/timeutils.c', '../../extmod/modonewire.c', '../../extmod/modnetwork.c', '../../extmod/modusocket.c']' returned non-zero exit status 1
../../py/mkrules.mk:104: recipe for target 'build-STM32F769DISC/genhdr/qstr.i.last' failed
make: *** [build-STM32F769DISC/genhdr/qstr.i.last] Error 1
make: *** Deleting file 'build-STM32F769DISC/genhdr/qstr.i.last'
```

dpgeorge, 2022-05-25:
You obviously have lvgl as part of the build, which is non-standard and not part of this repository.  If you check out this repository and build, it should work.

r00t1900, 2022-05-25:
> You obviously have lvgl as part of the build, which is non-standard and not part of this repository. If you check out this repository and build, it should work.

oh, I found it. it's the error on `lv_bindings` so yes you are right: it is the problem involved with LVGL. I am going to ask on `lv_micropython` repo. Besides, do you have any idea about the error?

mattytrentini, 2023-10-24:
Closing, as not related to mainline MicroPython (but please do report back if you had any clarifications!)

closed 2023-10-24 by mattytrentini