micropython/micropython #8705: ESP32 compilation problem----OSError: [Errno 8] Exec format error:

smf408fze, 2022-05-28:
[ 83%] Linking C static library liblibsodium.a
make[3]: Leaving directory '/home/book/ESP32/micropython/ports/esp32/build-GENERIC_S2'
[ 83%] Built target __idf_libsodium
[ 83%] Generating ../../genhdr/qstr.split
[ 83%] Generating ../../genhdr/qstrdefs.collected.h
QSTR updated
[ 83%] Generating ../../genhdr/qstrdefs.preprocessed.h
[ 83%] Generating ../../genhdr/qstrdefs.generated.h
[ 83%] Generating ../../frozen_content.c
MPY flashbdev.py
Traceback (most recent call last):
  File "/home/book/ESP32/micropython/tools/makemanifest.py", line 435, in <module>
    main()
  File "/home/book/ESP32/micropython/tools/makemanifest.py", line 370, in main
    res, out = system(
  File "/home/book/ESP32/micropython/tools/makemanifest.py", line 160, in system
    output = subprocess.check_output(cmd, stderr=subprocess.STDOUT)
  File "/usr/lib/python3.8/subprocess.py", line 411, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1702, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
OSError: [Errno 8] Exec format error: '/home/book/ESP32/micropython/mpy-cross/mpy-cross'
esp-idf/main/CMakeFiles/__idf_main.dir/build.make:84: recipe for target 'frozen_content.c' failed
make[3]: *** [frozen_content.c] Error 1
make[3]: Leaving directory '/home/book/ESP32/micropython/ports/esp32/build-GENERIC_S2'
CMakeFiles/Makefile2:5009: recipe for target 'esp-idf/main/CMakeFiles/__idf_main.dir/all' failed
make[2]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/all] Error 2
make[2]: Leaving directory '/home/book/ESP32/micropython/ports/esp32/build-GENERIC_S2'
Makefile:135: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/book/ESP32/micropython/ports/esp32/build-GENERIC_S2'
make failed with exit code 2
Makefile:34: recipe for target 'all' failed
make: *** [all] Error 2
What kind of question is that?How to solve

jimmo, 2022-05-28:
@smf408fze have you compiled mpy-cross?
See https://github.com/micropython/micropython/tree/master/ports/esp32#building-the-firmware

TomzBench, 2024-12-02:
I have the same error. Running `make` on `debian bookworm` seems to use the `arm-linux-gnueabi-gcc` compiler for building `mpy-cross` Which makes a bad binary and gives the `Exec format error` when you try and execute it.