micropython/micropython #8706: Flash encryption on esp32

MauroDiamantino, 2022-05-28:
Hello, have you implemented a way to allow a developer to compile micropython with flash encryption enabled? is it possible to modify sdkconfig to enable flash encryption? which is the best way to do that?

AmirHmZz, 2022-07-31:
You should use [project configuration menu](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#project-configuration-menu) to enable flash encryption : 
```shell
cd micropython/ports/esp32
idf.py menuconfig
```

MauroDiamantino, 2022-08-02:
Hi, what I've done is to compile micropython with the necessary frozen code and libraries (like ulab), then encrypt the fimware, enable the necessary efuses in the esp32 and finally burn the firmware in the board. So, I didn't need to modify the project configuration.

AmirHmZz, 2022-08-02:
Hey @MauroDiamantino. 

1. Can you help me in doing the same thing you've done step by step? I know how to build micropython with my frozen code and modules. I just want to know how to encrypt framework and which efuses need to be burned and how to flash encrypted framework into esp32?

2. Have you ever tried to flash another encrypted framework after flashing another one? I mean have you ever updated the encrypted framework?

3. Do you use secure boot too?

MauroDiamantino, 2022-08-03:
Hi @AmirHmZz ,

1. To encrypt the firmware you can use the tool `espsecure.py`. The efuses with their corresponding valuesa are the following:
FLASH_CRYPT_CONFIG 0xF
FLASH_CRYPT_CNT 0X1
DISABLE_DL_ENCRYPT 1
DISABLE_DL_DECRYPT 1
DISABLE_DL_CACHE 1
2. Yes, you can re-flashed the encrypted firmware this way.
3. I haven't yet used secured boot.
All the magic is in this [page](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html).

MauroDiamantino, 2023-10-12:
Hi, I have some doubts about this. Now I'm manually setting the following efuses in the ESP32 using `espsecure.py`:
FLASH_CRYPT_CONFIG 0xF
FLASH_CRYPT_CNT 0X1
DISABLE_DL_ENCRYPT 1
DISABLE_DL_DECRYPT 1
DISABLE_DL_CACHE 1
JTAG_DISABLE 1
And then I build the micropython firmware with my scripts as frozen codes and encrypt it with a pregenerated key using `espsecure.py`. After that, I write the encrypted firmware in the ESP32 using `esptool.py`.
I'm not using the reflashing option (see [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html#re-flashing-updated-partitions)) of the Flash encryption Development mode, bacause the efuse FLASH_CRYPT_CNT has just 7 bits so the board could be reflashed just 4 times, is it correct?
But I have realized that I'm encrypting all partitions nvs, phy-init, factory and vfs, when the Flash encryption docs state that only bootload, partition table, NVS Key Partition, Otadata, “app” type partitions (factory) and partitions marked with `encrypted` flag should be encrypted. So, nvs, phy-init, and vfs partitions shouldn't be encrypted, and even nvs partition doesn't support `encrypted` flag. Could this cause some troubles? Could it affect Wifi taking into account the calibration data is stored in the nvs partition?
With this configuration, could someone still stole and decrypt the firmware?

amplatzer, 2024-02-21:
> Hi, I have some doubts about this. Now I'm manually setting the following efuses in the ESP32 using `espsecure.py`: FLASH_CRYPT_CONFIG 0xF FLASH_CRYPT_CNT 0X1 DISABLE_DL_ENCRYPT 1 DISABLE_DL_DECRYPT [...]

Hi, have you figured out how to implement the flash encryption?
I have been trying this for a long time, but it doesn't work.
I don't think you have to burn all those fuses, as it described in the  "Flash Encryption Process" section.  once the proper firmware has been flashed, it will burn all necessary fuses, am I right?

I have successfully done the flash encryption with the hello-world example, but when I try the same process on micropython, the WIFI stops working with error 1101.

amplatzer, 2024-02-22:
> Hi, I have some doubts about this. Now I'm manually setting the following efuses in the ESP32 using `espsecure.py`: FLASH_CRYPT_CONFIG 0xF FLASH_CRYPT_CNT 0X1 DISABLE_DL_ENCRYPT 1 DISABLE_DL_DECRYPT [...]

Hi, [Mauro](https://github.com/MauroDiamantino).

> And then I build the micropython firmware with my scripts as frozen codes and encrypt it with a pregenerated key using espsecure.py. After that, I write the encrypted firmware in the ESP32 using esptool.py.
 
May I know which bin file you encrypted manually ? is it the firmware.bin ? do you need to encrypt the bootloader.bin and partition-table.bin and flash them ?

MauroDiamantino, 2024-02-22:
Hi @amplatzer, the file firmware.bin is the one you have encrypt (you can use `espsecure.py encrypt_flash_data` for that), but this is only needed if you want to reflash the ESP32, once the board has the flash encryption enabled (as long as the reflashing is enabled).

> Hi, have you figured out how to implement the flash encryption? I have been trying this for a long time, but it doesn't work. I don't think you have to burn all those fuses, as it described in the [...]

As you said there, it isn't needed to flash the efuses manually. You have to enable flash encryption before you compile micropython, modifying file micropython/ports/esp32/boards/sdkconfig.base (it is important to reduce log level to None, to avoid memory problems). If you are going to use frozen modules, it is important to add `include('.../micropython/ports/esp32/boards/manifest.py')` in the manifest file. Then compile, burn the firmware in the board and the flash encryption process will happen automatically.

amplatzer, 2024-02-23:
> Hi @amplatzer, the file firmware.bin is the one you have encrypt (you can use `espsecure.py encrypt_flash_data` for that), but this is only needed if you want to reflash the ESP32, once the board has the flash encryption enabled (as long as the reflashing is enabled).
> 
> > > Hi, I have some doubts about this. Now I'm manually setting the following efuses in the ESP32 using `espsecure.py`: FLASH_CRYPT_CONFIG 0xF FLASH_CRYPT_CNT 0X1 DISABLE_DL_ENCRYPT 1 DISABLE_DL_DECRYPT 1 DISABLE_DL_CACHE 1 JTAG_DISABLE 1 And then I build the micropython firmware with my scripts as frozen codes and encrypt it with a pregenerated key using `espsecure.py`. After that, I write the encrypted firmware in the ESP32 using `esptool.py`. I'm not using the reflashing option (see [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html#re-flashing-updated-partitions)) of the Flash encryption Development mode, bacause the efuse FLASH_CRYPT_CNT has just 7 bits so the board could be reflashed just 4 times, is it correct? But I have realized that I'm encrypting all partitions nvs, phy-init, factory and vfs, when the Flash encryption docs state that only bootload, partition table, NVS Key Partition, Otadata, “app” type partitions (factory) and partitions marked with `encrypted` flag should be encrypted. So, nvs, phy-init, and vfs partitions shouldn't be encrypted, and even nvs partition doesn't support `encrypted` flag. Could this cause some troubles? Could it affect Wifi taking into account the calibration data is stored in the nvs partition? With this configuration, could someone still stole and decrypt the firmware?
> > 
> > 
> > Hi, have you figured out how to implement the flash encryption? I have been trying this for a long time, but it doesn't work. I don't think you have to burn all those fuses, as it described in the "Flash Encryption Process" section. once the proper firmware has been flashed, it will burn all necessary fuses, am I right?
> > I have successfully done the flash encryption with the hello-world example, but when I try the same process on micropython, the WIFI stops working with error 1101.
> 
> As you said there, it isn't needed to flash the efuses manually (except for the flash encryption key if you want to use always the same, you have to burn it before the flash encryption process). You have to enable flash encryption before you compile micropython, modifying file ".../micropython/ports/esp32/boards/sdkconfig.base" (it is important to reduce log level to None, to avoid memory problems). If you are going to use frozen modules, it is important to add `include('.../micropython/ports/esp32/boards/manifest.py')` in the manifest file. Then compile, burn the firmware in the board and the flash encryption process will happen automatically. I'm using flash encryption with micropython without any troubles, so if you're facing some errors maybe you're making some mistakes. The wifi error maybe is related to a memory problem, however the compilation stops if you have that problem.

Hi @MauroDiamantino , thanks for the quick response. appreciate that. 
I still have some questions regarding to Flash Encryption process. 
1.  How did you compile micropython,  use Make command or idf.py?   let's say compile for board GENERIC_S3. 
2. If I don't use a pre-generated encrypt key, then the chip will use a random key, my question is , how the re-falshing works with a random key?  it doesn't know the key, how did it encrypt the updated firmware and re-flash it to the board?I thought once the encryption on, you can only re-flash it with the firmware that been encrypted with the same key, am I right? Otherwise you need to burn the efuse to turn if off( DEV mode of cause) then flash an un-encrypted firmware. 
3. include('.../micropython/ports/esp32/boards/manifest.py') in the manifest file. ,  which manifest file are you talking about, may I have the full path?
4.  I have changed the settings (like loglevel, partition offset, enable flash encrypt) in the menuconfig, do I still need to modify micropython/ports/esp32/boards/sdkconfig.base ?

breno174, 2024-02-29:
Hi @amplatzer , can you help me with a question? How are you managing to put your MicroPython code on the ESP32 and being able to modify it after encrypting?

amplatzer, 2024-03-02:
> Hi @amplatzer , can you help me with a question? How are you managing to put your MicroPython code on the ESP32 and being able to modify it after encrypting?

Hi I just compile them to .mpy file and replace the existing modules.

XBCSmart, 2025-03-20:
step1:  use mpy-cross to complie an  py file to mpy file
step2:  encypt mpy  then get a bin file
step3:   translate bin as c file which contains a bytearray
step4:   decrypt bytearray to bytearray
step5:  load and run this byte array

but i haven't  finished step5  because micropython hasn't  this feature now , 
i have tried many way  but failed,   You can try following this train of thought and i feel like success is already within reach

XBCSmart, 2025-03-20:
> > Hi [@amplatzer](https://github.com/amplatzer) , can you help me with a question? How are you managing to put your MicroPython code on the ESP32 and being able to modify it after encrypting?
> 
> Hi I just compile them to .mpy file and replace the existing modules.

step1: use mpy-cross to complie an py file to mpy file
step2: encypt mpy then get a bin file
step3: translate bin as c file which contains a bytearray
step4: decrypt bytearray to bytearray
step5: load and run this byte array

but i haven't finished step5 because micropython hasn't this feature now ,
i have tried many way but failed, You can try following this train of thought and i feel like success is already within reach