micropython/micropython #8707: ESP32-S3 USB HID

Red-M, 2022-05-28:
I'd love to see device USB HID support for the ESP32-S3, currently this is not implemented.

I am currently trying to implement this but it appears to be held up by espressif as their [library does not support it](https://github.com/espressif/esp-idf/blob/master/components/esp_hid/src/esp_hidd.c#L37-L50),, any assistance in implementing this would be greatly appreciated as this is my first piece of work dealing with microcontrollers directly.

dhalbert, 2022-05-28:
I just tried CircuitPython's USB HID support for keyboards on an S3, and it works, so I don't think there's an inherent problem.

Red-M, 2022-05-29:
The issue is that esp's documentation says to not mix raw tinyusb and the idf's tinyusb because they can have bad interactions.

stephanelsmith, 2022-05-30:
Fascinated by this prospect.  two bumps.

stroobandt, 2023-04-13:
> I just tried CircuitPython's USB HID support for keyboards on an S3, and it works, so I don't think there's an inherent problem.

@dhalbert Could you point me to some CircuitPython USB HID example code?
I am interested in getting USB HID to run on a Lilygo T7 S3 for an upcoming project. Thanks!

dhalbert, 2023-04-13:
@stroobandt Rather than discuss this here, could you post in our discord at https://adafru.it/discord in the `#help-with-circuitpython` channel, or in our forums, https://forums.adafruit.com/viewforum.php?f=60 ?

jonnor, 2025-10-03:
There is now support for USB devices dynamically for ESP32-S3 #15564 
There are examples for HID in micropython-lib

closed 2025-10-03 by jonnor