micropython/micropython #8717: There is still a cancellation/race issue in wait_for

Kalki70, 2022-05-31:
I know this behavior has been reported before, but I am still seeing it. Tracking a strange condition in our software, where cancelling a task soon after creating it did not work, we found the issue in wait_for and some other reported issues, the last one being #7386 

I also found this reported in Python, and a test program mentioned there causes also the issue on micropython : https://bugs.python.org/issue42130

Basically, this code :

```python
import uasyncio as asyncio

async def inner():
    return

async def with_for_coro():
    await asyncio.wait_for(inner(), timeout=100)
    await asyncio.sleep(1)
    print('End of with_for_coro. Should not be reached!')

async def main():
    task = asyncio.create_task(with_for_coro())
    await asyncio.sleep(0)
    assert not task.done()
    task.cancel()
    print('Called task.cancel()')
    await task  # -> You would expect a CancelledError to be raised.

asyncio.run(main())
```

Running this, no CancelledError is raised when awaiting task in main() and we get this output:

```
$ micropython cancel.py 
Called task.cancel()
End of with_for_coro. Should not be reached!
```

Now, if we add a small delay before cancelling, changing for instance to 0.1s:

```python
async def main():
    task = asyncio.create_task(with_for_coro())
    await asyncio.sleep(0.1)
```

Then we get the expect result :

```
$ micropython cancel.py 
Called task.cancel()
Traceback (most recent call last):
  File "cancel.py", line 21, in <module>
  File "/usr/lib/micropython/uasyncio/core.py", line 222, in run
  File "/usr/lib/micropython/uasyncio/core.py", line 191, in run_until_complete
  File "/usr/lib/micropython/uasyncio/core.py", line 176, in run_until_complete
  File "cancel.py", line 18, in main
  File "/usr/lib/micropython/uasyncio/task.py", line 145, in __next__
  File "/usr/lib/micropython/uasyncio/core.py", line 183, in run_until_complete
  File "cancel.py", line 9, in with_for_coro
CancelledError: 
```

This was tested using micropython 1.18

dpgeorge, 2022-05-31:
Thanks for the report.  I have a vague memory of seeing the issue myself, and also seeing it on CPython, so I ignored it because the MicroPython implementation at least matched CPython :joy: 

I think the problem is that the `wait_for` is being cancelled right at the moment that the thing it's waiting on is finishing.

dpgeorge, 2022-06-01:
Thanks for the patch.  Unfortunately your fix is not going to work well for MicroPython, because it adds a new member to the Task instance, which will add a lot of memory use for existing code.  We need to find another way to fix it.

But note that CPython has not yet merged a fix for this issue.  So it might be worth waiting to see what they do, then make sure we match their semantics (if possible).

dpgeorge, 2022-06-01:
I think I found a way to fix it, see #8719.

smurfix, 2022-06-17:
The patch from @dpgeorge has been merged. Can this issue be closed?

dpgeorge, 2022-06-17:
Yes this is completed.

closed 2022-06-17 by dpgeorge

Kalki70, 2022-06-17:
Yes, thanks.