micropython/micropython #8720: Transfer files to the NUCLEO_F091RC do not works

beyonlo, 2022-06-01:
Hi all

I'm using this board https://micropython.org/download/NUCLEO_F091RC/ but is impossible to copy files to the flash.

The mpremote works to connect, but not to copy files.
```
$ mpremote 
Connected to MicroPython at /dev/ttyACM0
Use Ctrl-] to exit this shell
>>
>>> import os
>>> os.uname()
(sysname='pyboard', nodename='pyboard', release='1.18.0', version='v1.18 on 2022-01-17', machine='NUCLEO-F091RC with STM32F091RCT6')
>>>

```
**Copy files With mpremote:**
```
$ mpremote cp gc1.py :
cp gc1.py :gc1.py
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV

```
**Copy files with ampy:**
```
$ ampy -p /dev/ttyACM0 -b 115200 put gc1.py 
Traceback (most recent call last):
  File "/usr/local/bin/ampy", line 8, in <module>
    sys.exit(cli())
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.8/dist-packages/ampy/cli.py", line 264, in put
    board_files.put(remote, infile.read())
  File "/usr/local/lib/python3.8/dist-packages/ampy/files.py", line 214, in put
    self._pyboard.exec_("f = open('{0}', 'wb')".format(filename))
  File "/usr/local/lib/python3.8/dist-packages/ampy/pyboard.py", line 279, in exec_
    raise PyboardError('exception', ret, ret_err)
ampy.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 1, in <module>\r\nOSError: [Errno 19] ENODEV\r\n')
```

**Ps:** Copy files with `mpremote` and `ampy` works fine in other boards, like as `pyboard`, just not works on this `NUCLEO F0`.

Anyone know what is the problem OR if is there another way to copy files do flash?

I see a similar problem in this PR #8713. Is it the same problem?

Thank you.

beyonlo, 2022-06-01:
I'm closing this issue because is not a issue. This is just because Micropython for the F0 do not has internal filesystem enable.  
Need to use a spi flash or implement internal flash driver.

closed 2022-06-01 by beyonlo

chrismas9, 2022-06-13:
@beyonlo See https://github.com/micropython/micropython/pull/8756 for a solution.
You need to create the filesystem after flashing the board:
`uos.VfsLfs1.mkfs(pyb.Flash(start=0))`

beyonlo, 2022-06-13:
Hello @chrismas9 that is great!

>See #8756 for a solution.

Is there a ready binary for the tests? If not, I need just to clone the official MicroPython master, configure disabling that features and build my firmware? Or I need to clone from your PR https://github.com/chrismas9/micropython/tree/F091_LFS1 ?

> You need to create the filesystem after flashing the board: `uos.VfsLfs1.mkfs(pyb.Flash(start=0))`

Did you added this feature to create the filesystem, or that already exists and you just disabled that modules and build a new firmware?

Sorry for many questions! I'm just learning more about that :)

Thank you for the clarifications.

chrismas9, 2022-06-13:
Hi @beyonlo For you convenience I have uploaded programming files (elf and dfu, depending on how you program) and my test scripts:

`Lfs1-mk.py` will create a filesystem and an empty main.py.
`Lfs1-rd.py` will test the filesystem.
These need to be executed from RAM, eg with rshell or Thonny. Then you can copy `main.py` to the board and it will flash the green LED.

[NUCLEO_F091RC.zip](https://github.com/micropython/micropython/files/8891213/NUCLEO_F091RC.zip)

beyonlo, 2022-06-13:
Hi @chrismas9 

Thank you very much!

That PR #8756 will be great to be merged in the master, so in the next versions of Micropython for the F0 can have a filesystem, even it is little (12k or less)!

rolandvs, 2022-06-13:
Super! I use the f091 a lot and thus makes it even more usable!

dpgeorge, 2022-06-17:
Filesystem on F091 was enabled in 868f23b362171ee131bd6bfa784911fb8fc66202