micropython/micropython #8723: Teensy 4.1 SDIO support

thezim, 2022-06-04:
Does MicorPython for the Teensy 4.1 support SDIO mode for the SD card?

robert-hh, 2022-06-04:
The driver supports the SD card as storage medium using the uSDHC host controller with the respective SDIO API calls. The data width is 4 bit.

thezim, 2022-06-04:
This is enabled specifically for the teensy 4.1? I ask because I’m seeing typical SPI speeds  with Sdcard access. Much slower write speed than the sdfat lib in Arduino. Microseconds in Arduino and milliseconds in MicroPython.

robert-hh, 2022-06-05:
SD-Card support is the same for all MIMXRT port, which supports it. My test gave numbers of < 1MB/s - 10 MB/s write speed for bulk transfers, depending on the block size and card type. Best speed is at block sizes with multiples of 4096 bytes. If you have small writes with non-block-aligned data, writing gets slow.