micropython/micropython #8727: Building ESP32 firmware failure

robhamerling, 2022-06-04:
Build of ESP32 modules fails (since 1.18-553 ?).
The console output shows:
'
../network_ppp.c:199:9: error: implicit declaration of function 'pppapi_set_auth'; did you mean 'ppp_set_auth'? [-Werror=implicit-function-declaration]
         pppapi_set_auth(self->pcb, parsed_args[ARG_authmode].u_int, username_str, password_str);
         ^~~~~~~~~~~~~~~
         ppp_set_auth
'

After the suggested change (pppapi_set_auth -> ppp_set_auth) compilation succeeds and firmware runs OK (checked with 1.18-567 on ESP32-S2_WROVER)

iot7755, 2022-06-04:
Build for board ESP32 GENERIC fails after latest pull:

../network_ppp.c: In function 'ppp_connect_py':
../network_ppp.c:199:9: error: implicit declaration of function 'pppapi_set_auth'; did you mean 'ppp_set_auth'? [-Werror=implicit-function-declaration]
         pppapi_set_auth(self->pcb, parsed_args[ARG_authmode].u_int, username_str, password_str);
         ^~~~~~~~~~~~~~~
         ppp_set_auth
cc1: some warnings being treated as errors

Same problem after latest pull

and 

make -C mpy-cross/
make BOARD=GENERIC clean
make BOARD=GENERIC -j4

LordGuilly, 2022-06-06:
it could be the esp-idf version, I am building ok with  MP latest version and idf v4.4 (have other problems around PPP though !)

dpgeorge, 2022-06-07:
Nothing has changed in MicroPython in PPP for a while, so it must be a recent IDF change.

LordGuilly, 2022-06-07:
in line 43, do you have the include of 

#include "netif/ppp/pppapi.h"

I can see that file in the IDF,that should fix the error (also worth checking the file is present, or if the tree is clean?)

robhamerling, 2022-06-07:
Yes, I do have line 43 with #include "netif/ppp/pppapi.h"
I expect a compiler msg when an include file cannot be found. 
I'm afraid I do not understand the meaning of "or if the tree is clean"...

LordGuilly, 2022-06-07:
the error you are getting is related to a missing function declaration. Functions are usually declared in .h files,
The MP project seems to rely a lot in the ESP-IDF code, so if you have that include, then you need to go to the IDF directory and check what does that file say

in mine, I can see the following:

```
#if ESP_PPP && PPP_AUTH_SUPPORT
void pppapi_set_auth(ppp_pcb *pcb, u8_t authtype, const char *user, const char *passwd);
#endif
#if PPP_NOTIFY_PHASE
err_t pppapi_set_notify_phase_callback(ppp_pcb *pcb, ppp_notify_phase_cb_fn notify_phase_cb);
#endif /* PPP_NOTIFY_PHASE */
#if PPPOS_SUPPORT
ppp_pcb *pppapi_pppos_create(struct netif *pppif, pppos_output_cb_fn output_cb, ppp_link_status_cb_fn link_status_cb, void *ctx_cb);
#endif /* PPPOS_SUPPORT */
```

the prototype is under two config defines, maybe you have one of them missing in your build config?

Regarding the "clean tree", if you start making changes in a source tree, sometimes the version control system might mess up the updates - as it tries to keep or merge the changes. 
Usually a "clean tree" would mean that there are no local changes, just whatever you got from the original repo. It's not a rare problem, specially if you try changes or things and then forget to revert

LordGuilly, 2022-06-07:
sorry, I missed that part, I thought you were doing the build yourself and I was trying to help :-)

robhamerling, 2022-06-07:
No excuses needed! Your involvement is probably helpful for the pre-built maintainer. 
Indeed I do builds myself for boards with which I want the st7789_mpy module included. For most boards I rely on the pre-built firmware.

jimmo, 2022-06-08:
> Thanks for the explanation. But may I remind you: this issue is about a problem with the pre-built firmware.

@robhamerling Sorry I don't follow what you're asking for here, you're posting a compile error but also saying the issue is with pre-built firmware? (How does using pre-built firmware give you a compile error?)

dpgeorge, 2022-06-08:
The problem is due to this commit: 15fea3a1ff9b7437e0fed54bd4a16c1708faf321

dpgeorge, 2022-06-08:
In particular these lines, which change the lwip include dir unconditionally if `lib/lwip/` is checked out:
```
   list(APPEND MICROPY_INC_CORE
       ${MICROPY_LIB_LWIP_DIR}/include
   )
```

dpgeorge, 2022-06-08:
Fixed by bd375df02cfacff82b8c74f4adabeb0cade9a496

closed 2022-06-08 by dpgeorge