micropython/micropython #8728: ESP32 and PPP not working

LordGuilly, 2022-06-05:
I am trying to raise a PPP connection with a modem (two different ones actually, an EG91 and a BG95)
Once the modem is registered into the network, I start the data call with AT commands, and after the "CONNECT" string is received, do the following code in MicroPython:
```
my_log.info("Starting Data Call", fn="main")
modem.startDataCall()
my_log.info("Creating PPP object", fn="main")
sleep(5)
ppp = network.PPP(modem.getSerialPort())
ppp.active(True)
ppp.connect(authmode=ppp.AUTH_CHAP,username ="", password="")
```

I am monitoring the PPP frames in the serial line, and I can see the following:
- the ESP32 sends the first ConfRequest
- the moddem replies with the ACK and its own ConfRequest
- but the ESP32 does not reply anything for about 6 seconds, and in the meantime the modem sends new ConfRequests
- after 6 seconds, the ESP32 seems to process _all_ the requests and ACK them, but this causes abnormal transitions from the PPP FSM. 
- The modem replies the ConfRequest, but again nothing happens for another 6 seconds.
-  The modem times out every second, and sends a new ConfRequest that causes the same problem again and again, until one or both of the parts give up

I enabled LWIP and PPP logs. It seems to me the problem is the UART object is not passing the data received in the serial port to the PPP object straight away, but only after a timeout. I played a bit with some of the hardcoded values (like FSMTIMEOUT), that seems to confirm the guess.
Unfortunately I am unable to find what mechanism will pass the data from the UART object ot the PPP object - all I can see is there is a task created that waits to be notified, but cannot find what's sending such notification.

I am using latest micropython version (master branch, commit 2111ca0b8fdcbef6177a8d37fde53085de2a798a) , and building with ESP-IDF 4.4

any help would be appreciated, I can supply the logs (or enable more segments and capture more if it helps)

LordGuilly, 2022-06-07:
these are the log files, the PCAP does not have timestamps, just the exchange sequence. 

[ppp_debug.zip](https://github.com/micropython/micropython/files/8852290/ppp_debug.zip)

Tangerino, 2022-07-13:
Hi @LordGuilly could you guide me on how to enable LWIP and PPP logs so I can do the tests here as well?

LordGuilly, 2022-07-19:
this is the diff output, I haven't tested the new FW yet, to see if it's fixed - this project went down in the priority list!

you need to enable the PPP debug segments and the LWIP too

diff --git a/ports/esp32/boards/sdkconfig.base b/ports/esp32/boards/sdkconfig.base
index 4b2901343..261670a16 100644
--- a/ports/esp32/boards/sdkconfig.base
+++ b/ports/esp32/boards/sdkconfig.base
@@ -18,8 +18,11 @@ CONFIG_BOOTLOADER_LOG_LEVEL_WARN=y
 
 # Change default log level to "ERROR" (instead of "INFO")
 CONFIG_LOG_DEFAULT_LEVEL_INFO=n
-CONFIG_LOG_DEFAULT_LEVEL_ERROR=y
-CONFIG_LOG_DEFAULT_LEVEL=1
+CONFIG_LOG_DEFAULT_LEVEL_VERBOSE=y
+CONFIG_LOG_DEFAULT_LEVEL_DEBUG=m
+CONFIG_LOG_DEFAULT_LEVEL_ERROR=n
+#CONFIG_LOG_DEFAULT_LEVEL=5
+#CONFIG_LOG_MAXIMUM_LEVEL=5
 
 # ESP32-specific
 CONFIG_ESP_TASK_WDT_CHECK_IDLE_TASK_CPU0=n
@@ -39,9 +42,12 @@ CONFIG_FREERTOS_SUPPORT_STATIC_ALLOCATION=y
 CONFIG_FREERTOS_ENABLE_STATIC_TASK_CLEAN_UP=y
 
 # UDP
+CONFIG_LWIP_NETIF_DEBUG=y
 CONFIG_LWIP_PPP_SUPPORT=y
 CONFIG_LWIP_PPP_PAP_SUPPORT=y
 CONFIG_LWIP_PPP_CHAP_SUPPORT=y
+CONFIG_LWIP_PPP_DEBUG_ON=y
+CONFIG_LWIP_DEBUG=y
 
 # SSL
 # Use 4kiB output buffer instead of default 16kiB (because IDF heap is fragmented in 4.0)
~
~

victorallume, 2022-11-16:
Did you get anywhere with this? Did you write the code for the `modem` object, or did this come from a library somwhere?

ThinkTransit, 2023-04-09:
I have a similar issue, PPP connects but it is very slow, maybe this is caused by the timeout.

Has anyone made any progress on getting this working?

Tangerino, 2023-04-13:
I never make it work.
Anyone in Europe that is willing to play with my board?
I can ship it to you