micropython/micropython #8729: socket.SO_BROADCAST

JarynCode, 2022-06-06:
Use socket.SO_BROADCAST to WOL, but get "AttributeError: 'module' object has no attribute 'SO_BROADCAST'".how to fix it?

dpgeorge, 2022-06-07:
Which port/hardware are you using?

You can try using the value 0x20 instead (which is what lwIP defines it to be).

msetina, 2023-05-04:
I'm using RP2 and can confirm that socket.SO_BROADCAST is missing in 1.20

robert-hh, 2023-05-04:
Which RP2 board do you use? Pico W or Arduino Nano RP2040 Connect?

msetina, 2023-05-04:
Raspberry Pi Pico W

msetina, 2023-05-04:
Maybe there is a reason in implementation, as after using 0x20 send_to returns OSError: [Errno 113] EHOSTUNREACH
Will be checking sam code on ESP32 later

msetina, 2023-05-04:
ESP32 generic image warns:
Warning: lwip.setsockopt() option not implemented
so...
and sendto returns OSError: [Errno 113] EHOSTUNREACH
and also can not find definition of SO_BROADCAST

xuancong84, 2023-06-17:
I also encountered this bug for ESP8266. It seems that some previous version micropython supports SO_BROADCAST. Not sure why developers have removed it.

xuancong84, 2023-06-19:
Issue solved (see pull request at https://github.com/micropython/micropython/pull/11815#issue-1762602891 )