micropython/micropython #8732: STM32L4: Timer callbacks broken after stmlib L4-1.17.0

chrismas9, 2022-06-07:
Timer callbacks seem to be broken for STM32L4 series after stmlib was updated to L4-1.17.0. Tested on custom L496 board and NUCLEO-L452RE. Works on F401, F411 and F405. Not tested on L0, F0, F7, H7, G0, G4 or WB.

On NUCLEO-L452RE the attached test script works on V1.17-147-gd72d699da but fails on V1.17-240-gbc1b0fd2c. I can't see any relevant changes to MicroPython code between these, except for the library update. Also fails with the latest nightly build of NUCLEO-L452RE, so not dependant on build environment.

The issue can be duplicated by running the attached test script on any board with PB15 (output) jumpered to PA1 (input). Rising edges on PA1 should trigger a calback, flash LED1 and print the capture time once per second.

[callback.zip](https://github.com/micropython/micropython/files/8849938/callback.zip)

yn386, 2022-09-01:
I tried to be duplicated by running the attached script with NUCLEO-F446RE.

I could not be duplicated with MicroPython v1.19.1-358-g0b26efe73.
This version uses STM32CubeF4 version 1.16.0.
But using STM32CubeF4 version v1.27.1, this issue could be duplicated on NUCLEO-F446RE.

I found some changes about timer between v1.16 and v1.27.1 and the implementation of timer.c should change for v1.27.1.
So I'm trying to implement it. 
STM32CubeL4 also includes these changes, so this implementation may also fix this issue on STM32L4.

For workaround, specify callback when creating gps_PPS_IC like following change can prevent this issue.

```
--- callback.py.1	2022-09-01 17:22:03.976472646 +0900
+++ callback_f4.py	2022-09-01 17:20:48.323016890 +0900
@@ -55,8 +55,7 @@
 
     timer_PPS = Timer(2, prescaler=0, freq = 1)
     fake_PPS_out_PWM = timer_PPS.channel(1, Timer.PWM, pin=fake_PPS_out, pulse_width_percent=10) # PA0
-    gps_PPS_IC = timer_PPS.channel(2, Timer.IC, pin=gps_PPS_in, polarity=Timer.RISING) # PA1
-    gps_PPS_IC.callback(gps_callback)
+    gps_PPS_IC = timer_PPS.channel(2, Timer.IC, callback=gps_callback, pin=gps_PPS_in, polarity=Timer.RISING) # PA1
     print('Waiting for callbacks')
 
     while 1:
```

dpgeorge, 2022-09-06:
Fixed by 989b8c728b296969e6489908b8e94ed23625a1e8

closed 2022-09-06 by dpgeorge