micropython/micropython #8737: Adding mongoose as a package

AmirHmZz, 2022-06-07:
Had anyone tried to port https://github.com/cesanta/mongoose to micropython?

jimmo, 2022-06-08:
@AmirHmZz this is MicroPython's bug tracker. This sort of question is better suited to the forum.

If you would like this to be considered as a feature enhancement please see the questions in the "Feature request" placeholder text at https://github.com/micropython/micropython/issues/new/choose

As MicroPython uses LWIP and mbedTLS it probably isn't too difficult to integrate but the the main question that would need to be addressed is how someone would actually use this -- obviously Python bindings would need to be implemented. For the same code size, as well as ease of supporting asyncio, I suspect you might be better off doing the whole thing in Python.

AmirHmZz, 2022-06-10:
@jimmo Yes I completely agree with you. The code size would be the same.
But as you know there's lack of a suitable HTTP server in Micropython. especially support for HTTP pipelining which can reduce number of open socket connection for a web server on an embedded system.

AmirHmZz, 2022-08-14:
I've had a [small conversation](https://github.com/cesanta/mongoose/discussions/1671#discussioncomment-3392841) with mongoose repository maintainer @cpq and he asked me some questions about license and other stuff. I've also invited him to join this issue in order to make some negotiation with Micropython maintainers.

cpq, 2022-08-14:
Thanks @AmirHmZz 

Mongoose is a C/C++ networking library that provides API for HTTP, Websocket, MQTT, SNTP, and async DNS resolver. It makes it easy to implements rich networking on embedded devices.

My question is - if mongoose bindings for micropython are developed, can they be integrated into the mainstream micropython repo? The biggest issue I see is the licensing. Mongoose is dual licensed, GPLv2 / commercial. Micropython is MIT, though some parts are under different licenses including GPLv2.

@jimmo could you shed a light on that please?

jimmo, 2022-08-15:
> My question is - if mongoose bindings for micropython are developed, can they be integrated into the mainstream micropython repo? The biggest issue I see is the licensing. Mongoose is dual licensed, GPLv2 / commercial.

Yes we are unlikely to add any code to the main repo that results in the GPL license being applied to the resulting binary.

It';s worth noting we do something _similar_ with the bluekitchen btstack stack. When you enable this, your binaries are now subject to their (dual commercial) license. We don't publish any official binaries with this enabled though.

However in the case of a GPL dependency, I'm not quite sure we can do this within the rules.

You are more than welcome to develop bindings in an external repo that a user can integrate into their own builds, this is exactly what the USER_C_MODULES feature is for. See https://docs.micropython.org/en/latest/develop/cmodules.html for more information. https://github.com/v923z/micropython-ulab/ is a great example of this.

> Micropython is MIT, though some parts are under different licenses including GPLv2.

We have a note in LICENSE to clarify this -- "Note that (L)GPL-licensed code listed below is only
used during the build process and is not part of the compiled source code."

(The main example here is dfu.py which although not a problem, we plan to replace anyway)

closed 2022-08-15 by jimmo

scaprile, 2022-08-16:
One more question.
Mongoose is an event-driven library, it requires an event manager to be periodically called, which in turn will call the relevant callbacks for the desired handlers. Can you provide links to documentation on how to achieve this (besides observing the source code). Thanks!

jimmo, 2022-08-17:
@scaprile Probably the best example of this is what we do with the BLE stack (NimBLE) and the IP stack (LWIP).

Sorry I know you said documentation but unfortunately all I have to offer is the source code. I'll try and explain it a bit though.

It's slightly different for each port. On ESP32 they run in background FreeRTOS tasks. On STM32 we use the [systick to run the LWIP poll every n'th tick](https://github.com/micropython/micropython/blob/master/ports/stm32/main.c#L425) via [pendsv](https://github.com/micropython/micropython/blob/master/ports/stm32/mpnetworkport.c#L71), and we use the [soft timer to run the BLE poll](https://github.com/micropython/micropython/blob/master/ports/stm32/mpbthciport.c#L52).

The reason LWIP uses systick is because it's safe to run in interrupt context (although note we bounce it via pendsv so that it can be interrupted by higher-priority interrupts), whereas we want to run BLE in scheduler context as the BLE stack will end up invoking Python callbacks which we want to be "soft" (see https://docs.micropython.org/en/latest/reference/isr_rules.html). The SoftTimer is itself built on systick and pendsv but it's just a really convenient way of essentially saying "call micropython.schedule/mp_sched_schedule at this interval".

On ESP32 because the BLE stack runs in a FreeRTOS task, we need to do [https://github.com/micropython/micropython/blob/master/extmod/modbluetooth.c#L1189](some extra tricks) to get it into scheduler context.

The only issue right now is that SoftTimer isn't yet available on ESP32 and Unix.

However. If you just want to get this working as a proof of concept that can be improved later, what I would suggest is instead making your event manager poll method accessible from Python, and then have the user code call it in their main loop. Then later (i.e. when SoftTimer is available on all ports...) you can make this automatic.

cpq, 2022-08-17:
@jimmo thanks!

Btw mongoose provides a simple soft timer implementation [timer.c](https://github.com/cesanta/mongoose/blob/master/src/timer.c) that uses a monotonic clock primitive like systick.

Unrelated - Jim, your name seems familiar... Did you happen to work as Google SRE at some point?

jimmo, 2022-08-17:
@scaprile @AmirHmZz I wonder if perhaps rather than trying to run Mongoose's http server in MicroPython, it might be more useful to work on adding MicroPython into Mongoose as a way of adding scripting capabilities, and then expose all your existing services with Python APIs. The point is that rather than MicroPython being the "OS", you make Mongoose the OS / task manager / peripheral controller (which is of course its natural way of operating) running MicroPython as a task. This has been done in other RTOS projects (including Riot), and we maintain a Zephyr port in this repo, and for the most part this is how our ESP32 port works too. There's been some discussion of doing this for Mynewt too.

> Unrelated - Jim, your name seems familiar... Did you happen to work as Google SRE at some point?

@cpq (lsm iirc...? it's been a while!) Yes! A fellow member of the borgmon appreciation society! I'm sure we probably discussed bigtable replication delay at some stage :) Enjoying working on different problems...12-or-so orders of magnitude different.

cpq, 2022-08-17:
@jimmo yes, lsm I was! Nice to meet again, and see you contributing to micropython!

scaprile, 2023-11-02:
https://github.com/cesanta/mongoose/pull/2450