micropython/micropython #8746: user modules do not compile with the newest version

uraich, 2022-06-10:
Hi,
I pulled the newest version of micropython and I now see errors when trying to add my user modules. I try to add ulab and a module I wrote myself. This is what I see:

<command-line>: error: expected identifier or '(' before numeric constant
/opt/ucc/micros/esp32/micropython/ports/esp32/build-GENERIC/genhdr/moduledefs.h:35:61: note: in expansion of macro 'MODULE_DHT11RAW_ENABLED'
 extern const struct _mp_obj_module_t dht11Raw_user_cmodule, MODULE_DHT11RAW_ENABLED;
                                                             ^~~~~~~~~~~~~~~~~~~~~~~
<command-line>: error: expected identifier or '(' before numeric constant
/opt/ucc/micros/esp32/micropython/ports/esp32/build-GENERIC/genhdr/moduledefs.h:111:57: note: in expansion of macro 'MODULE_ULAB_ENABLED'
 extern const struct _mp_obj_module_t ulab_user_cmodule, MODULE_ULAB_ENABLED;
                                                         ^~~~~~~~~~~~~~~~~~~
/opt/ucc/micros/esp32/micropython/py/objmodule.c:169:5: error: macro "MP_ROM_PTR" passed 2 arguments, but takes just 1
     MICROPY_REGISTERED_MODULES
     ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /opt/ucc/micros/esp32/micropython/py/objmodule.c:40:
/opt/ucc/micros/esp32/micropython/ports/esp32/build-GENERIC/genhdr/moduledefs.h:37:70: error: 'MP_ROM_PTR' undeclared here (not in a function); did you mean 'MP_QSTR_PTR'?
 #define MODULE_DEF_MP_QSTR_DHT11RAW { MP_ROM_QSTR(MP_QSTR_dht11Raw), MP_ROM_PTR(&dht11Raw_user_cmodule, MODULE_DHT11RAW_ENABLED) },
                                                                      ^~~~~~~~~~
/opt/ucc/micros/esp32/micropython/ports/esp32/build-GENERIC/genhdr/moduledefs.h:176:5: note: in expansion of macro 'MODULE_DEF_MP_QSTR_DHT11RAW'
     MODULE_DEF_MP_QSTR_DHT11RAW \
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/ucc/micros/esp32/micropython/py/objmodule.c:169:5: note: in expansion of macro 'MICROPY_REGISTERED_MODULES'
     MICROPY_REGISTERED_MODULES
     ^~~~~~~~~~~~~~~~~~~~~~~~~~

Which modifications are needed to adapt to the newest version?

dpgeorge, 2022-06-10:
You need to remove the 3rd (the last) argument of any `MP_REGISTER_MODULE()` uses.

uraich, 2022-06-10:
Hi Damien,
Thanks! This did the trick.
Uli