micropython/micropython #8748: Support for buffer protocol in btree module

AmirHmZz, 2022-06-10:
Allowing `btree` module to get binary data as `memoryview` and `bytearray` will deduplicate **RAM allocation** for converting `memoryview` and `bytearray` to `bytes`.

```Python
db = btree.open(open("mydb", "w+b"))
db[memoryview(b"test_key")] = memoryview(b"test_value")
```

Currently the snippet generates this exception : 
```Python
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't convert 'memoryview' object to str implicitly
```

dlech, 2022-06-10:
I don't see how this saves memory. Wrapping a bytes object in a memory view creates two objects whereas using a bytes object (or as required in this case str object) directly will use less memory.

AmirHmZz, 2022-06-10:
The snippet that I wrote was just for reproducing that exception. Actually in my own project I have a streaming network buffer (`bytearray`) which is used to process received data from network. Data throughput of that stream is big enough to avoid additional memory allocations. My own code looks something like this : 

```Python
import btree
db = btree.open(open("mydb", "w+b"))
buf = bytearray(256)
mv = memoryview(buf)
while (size := stream.readinto(buf)) != 0 : 
    db[mv[:10]] = mv[10:size]
```

dlech, 2022-06-10:
The problem here is that the key needs to be a string, not bytes. In the underlying C code, strings are expected to be zero-terminated. So unless the data in `buf` includes the zero termination, this wouldn't work as any code that expects the zero-termination would read past the end of the string. Since the btree is going to have a string anyway, changing the code to:

    db[str(mv[:10], 'utf-8')] = mv[10:size]

probably doesn't make a difference in what would be done internally anyway.

AmirHmZz, 2022-06-10:
But according to [documentation](https://micropython-glenn20.readthedocs.io/en/latest/library/btree.html) both _key_ and `value` must be `bytes` : 

> BTree database work as close a possible to a way standard [dict](https://micropython-glenn20.readthedocs.io/en/latest/library/builtins.html#dict) type works, one notable difference is that both keys and values must be [bytes](https://micropython-glenn20.readthedocs.io/en/latest/library/builtins.html#bytes) objects (so, if you want to store objects of other types, you need to serialize them to [bytes](https://micropython-glenn20.readthedocs.io/en/latest/library/builtins.html#bytes) first).

dlech, 2022-06-10:
I stand corrected. It does look like `mp_obj_str_get_data()` could be replaced with `mp_get_buffer_raise()` in the btree get and set methods.

AmirHmZz, 2022-06-10:
@dlech Does your PR supports **both** of reading and writing values in database?

dlech, 2022-06-10:
For reading, to avoid allocating a new bytes object for the value read, a new `readinto` method  would need to be added.

AmirHmZz, 2022-06-10:
In case of reading I just meant the key which is needed to retrieve it's corresponding value but It would be great to have something live `getinto()` which works similar to `FileIO.readinto()`. I guess it might add some complexity to code or maybe some additional allocation to cache retrieved record in order to avoid repetitive searches every time `getinto()` is called. Or might It would be better that `getinto()` returns something like a `ByteIO` object.

dlech, 2022-06-10:
All of the input parameters for key and value and index (key) of `[]` now allow any buffer protocol object with my PR.

AmirHmZz, 2022-06-10:
Another approach that many databases use is returning a `Cursor` as the result which behavies like a byte stream.
Another advantage of beeing able to read data of database in a streaming manner, is controlling RAM usage while getting a range of data ( many values ) from database.

dlech, 2022-06-10:
> ```python
>     db[mv[:10]] = mv[10:size]
> ```

I think this will still allocate two new slice objects and two new memory view objects.

AmirHmZz, 2022-06-10:
Yes you're correct. But passing memoryview instead of bytes is a common optimization in Micropython. This technique is supported in `socket` and `FileIO`. I think going further than this might need some low level APIs like `mp_get_buffer_raise()` which is able to read a slice of buffer instead of reading all of it. Anyway passing a memoryview is better than passing bytes or slices of bytearray which finally leads to duplication of data.

marcidy, 2022-06-16:
memoryview is common optimization to view bytes in RAM, not bytes on the filesystem/on disk.  I realize we're talking about a filesystem in flash, but it's still a filesystem more similar to a ramdisk than actual RAM pointed to by an array or similar.

Do you have an example where memoryview is used to access the raw bytes directly on the filesystem?  That sounds more like mmap to me.

dpgeorge, 2022-06-20:
Implemented in a565811f23c675ccbeed636c065828cdaf5fa96a

closed 2022-06-20 by dpgeorge