micropython/micropython #8751: [ESP32-S3] BLE.init() interrupts serial output

n-elia, 2022-06-10:
Hi,
I'm running MicroPython v1.18 on a ESP32-S3 based board (_Unexpected Maker TinyS3_). I'm using the onboard USB to get a remote REPL.

We're successfully running a firmware on older ESP32 boards and, after moving to the new ESP32-S3, the method `bluetooth.active()` appears to break the serial connection.

**To quickly reproduce the issue:**
```python
# main.py
import bluetooth

ble = bluetooth.BLE()
print("Activating BLE...")
ble.active(True)
print("Activated BLE.")
```

**Output:**
```
MPY: soft reboot
Activating BLE...
```

**Expected output** (from an older ESP32 board):
```
MPY: soft reboot
Activating BLE...
Activated BLE.
MicroPython v1.18 on 2022-01-17; ESP32 module with ESP32
Type "help()" for more information.
>>>
```

As you may see, after executing `bluetooth.active()` the connection hangs. The terminal becomes unresponsive (tried with rshell, picocom, Thonny IDE) and the only way to re-gain the control is to have a `sleep` statement at the beginning of the script and to hard reset the board.

This was tested at least on the following builds:
```
Nightly builds
[v1.18-610-gcf7d962cf (2022-06-10) .uf2](https://micropython.org/resources/firmware/tinys3-20220610-unstable-v1.18-610-gcf7d962cf.uf2) [[.bin]](https://micropython.org/resources/firmware/tinys3-20220610-unstable-v1.18-610-gcf7d962cf.bin) [[.elf]](https://micropython.org/resources/firmware/tinys3-20220610-unstable-v1.18-610-gcf7d962cf.elf) [[.map]](https://micropython.org/resources/firmware/tinys3-20220610-unstable-v1.18-610-gcf7d962cf.map)
[v1.18-605-gee9feacc0 (2022-06-10) .uf2](https://micropython.org/resources/firmware/tinys3-20220610-unstable-v1.18-605-gee9feacc0.uf2) [[.bin]](https://micropython.org/resources/firmware/tinys3-20220610-unstable-v1.18-605-gee9feacc0.bin) [[.elf]](https://micropython.org/resources/firmware/tinys3-20220610-unstable-v1.18-605-gee9feacc0.elf) [[.map]](https://micropython.org/resources/firmware/tinys3-20220610-unstable-v1.18-605-gee9feacc0.map)
[v1.18-602-gf63b4f85a (2022-06-09) .uf2](https://micropython.org/resources/firmware/tinys3-20220609-unstable-v1.18-602-gf63b4f85a.uf2) [[.bin]](https://micropython.org/resources/firmware/tinys3-20220609-unstable-v1.18-602-gf63b4f85a.bin) [[.elf]](https://micropython.org/resources/firmware/tinys3-20220609-unstable-v1.18-602-gf63b4f85a.elf) [[.map]](https://micropython.org/resources/firmware/tinys3-20220609-unstable-v1.18-602-gf63b4f85a.map)
[v1.18-601-g3452ee58d (2022-06-09) .uf2](https://micropython.org/resources/firmware/tinys3-20220609-unstable-v1.18-601-g3452ee58d.uf2) [[.bin]](https://micropython.org/resources/firmware/tinys3-20220609-unstable-v1.18-601-g3452ee58d.bin) [[.elf]](https://micropython.org/resources/firmware/tinys3-20220609-unstable-v1.18-601-g3452ee58d.elf) [[.map]](https://micropython.org/resources/firmware/tinys3-20220609-unstable-v1.18-601-g3452ee58d.map)
```

<details>
  <summary>Firmware which caused the problem</summary>
  
    
```python
# main.py
import sys
import time
import uasyncio
import bluetooth

from ble_advertising import decode_name

_IRQ_SCAN_RESULT = const(5)
_IRQ_SCAN_DONE = const(6)

async def main_loop():
    ble_coro_handle = uasyncio.create_task(ble_coro())
    await uasyncio.sleep(5)

async def ble_coro():
    ble = bluetooth.BLE()
    print("Activating BLE...")
    ble.active(True)
    print("Activated BLE.")
    ble.irq(irq_handler)
    try:
        ble.gap_scan(0, 30000, 30000)
        while True:
            await uasyncio.sleep(5)

    except uasyncio.CancelledError:
        ble.gap_scan(None)
        print("ble_coro: task cleaned up")
        
  
def irq_handler(self, event, data):
    if event == _IRQ_SCAN_RESULT:
        addr_type, addr, adv_type, rssi, adv_data = data
        device_name = decode_name(adv_data) or 'unknown'
        device_rssi = int(rssi)
        ts = time.time()
        print(f"Found device with name {device_name}, RSSI {device_rssi}, at time {str(ts)}")

    elif event == _IRQ_SCAN_DONE:
        print("Scan cycle terminated.")

if __name__ == "__main__":
    try:
        uasyncio.run(main_loop())
    except KeyboardInterrupt:
        print('Interrupted')
    finally:
        uasyncio.new_event_loop() 
```

```python
# ble_advertising.py
from micropython import const
import struct

_ADV_TYPE_NAME = const(0x09)

def decode_field(payload, adv_type):
    i = 0
    result = []
    while i + 1 < len(payload):
        if payload[i + 1] == adv_type:
            result.append(payload[i + 2 : i + payload[i] + 1])
        i += 1 + payload[i]
    return result

def decode_name(payload):
    n = decode_field(payload, _ADV_TYPE_NAME)
    return str(n[0], "utf-8") if n else ""
```
</details>

Thanks for your help.

UnexpectedMaker, 2022-06-13:
@jimmo Any idea what this could be mate?

jimmo, 2022-06-14:
@n-elia @UnexpectedMaker I have an S3 here, will test and get back to you (unfortunately won't be able to look until next week though).
We have to jump through some hoops to wire up the BLE events on ESP32 (and this is why some features are unavailable on ESP32) and maybe that's broken somehow on S3. See https://github.com/micropython/micropython/pull/7046 for the ongoing work to fix this, it might be worth trying this.

FWIW (unrelated to your problem, but FYI), if you're using BLE and asyncio, have you seen https://github.com/micropython/micropython-lib/tree/master/micropython/bluetooth/aioble

n-elia, 2022-06-14:
Thank you for your detailed answer!

Yes, we are using uasyncio and aioble will be great, we'll implement it for sure, thanks!

I don't know if it can help, but we noticed that if we let the device run headless (i mean without opening a remote REPL over serial connection) as BLE scanner and log its operation through WiFi, the device runs fine and we are able to receive the scanned devices over WiFi.
Instead, if we start a remote REPL it hangs as in the issue description, therefore no data is sent through WiFi.

n-elia, 2022-07-06:
Hi,

the latest stable MicroPython release for TinyS3 (`v1.19.1 (2022-06-18)`) fixed the issue.

Also, `aioble` works great and it's by far more convenient and intuitive than the builtin `bluetooth`. Thanks for the hint!

Thanks!

closed 2022-07-06 by n-elia