micropython/micropython #8753: docs: alternative build options required for `port/rp2`

r00t1900, 2022-06-11:
## description
Currently I bought a official raspberry pi pico board and a third-party pico board. One of the difference between them is the size of flash: the third-party board use an 4MB Flash instead of 2MB Flash in the original disign of pi pico.

However, If I download the precompiled `.u2f` firmware file for pi pico and then flash it to both of them, I got the same available space of filesystem:
```python
os.statvfs('/')
```
It appears to be about `1.38MB` in total(blocks*bsize), as the firmware has already took about `500-600KB` size for storage.'

## some thoughts

I guess that may be something define in the `make` steps. I lookup in `port/rp2` and found a marco in `rp2_flash.c`:
```c
#ifndef MICROPY_HW_FLASH_STORAGE_BYTES
#define MICROPY_HW_FLASH_STORAGE_BYTES (1408*1024)
#endif
```
And `1408KB` is exactly almost equal to `1.37MB`.
Is this the main reason why either W25Q16 or W25Q32 perform the same available filesystem size? I've also notice that the include part of `rp2_flash.c`:
```c
...
#include "hardware/flash.h`
...
```
But I can not find the corresponding file `hardware/flash.h`.

## questions

- **how to reconfigure for other flash size**
  What should I do if I would like to build an alternative micropython firmware for a pico-like board? Maybe different in flash,crystal and pin layout. Where should I reconfigure before building?(mainly is the flash size)

- **how to operate more flash components**
  The official pi pico is designed with flash connected to XIP via a single QSPI interface. But what should I do if I want to operate one more flash via the normal/common spi interface?(the speed is not a problem). 
  I would like to storage data into different flash components, like 8 flash chips and switch them by CS signal. Can they be use as an another standalone filesystem with littlefs?
  If no, is there any spi-flash api for micropython to `read/write/erase` external flash component? If no, then I have to implement all basic spi-flash functions.

Thanks for reading and looking forward to your replies :)

> In this case we can assume that the flash is W25QXX for getting started, while XX is range from `16` to `128`.

peterhinch, 2022-06-11:
This support query should be raised in [the forum](http://forum.micropython.org/index.php) rather than in this bug tracker.

Re using a set of SPI flash chips to create a filesystem see [this repo](https://github.com/peterhinch/micropython_eeprom).

r00t1900, 2022-06-14:
> This support query should be raised in [the forum](http://forum.micropython.org/index.php) rather than in this bug tracker.
> 
> Re using a set of SPI flash chips to create a filesystem see [this repo](https://github.com/peterhinch/micropython_eeprom).

- I found some interesting posts on the forum and some marco I found useful is `PICO_FLASH_SIZE_BYTES` and `MICROPY_HW_FLASH_STORAGE_BYTES`. Is that correct? I would like to try it and find out myself. you are so brilliant! I've also found your replies on the micropython forum!  Posts are:
  - https://forum.micropython.org/viewtopic.php?f=21&t=10745&hilit=flash
  - https://forum.micropython.org/viewtopic.php?f=21&t=11438&hilit=flash
  - https://forum.micropython.org/viewtopic.php?f=21&t=11054&hilit=flash
  - https://forum.micropython.org/viewtopic.php?f=21&t=10842&start=10&hilit=flash
  - https://forum.micropython.org/viewtopic.php?f=21&t=10658&p=58837#p58837
  - https://forum.micropython.org/viewtopic.php?f=21&t=9646&hilit=flash&start=20

- Rpeo [micropython_eeprom](https://github.com/peterhinch/micropython_eeprom) is very good! that's what I am pointing for. What's the different bewtween `rp2.Flash` and this `spi_flash.Flash`? Is the former can only operate internal flash mounted on QSPI and the latter can operate most of the external flash wired on SPI bus?

Josverl, 2025-11-06:
Closing as not an actual bug report