micropython/micropython #8754: incomplete `hashlib` library and no crypto library found

r00t1900, 2022-06-11:
I use the latest version(1.18) of pi-pico micropython firmware for some application development. 

However I have some unresolved problems:

- `hashlib`: I could only find the `hashlib.sha256` is available, neither `md5` nor `sha1` is found. What is the reason?
- `crypto features`: No AES or RSA algorithm found but I heard that this could be added by reconfigure some parameters before building. Is there any instruction?
- ` customize api & stubs reference generation`: I can not find a suitable micropython api or stubs when develop micropython with pi-pico board or self-made rp2040 board. I use stubs for rp2 in [micropython-stubs](https://github.com/Josverl/micropython-stubs) and found the hints is not 100% exactly corresponding the firmware. For example, I can get hint that `hashlib.md5`, `hashlib.sha1` is available when programming with VSCode and [micropython-stubs](https://github.com/Josverl/micropython-stubs) , but after download `main.py` it raise an error, which mean they are not 100% corresponding for some building difference. How can I make a fit-to-board api&stubs after I build micropython with some custom libraries or options?

MicroPython is such a good thing and I am looking forward to your replies :)

oliver-joos, 2022-06-12:
For ucryptolib.aes(), uhashlib.sha1()/md5() and ussl.wrap_socket() you need USSL. USSL in turn requires MBEDTLS and MBEDTLS uses about 64kB of flash memory! I think that's why these are not enabled by default.

You can enable USSL and MBEDTLS by adding the following 2 lines to mpconfigboard.mk of your board:
```
MICROPY_PY_USSL = 1
MICROPY_SSL_MBEDTLS = $(MICROPY_PY_USSL)
```
(And thanks for pointing out "micropython-stubs"!)

r00t1900, 2022-06-14:
:+1: :+1: :+1: That's cool and I guess the marco symbols are what I need! 

## :question: How to explore more?

Now by submitting this issue I know there are `ucryptolib.aes()`, `uhashlib.sha1()` and other features that are hidden from default settings. **Where can I have a overlook of all micropython features?** Maybe I need more features on my micropython firmware and the flash size is not a problem: I would like to redesign a pico board with a bigger flash memory to fix this. 

## :question: How to increase filesystem size?

If this will increase the flash memory usage, what should I do while building? Assuming I change the flash chip from 2MiB(W25Q16) to 16MiB(W25Q128), **is there any other macro symbols that I should notice and change as well to support the new flash size?**
> I've also search `how to use flash of different size on rp2040` on the forum but I would also like to hear the official voice :)

never mind the `micropython-stubs` discussing :sun_with_face:

r00t1900, 2022-06-17:
yes I found it as well. The description in .h file is pretty good. And understood, questions in forum and bugs in issues :)