micropython/micropython #8770: RP2: sleep modes do not save power

peterhinch, 2022-06-17:
In my testing machine.lightsleep and machine.deepsleep still don't work (with V1.19 firmware).

The Pico was powered from a bench supply with 5V on vsys. There was no connection to the USB port. main.py was as follows:

import time
time.sleep(5)
import machine
machine.lightsleep(10_000)

Current consumption was ~21mA with a tiny glitch visible after 5s but no change otherwise. The same was true with deepsleep (which seems to call lightsleep).

dpgeorge, 2022-06-17:
Yes, power saving sleep is not yet implemented on rp2.  This is a good starting point to getting it working: https://ghubcoder.github.io/posts/deep-sleeping-the-pico-micropython/

dpgeorge, 2022-06-30:
Low power implemented in b004e7e397577d95404fd31aec68a5c54904a48c

xavier-draye, 2023-08-02:
Hi @dpgeorge,
Thanx for implementing lightsleep() support in rp2. Would you agree to provide additional hints to enable wake up from dormant mode  using GPIO trigger ? I am referring to your comment:
"Calling lightsleep without a specified period, in which case it uses xosc  dormant mode.  There's currently no way to wake it up from this state,  unless you write to raw registers to enable a GPIO wake up source.". 
I have a good experience of register manipulation (did that for PIO and DMA in another application), but the dormant mode manipulation seems much more complex to me...
Thanx in advance,
Xavier.