micropython/micropython #8772: RP2: Get read buffer size / partial read on USB stdin

smurfix, 2022-06-17:
I want to run a binary protocol on the RP2's USB port. The problem is that I have no way to determine how many characters are available, reading more than what's available blocks, and reading one at a time ends up being too slow.

The UART has a nice `any()` function that tells me how many characters are in the buffer. I'd like to request adding that method to its stdin handler, *or* a way to tell it to not block on partial reads, or both.

ubidefeo, 2023-06-06:
@smurfix 
I ended up here because I'm having some data loss issues when pushing data chunks to any MP board.
Have you found any workaround?
This issue doesn't seem to have gotten any attention 😬

smurfix, 2023-06-06:
Unfortunately the best I came up with is to do one-character reads. Definitely less than optimal but it gets the work done for now.

dpgeorge, 2023-06-06:
How are you currently reading the data, is it using `sys.stdin.read()`?

smurfix, 2023-06-06:
`sys.stdin.buffer.read()` actually – I need bytes, not unicode characters.

dpgeorge, 2023-06-06:
I see.

It looks like there's currently no good way to do what you want, to read more than 1 byte without blocking.

In CPython, doing `sys.stdin.buffer.read(10)` will block until 10 bytes are available.

But, CPython does have `sys.stdin.buffer.read1` (and `readinto1`) which does at most one read from the underlying device and so is effectively non-blocking if at least one character is available.  That would probably work well here, and is easy enough to implement.

CPython also has `fnctl` hacks to make stdin non-blocking, but that won't really work in MicroPython.

Need to also consider how to make it work with asyncio, ie be able to efficiently read from stdin as an async stream.

smurfix, 2023-06-06:
`read1` would work for me, of course.

I do use this with asyncio. Right now I simply assemble the input in a buffer, looping until `sys.stdin.buffer` is no longer readable, then return this. With `read1` my code would reduce to `yield uasyncio._io_queue.queue_read(stream)` plus `return stream.readinto1(buf)`.

Yes, that does mean that I should write a PR to explicitly export those `queue_read` and `queue_write` methods … how would you call them? Simply `await readable(stream)`?