micropython/micropython #8775: [STM32H755_Nucleo-144/stm32h755]: there is a working config

ska-la, 2022-06-17:
I've got a working instance of MicroPython on a board STM32H755 Nucleo-144 (CORE_CM7). The base for this were NUCLEO_H743ZI(2) board configs. It's not tested thoroughly, only user leds (LED1/2/3) and Ethernet (got ip address from router). There are some problems (may be not a problems just only a drawbacks). First of all, i'm tuning/checking pins.csv (from stm32h743) and stm32h755_af.csv (derived from stm32h743_af.csv) to correspond the board. I compare the same files, which provided with STM CubeMX. There is firmware for CORE_CM4 also, but it requires modified stm32h755.ld (not implemented yet). First question: may i ask request? Second one: i found dubbed pin definitions into pins.csv (from stm32h743), is it a real problem or not? Thanks.

RasailiTilak, 2025-01-07:
@ska-la 
Hello, I hope you are doing well. STM32H755 is dual core M7 and M4, how do we implement that? if you have any solution please could you provide it to me?

I did for NUCLEO_H753ZI.
 But for the Dual core, I didn't find the solution. 
Thank you.

ska-la, 2025-01-08:
Hello. You can filtered out Pull requests with Author: ska-la, to get my outcome. Use the [README.md](https://github.com/micropython/micropython/pull/8885/commits/4212d0d31a25a2710681c700bf787bab0c557b04) as help to build images. I used STM's CubeProgrammer to flash the images. You can replace all mentions of 755 with 753 (dir names and into a files).
I didn't complete my PR's. May be you'll do it for both boards. If you'll get working result, please notice me.
The form doesn't allow to pin archive with all files. I can send it on your email.

ska-la, 2025-01-08:
@RasailiTilak forgot to mention you in prev. comment