micropython/micropython #8776: Bug with WiFi on ESP32-S3 still there on 1.19 Micropython version.

beyonlo, 2022-06-17:
I already related this Bug here #8635 (using the MicroPython 1.18), but it still there on 1.19 MicroPython version. I'm using the board GENERIC_S3 (https://micropython.org/download/GENERIC_S3/)

**Testing WiFi as STA mode:** 

```
>>> import os
>>> os.uname()
(sysname='esp32', nodename='esp32', release='1.19.0', version='v1.19 on 2022-06-16', machine='ESP32S3 module with ESP32S3')
>>> import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True) # MicroPython congeals/stop here.
```

**Ps**: if I build by myself the firmware (without changes, just compile) WiFi works. Problem looks like be just on the ready compiled binary from MicroPython page.

This BUG was confirmed by other users on the #8635

dpgeorge, 2022-06-18:
I'll close this as a duplicate of #8635.

closed 2022-06-18 by dpgeorge

MATTYGILO, 2022-09-21:
I'm having the same problem uart works fine, usb doens't work though for esp32s3 on espidf 4.4.2. I have modified micorpython to enable usb modem instead of uart.