micropython/micropython #8778: Serial behavior under low baud rate

Tangerino, 2022-06-18:
MP Version 1.19
Target: ESP32
Serial TX <-> RX in the same port

Under low baud rate, the information takes one second to be received.
In 110bps it takes 1/110 * 10 sec = 0.09 seconds to be transmitted and the following code demonstrate that the character is received after 1 second only.

Using legacy devices where the protocol is time sensitive is a huge problem.

```python
from time import ticks_diff
from time import ticks_ms
import machine

def test(baud_rate, loops):
    uart = machine.UART(2, tx=12, rx=34, timeout=1, baudrate=baud_rate, timeout_char=1)
    for loop in range(loops):
        uart.write("U" * 1)
        now = ticks_ms()
        while True:
            if uart.any():
                elapsed = ticks_diff(ticks_ms(), now)
                rx = uart.read()
                print(f"{loop} - Baud rate:{baud_rate}. Got {len(rx)} bytes - Elapsed {elapsed} - {rx}")
                break

for baud_rate in [9600, 300, 110, 115200]:
    print(f"Testing baud rate {baud_rate}")
    test(baud_rate=baud_rate, loops=2)

```

Results
```
Testing baud rate 9600
0 - Baud rate:9600. Got 1 bytes - Elapsed 12 - b'U'
1 - Baud rate:9600. Got 1 bytes - Elapsed 12 - b'U'
Testing baud rate 300
0 - Baud rate:300. Got 1 bytes - Elapsed 373 - b'U'
1 - Baud rate:300. Got 1 bytes - Elapsed 371 - b'U'
Testing baud rate 110
0 - Baud rate:110. Got 1 bytes - Elapsed 1018 - b'U'
1 - Baud rate:110. Got 1 bytes - Elapsed 1016 - b'U'
Testing baud rate 115200
0 - Baud rate:115200. Got 1 bytes - Elapsed 1 - b'U'
1 - Baud rate:115200. Got 1 bytes - Elapsed 1 - b'U'
```

robert-hh, 2022-06-18:
It is indeed the time, the RTOS takes to report, if characters are available. It does not matter, if uart.any() is called to tell or uart.read(), even if in Micropython the timeout is set to 0. Sending starts immediately after the call to uart.write().
I faintly recall that we had that discussion earlier, but cannot remember the result.

Tangerino, 2022-06-19:
Got it. @dpgeorge Can something be done to integrate this to the main branch?
I'm using the unmodified code (I thing many of us), so no need to maintain my own code.
Cheers

Tangerino, 2022-06-19:
Implementing the above suggestion and using the same Python script I got these figures.
Bottom line, no changes in low baud rate.

```
Testing baud rate 9600
0 - Baud rate:9600. Got 1 bytes - Elapsed 3 - b'U'
1 - Baud rate:9600. Got 1 bytes - Elapsed 4 - b'U'
Testing baud rate 300
0 - Baud rate:300. Got 1 bytes - Elapsed 1493 - b'U'
1 - Baud rate:300. Got 1 bytes - Elapsed 1491 - b'U'
Testing baud rate 110
0 - Baud rate:110. Got 1 bytes - Elapsed 1018 - b'U'
1 - Baud rate:110. Got 1 bytes - Elapsed 1016 - b'U'
Testing baud rate 115200
0 - Baud rate:115200. Got 1 bytes - Elapsed 0 - b'U'
1 - Baud rate:115200. Got 1 bytes - Elapsed 0 - b'U'
```

robert-hh, 2022-06-19:
Tested again this morning. There is some improvement if the timeout in the call to uart_set_rx_timeout() is set to 1, which is the lowest accepted value. The expected timeout value has to be given as character times. Then, the additional wait time is 1 character time @ 12 bit times/character (1 + 8 + 1 + 2).
```
Testing baud rate 300
0 - Baud rate:300. Got 1 bytes - Elapsed 79 - b'U'
1 - Baud rate:300. Got 1 bytes - Elapsed 77 - b'U'
Testing baud rate 110
0 - Baud rate:110. Got 1 bytes - Elapsed 218 - b'U'
1 - Baud rate:110. Got 1 bytes - Elapsed 216 - b'U'
```

robert-hh, 2022-06-19:
The section in the code should then be:
```
    if (args[ARG_timeout_char].u_int != -1) {
        self->timeout_char = args[ARG_timeout_char].u_int;
        uint32_t min_timeout_char = 13000 / baudrate + 1;
        if (self->timeout_char < min_timeout_char) {
            self->timeout_char = min_timeout_char;
        }
        uart_set_rx_timeout(self->uart_num, self->timeout_char / min_timeout_char);
    }
```
Otherwise, timeout_char is not used at all.

Tangerino, 2022-06-19:
I just tested this but did not impose any limitation and set the timeout to 1 and I got this result
```
Testing baud rate 9600
0 - Baud rate:9600. Got 1 bytes - Elapsed 2 - b'U'
1 - Baud rate:9600. Got 1 bytes - Elapsed 2 - b'U'
Testing baud rate 110
0 - Baud rate:110. Got 1 bytes - Elapsed 218 - b'U'
1 - Baud rate:110. Got 1 bytes - Elapsed 216 - b'U'
```

Still a big lag between send and receive.

Tangerino, 2022-06-19:
I fixed this a long time ago and tried to do a PR but it not went trough.
```url
https://github.com/micropython/micropython/pull/7883
```

I can se some interrupt options and I get the desired behavior. If someone could help I would appreciate.

Tangerino, 2022-06-19:
Just a bit of context.

Millions of utility energy meters in Brazil, deployed more than 30 years ago (still nowadays), use a simple half-duplex protocol, much like a mores code, where it sends, every second, 8 bytes with energy information. Not header, no footer in protocol, just a 'silence' between frames (around 270 msec).
So timing if very important

In my RP, I can use:
```python
        self.uart = UART(app.port,
                         baudrate=110,
                         rx=pins.rx,
                         tx=pins.tx,
                         bits=8,
                         parity=None,
                         stop=1,
                         rxfifo_full_thresh=3)
```

robert-hh, 2022-06-19:
> Still a big lag between send and receive.
That's maybe caused by the UART hardware, which needs a character time to tell, if more data is on the road or not. Changing the code into:
```
    if (args[ARG_timeout_char].u_int != -1) {
        self->timeout_char = args[ARG_timeout_char].u_int;
        uint32_t char_time_ms = 12000 / baudrate + 1;
        uint32_t rx_timeout = self->timeout_char / char_time_ms;
        if (rx_timeout < 1) {
            uart_set_rx_full_threshold(self->uart_num, 1);
            uart_set_rx_timeout(self->uart_num, 1);
        } else {
            uart_set_rx_timeout(self->uart_num, rx_timeout);
        }
    }
```
results in:
```
Testing baud rate 300
0 - Baud rate:300. Got 1 bytes - Elapsed 32 - b'U'
1 - Baud rate:300. Got 1 bytes - Elapsed 32 - b'U'
Testing baud rate 110
0 - Baud rate:110. Got 1 bytes - Elapsed 87 - b'U'
1 - Baud rate:110. Got 1 bytes - Elapsed 88 - b'U'
```
So it still uses only timeout_char, but for times less than a character time the fifo_full_signal is set to 1.

robert-hh, 2022-06-19:
> Millions of utility energy meters in Brazil, deployed more than 30 years ago (still nowadays), use a simple half-duplex protocol, much like a mores code, where it sends, every second, 8 bytes with [...]

I hope there is at least a checksum in the data.

Tangerino, 2022-06-20:
Hi.
Thank for helping.
I confirmed it works fine.

closed 2022-06-20 by Tangerino