micropython/micropython #8781: mpy_ld.py: R_XTENSA_NDIFF32 not supported

phlash, 2022-06-19:
Similar to #8436, this appears to be yet another informational / relaxation-only reloc record type (62), which I have encountered in the object file output of the latest ESP-IDF compiler for ESP32S3 `xtensa-esp32s3-elf-gcc (crosstool-NG esp-2022r1-RC1) 11.2.0`

Reloc type details found here: https://github.com/jcmvbkbc/xtensa-abi/blob/master/relocations

I have applied a similar fix (ignoring the entry) to my local `mpy_ld.py`, which seems happier (although I am finding other issues, to be reported shortly!)

jimmo, 2022-08-22:
@phlash  Just came across https://www.ashbysoft.com/articles/doom-tidal/  :) Amazing!

Sorry about the delay getting to this (and #8782, #8783) but thanks for the detailed report and suggested fixes.

phlash, 2022-08-22:
No worries, glad these were useful!