micropython/micropython #8783: mpy_ld.py: Unable to emit relocations in more complex .rodata.* sections

phlash, 2022-06-19:
While linking a large(ish) project with the latest micropython and ESP-IDF compiler, I had a failure during writing of relocations in `build_mpy` due to a section named `.rodata.str1.1` which was not matched by the `kind` checking logic, resulting in an attempt to concatenate an `int` with a `str`. Given such sections are part of the general read-only data, I applied the following fix which seems to work:
```diff
@@ -965,7 +976,7 @@ def build_mpy(env, entry_offset, fmpy, native_qstr_vals, native_qstr_objs):
     for base, addr, kind in env.mpy_relocs:
         if isinstance(kind, str) and kind.startswith(".text"):
             kind = 0
-        elif kind in (".rodata", ".data.rel.ro"):
+        elif isinstance(kind, str) and (kind.startswith(".rodata") or kind.startswith(".data.rel.ro")):
             if env.arch.separate_rodata:
                 kind = rodata_const_table_idx
             else:
```

jonnor, 2023-07-30:
I hit the same issue when working on https://github.com/emlearn/emlearn-micropython - which was some quite simple 100 line code. I did some other hack, but will try to test this fix.

jonnor, 2023-07-30:
This fix looks to work for me. I have created a merge request with your fix @phlash - hope you dont mind

phlash, 2023-07-30:
You're welcome, I didn't have a fork to commit to / issue a PR from, so thanks!

dpgeorge, 2023-09-01:
Fixed by 4837ec336a8047a1707534315f6b4b8aeac3f891

closed 2023-09-01 by dpgeorge