micropython/micropython #8785: Errors with "round" for some values.

jkowalch, 2022-06-19:
For certain values, "round" produces incorrect results.  Python 3.7.8 works correctly.
Example:
>>> round(80.36, 1)
80.4
>>> round(80.25, 1)
80.2
>>> round(80.26, 1)
80.30001
>>> round(80.27, 1)
80.30001

MicroPython version:  1.18 on 2022-01-17 w/ Raspberry Pi Pico.  Same results on prior version 1.17.

dlech, 2022-06-19:
Sounds like #8117 and related issues. 80.3 doesn't have an exact representation in 32-bit (single precision) floats used in many MicroPython boards while CPython uses 64-bit (double precision) floating point values.

dpgeorge, 2022-06-20:
This is the same as #4212, it's a problem with printing the float (not rounding it).

So I'll close the issue.  If you think it is a problem with `round()` itself then please provide details.

closed 2022-06-20 by dpgeorge

jkowalch, 2022-06-20:
You’re correct (of course).  Just printing the result produces the format issue.  It is not an issue with “round.”  This was my very first bug report and I didn’t research it like I probably should have before I posted.  Sorry to have caused you unnecessary work.  Love MPy and the Pi Pico.  

J

 

Closed #8785 <https://github.com/micropython/micropython/issues/8785>  as completed.

dpgeorge, 2022-06-20:
No problem.  This issue comes up quite a lot, so would be good to do something about it (but see linked issue as to why it's hard).