micropython/micropython #8789: Add .toggle() function on ESP32 Port

uPesy, 2022-06-20:
* ESP32 Port

* The function `Pin.toggle()` in RP2040 port is not implemented in the ESP32 Port. This function enables to reverse the logic state of a Pin.
[image]

fconrotte, 2022-09-18:
Hello !

I teach kids about microPython on the cansat project http://cansat.lu/

I tell them to look at the doc to find the meaning of functions but the first one they use PIN.toggle() is indeed not documented :)

Where can I file a bug report for this ?

robert-hh, 2022-09-18:
> Where can I file a bug report for this ?

Here, under the issues tab.

millosolomillo, 2022-10-17:
> Hello !
> 
> I teach kids about microPython on the cansat project http://cansat.lu/
> 
> I tell them to look at the doc to find the meaning of functions but the first one they use PIN.toggle() is indeed not documented :)
> 
> Where can I file a bug report for this ?

Issue opened at  #9652