micropython/micropython #8793: RPico congeals/stop when using two cores and writing on the flash

beyonlo, 2022-06-20:
Hello.

I'm using `RPico` with MicroPython version `1.19.1`. I'm using the ready MicroPython firmware from the the download page: https://micropython.org/resources/firmware/rp2-pico-20220618-v1.19.1.uf2

This example write data on flash using `core0` and use the `core1` just for watch the pin_out

**Code:**

```
# This example write data on flash using core0 and use the core1 just for watch the pin_out
import time, gc, _thread, os
from machine import freq, Pin
freq(270_000_000)
print('Clock running at: {}'.format(freq()))
gc.disable()

system = os.uname()
print('{}: {}\n'.format('os.uname()', system))

global p_out
p_out = Pin(0, Pin.OUT)
p_out.off()

def write_data(msg):
    f = open('data.txt', 'w')
    res = msg
    res = f.write(res)
    f.close()
    return res

def core0():
    time.sleep(2)
    while True:
        gc.collect()
        time.sleep(0.5)
        msg = 'asdhaksd'
        res = write_data(msg)
        print(res, p_out.value())

def core1():
    while True:
        p_out.on()
        time.sleep_ms(1)
        p_out.off()
        time.sleep_ms(1)

_thread.start_new_thread(core1, ())

core0()
```

**Output:**

```
$ mpremote run rpico_th1.py 
Clock running at: 270000000
os.uname(): (sysname='rp2', nodename='rp2', release='1.19.1', version='v1.19.1 on 2022-06-18 (GNU 11.2.0 MinSizeRel)', machine='Raspberry Pi Pico with RP2040')

8 0
8 1
8 0
8 1
8 1
8 0
8 0
8 1
8 1
8 0
8 0
8 1
8 1
8 0
8 0
8 1  <- Congeals/stop here!

```

**Ps 1**: Sometimes the Bug happen, sometimes not. So, if Bug do not happen at the first time, just try more times. It is very easy to repeat until problem happen.
**Ps 2**: I tested in two different `RPico` to be sure that is not board problem
**Ps 3**: If you invert the example: writing data on flash using `core1` and use the `core0` just for watch the pin_out, will be very worst, I mean, the problem always happen!

Thank you.

jimmo, 2022-06-21:
> freq(270_000_000)

@beyonlo !!!

Does this also happen at the rated 133MHz?

beyonlo, 2022-06-21:
Hello @jimmo

> Does this also happen at the rated 133MHz?

Yes. Below test running at 133MHz.

Output:

```
$ mpremote run rpico_th1.py 
Clock running at: 133000000
os.uname(): (sysname='rp2', nodename='rp2', release='1.19.1', version='v1.19.1 on 2022-06-18 (GNU 11.2.0 MinSizeRel)', machine='Raspberry Pi Pico with RP2040')

8 1
8 1
8 0
8 1
8 1
8 0 <- Congeals/stop here!

```

dpgeorge, 2022-07-12:
Fixed by 662dc8602bf0ab99332748d178b0e950189df9e7

closed 2022-07-12 by dpgeorge