micropython/micropython #8794: UART sendbreak bug

AeroSynth, 2022-06-21:
There is a bug with the uart.sendbreak() method.
The manual states the break signal lasts 13 bits.  It actually is 11.
This has caused errors with external sensors that utilize the SDI-12 protocol.  It requires at least 13 bits @1200 baud.  Or line needs to go for at least 12ms.  Now, it is 9.167 ms.

AeroSynth, 2022-06-22:
Just verified it's the same on V1.18.

AeroSynth, 2022-06-28:
I found a work-around that may actually be better.
the follwoing example sends a 15ms break over the SDI-12 interface.  This interface is similar to a UART, but is simplex.  So a gpio is used to switch between Rx and Tx.  SDI-12 runs at 1200 baud.  Three wires are used: +12V, data, and ground.  Data switches between 0 and 5V.  It's an older standard, slow, but quite robust for sensors as it can provide power to the sensors.
One reason this may be better is that a uart.write() tends to return immediately, so you must add the delay otherwise switching the Rx to Tx will interrupt the break..
Here is the code snippet:
```
from pyb import UART, Pin, delay, ADC
​
def send_break():
    tx = Pin('C12',Pin.OUT) #redefine uart tx line as gpio output
    tx.value(1)
    d.value(0) #set direction register.
    tx.value(0)
    delay(15)
    tx.value(1)
    d.value(1) #set to receive
    #delay(20)
    uart = UART(5,1200) #set C12 back to UART
    
uart = UART(5,1200) #set C12 to UART
v12=Pin('B9',Pin.OUT) #v12 enable
sel=Pin('A4',Pin.OUT) #v select 1=12V, 0=5V
sdi_en = Pin('C1',Pin.OUT) #sdi enable 12V to SDI and 5V supply
d = Pin('C2',Pin.OUT) #uart data direction, 0=xmit, 1=recv
Vin = ADC(Pin('C5'))
​
#set pin values
v12.value(0) #turn on VSW
sel.value(1) #set VSW to 12V
sdi_en.value(1) #enable SDI output voltage
d.value(0) #set SDI to transmit
delay(100) #wait for voltage to come up
print(6.*Vin.read()*3/4096.) #print the voltage
​
#delay(100)
send_break() #send break.  returns in Rx mode
​
#send a message
d.value(0) #set for xmit mode
uart.write(b'test')
d.value(1) #set for recv mode
#uart.write(b'there.')
delay(50)
​
#now send a regular break
''' NOTHING IS OUTPUTTED.  IT DOESN'T WORK '''
d.value(0) #set for xmit mode
uart.sendbreak()
d.value(1) #set for recv mode
​
#confirm we can send more data
delay(25)
d.value(0) #set for xmit mode
uart.write(b'hello')
d.value(1) #set for recv mode
​
​
```

robert-hh, 2022-06-28:
That's the way how I iimplemented sendbreak() in the SAMD port in C, since it had the same issue.