micropython/micropython #8795: RFC: Move forums to GitHub Discussions

jimmo, 2022-06-21:
The forums continue to run well on phpbb and I am hesitant to mess with something that is working, but there are definitely a few things I'd like to improve. We looked at migrating to Discourse, which I think would be an overall improvement over phpbb, but recently GitHub have added "Discussions" support, and I think this might be a better direction.

- Keep everything in one place (GitHub) for code, issues, PRs, and now discussion.
- Much more modern editing (including emoji response, threads, accepted answer, etc) and markup tools (code formatting, etc).
- Many users already have GitHub accounts, whereas current creating a forum.micropython.org account is an extra step (also it's not possible for new users to create new threads on the existing forum until they reply to some other threads).
- Better support for notifications and subscriptions.
- Don't need to run/maintain our own server.

The modern editing/markup tools are a big factor for me. It may sound a bit trivial but the emoji response would be really nice just to judge engagement, and the accepted answer feature would make it easier for people to make use of previous threads. There's also category support (just like phpbb) but also tagging.

Downsides:
 - There's no automatic way to import history, so we'd need to leave the current forum running in read-only mode. (FWIW, this was also an issue with moving to Discourse, there's a phpbb importer but it had issues). There is a huge wealth of information in the forums. (It looks like it _should_ be possible to use the GitHub API to import thread content, although all the posts would be from a robot account but we could annotate the message text with the original author).
 - No way to import users, so people would need to create GitHub accounts to post/reply (this is really no different to the current forum, so this really only impacts the (small number of?) people who have forum accounts but not GitHub accounts).
 - More dependence on a third-party service (GitHub).

Here's some example projects currently using GitHub notifications:

- https://github.com/home-assistant/frontend/discussions/
- https://github.com/nodejs/node/discussions
- https://github.com/faster-cpython/ideas/discussions

More info here: https://docs.github.com/en/discussions

I propose that we could create an [organisation-level discussion](https://github.blog/changelog/2022-04-12-organization-discussions/) (conceptually, spanning micropython and micropython-lib).

Keen to hear what people think! cc @peterhinch @robert-hh @davehylands @stinos @kevinkk525 @mcauser @mattytrentini @scruss @projectgus (and @dpgeorge )

iabdalkader, 2022-06-21:
I moved our forums from phpBB to Discourse a little over a year ago maybe, they did the migration for us (there was an option to do migration yourself, but I didn't want to risk it). It went really well, as I was told migration from phpBB is straightforward, and you get a chance to review everything before the handover. Also they offer different plans for hosting, so you don't have to worry about self-hosting either. Downside, it will cost more if you require more bandwidth, K-views/emails per month and storage etc... You should checks phpBB stats first to determine which plan is needed and if it works with the current budget. FWIW we're on the standard plan (20 GB storage, 100k emails, and 100k pageviews per month) and it's more than enough for now. Overall, I'm very happy with the decision and it is definitely worth it.

cwalther, 2022-06-21:
I have no experience with GitHub Discussions, am pretty happy with phpBB except for minor gripes like lack of inline-code markup (` `` ` in Markdown), and the few times I have used Discourse I just found it confusing.

One major feature of phpBB for me, that I would hate to lose, is Atom/RSS feeds. They allow me to 1. read everything in one place in my newsreader, without having to check dozens of websites and 2. keep a local, offline-searchable archive. At a quick glance, GitHub Discussions doesn’t seem to have that, is that correct?

beyonlo, 2022-06-21:
My opinion is that the best option is to migrate to GitHub Discussions, "Keeping everything in one place (GitHub) for code, issues, PRs, and now discussion". That will be amazing at the long run!

dlech, 2022-06-21:
I'm just one data point, but I currently don't participate in the forums because it is another thing I would have to subscribe to and one more reminder email to delete every day. Discord is the same in this regard. But if it was on GitHub, I might actually chime in from time to time.

chrismas9, 2022-06-21:
It appears you can move issues to discussions. That could eliminate all the "This is the bug tracker, this question would be better asked on the forum" by just bumping them to discussions.

You can create an issue from a discussion. This would allow discussion of potential bugs, eg is it a bug, a misunderstanding or deliberate limitation without cluttering the issue tracker and then create an issue from it if it is a bug or feature request. All the discussion history is copied into the issue.

RFCs might be better started in Discussions with a wider audience, then moved to issues when a consensus is nearer.

If you can search Discussions and Issues with the same search that might yield answers more quickly.

I think they are compelling reasons to use Discussions and would help keep the issue tracker clean.

dlech, 2022-06-21:
> It appears you can move issues to discussions.

Technically speaking you can't "move" the issue. You can "convert to a discussion" which means that the issue is copied to a new discussion and the issue is closed and locked but not removed or hidden.

robert-hh, 2022-06-21:
The question is how to keep the Discissions forum clean. Dave, Peter, me and others do regularly clean out unwanted contributions from the forum. That task should not be done by Damien or other people, who seem already need more than a 24 hour day for maintaining the firmware.

scruss-elmwood, 2022-06-21:
Note that opinions here will skew very strong towards active github users because you have to be an active github user to comment here. Github is not a welcoming interface for new users.

jimmo also started a thread on the forum: [RFC: Move forums to GitHub Discussions - MicroPython Forum](https://forum.micropython.org/viewtopic.php?f=8&t=12577). So now we have *two* places to look at ...

mattytrentini, 2022-06-22:
I believe we _must_ move away from phpbb. It's an impediment to get people involved - particularly for beginners - as the interface and markup and awkward to use. Further, it _looks_ old and dated, and I think it reflects poorly on MicroPython. These aren't just my opinions, I've heard it from _many_ people new to MicroPython.

Given that I think we need to move away from phpbb, Github Discussions is an _excellent_ replacement. It's modern, lightweight and the markup is sane (coloured syntax highlighting will be _very_ welcome!). I was advocating for Discourse before Discussions became available but Discussions is, generally, at least as good and better in many areas; such as the ease with referring to issues and PRs.

I'd be ok with making phpbb readonly and _not_ importing the history since it will be easily searchable if necessary. We'd need to figure out how to handle active discussions during the transition but that should be possible.

(I can imagine RSS may be a loss @cwalther - note that it is an open discussion: github-community/community/discussions/31)

So yes, +1, let's make it happen!

mcauser, 2022-06-22:
+1 for Github Discussions from me

Side note: how easy is it to get data back out of Github Discussions if a better tool comes along?

dpgeorge, 2022-06-22:
> Would this move load the maintainers with the task of moderating the discussions?

If we put the GitHub Discussions in a new repo (eg micropython/micropython-discuss) then we can give write permission on that repo to the existing forum moderators, so they can modify/remove content.  Then link that Discussions to the micropython organization so that it covers all the repositories.

alphaFred, 2022-06-22:
If we move than I would vote for GitHub Discussions rather than something like Discord. I believe having the actual code, issues, PRs, and forum so close together will have benefits. 

I am not sure if this is possible but I would really like to see something similar to the  “Boards running MicroPython” sub forums.  For contribution purposes it would however be better if there could be a sub forum on a port rather than a board level.

andypiper, 2022-06-22:
I've worked with several different discussion platforms as part of my DevRel work, including content migrations. I've specifically supported a large high-volume Discourse community day-to-day for ~8 years. The past few years have seen a lot of projects move to Discourse and I'm pretty familiar with that - it's effective, and the trust levels can help with moderation (although can also be a barrier to entry sometimes). GH Discussions are a more recent entrant to the space but growing in popularity of course since a lot of activity happens alongside in Issues, Projects etc.

I really think it's time to move on from phpBB. As a relative newcomer to this community, I've engaged with the forums but I find it very difficult to see, at a glance, the most recent new topics or updates - I have to actively click into the categories. I've flirted with an RSS reader to keep up, but I just find myself not using the forums as much as I'd like. I imagine there are probably plugins to improve UX, but by default I don't find it great. Also phpBB markup / bbcode is something I constantly have to go lookup, but that's likely a familiarity thing now I mostly work in GfM... 

In general I'd support either Discourse or GH Discussions as a new home, and I'd be happy to help with a migration to either. I lightly lean towards GH Discussions since almost all of the project appears to be centered here, barring the website, docs, and wiki (and the latter lives in 2 or more places...)

One more thing to note is that there's absolutely a difference between forums, and chat - there exist multiple live chat platforms (Slack, Discord, IRC) but those conversations aren't kept in a searchable manner.

**Considerations / decision points**

- lifetime of the platform
  - per @dpgeorge comment in the meetup, ideally we want something that will last 10 years (which is more-or-less the life phpBB has served)
- how to manage history
  - can do an export/import for Discourse
  - could store an archive in GH, or leave the phpBB in place read-only (although then there's a challenge to encourage people to refer back / search for previous discussions in Yet Another Place)
- RSS / external access
  - Discourse ✅ and there's also an app
  - GH ❌ but there's also their app
- backup / future migration 
  - Discourse has export capability ✅ 
  - GH ❓ 
- moderation and access
  - Discourse has built-in education / get started, trust levels, and fine-grained roles, and it also enables you to set up different methods of sign-in outside of requiring a GH account
  - I'm yet to dig deeply on how much [control](https://docs.github.com/en/discussions/managing-discussions-for-your-community/moderating-discussions) GH Discussions provide, but I've seen a couple of projects deal with problematic discussion contributors so I know it's possible. I strongly agree with @robert-hh that this is not something that should fall on the core maintainers. 

A final comment in favour of GH Discussions is that it would enable us to convert issues into discussions when appropriate, which may help with chipping away at the large number of existing issues.

On @dpgeorge comment about a new org-level repo - GitHub [does support this](https://docs.github.com/en/discussions/quickstart#enabling-github-discussions-on-your-organization). 

I'll go check the conversation on this same topic on phpBB as well 🙂 please count me in as someone to help with this work as needed.

stinos, 2022-06-22:
> I've engaged with the forums but I find it very difficult to see, at a glance, the most recent new topics or updates

Does https://forum.micropython.org/search.php?search_id=newposts not work for you? Ok it doesn't order by category, but still pretty convenient. And I hope github discussions has something like this as well.

dpgeorge, 2022-06-23:
It looks like GH Discussions sorts by default on most recent activity first: https://github.com/home-assistant/frontend/discussions?discussions_q=

You can also add filters, eg activity since 2022/6/15 sorted by most recent activity first: https://github.com/home-assistant/frontend/discussions?discussions_q=created%3A%3E%3D2022-06-15+

IhorNehrutsa, 2022-06-28:
+1 for GitHub Discussions

dpgeorge, 2022-08-25:
OK, let's do it, let's move to GitHub Discussions!

See https://github.com/orgs/micropython/discussions

I've set it up as a discussion on this main repository, so that it's more discoverable (it appears as a tab at the top of this page).  For moderators, they can have "triage" permissions on this repository which should be enough to be able to moderate the discussions.  I'll add @peterhinch , @dhylands  and @robert-hh to this "triage" team.

closed 2022-08-25 by dpgeorge

jimmo, 2022-08-25:
PSA: If anyone wants to filter out the discussions from issue/PR emails, there's unfortunately very little difference in the headers. It seems that the best option is to match the "repo-discussions" part of the message-id.

In GMail, you can't match on a partial message-id, however it seems to work to use `listid:micropython.micropython.github.com "repo-discussions"`.

i.e. I have
`listid:micropython.micropython.github.com -"repo-discussions"` --> archive, label `micropython`
`listid:micropython.micropython.github.com "repo-discussions"` --> archive, label `micropython/discussions`

Hopefully GitHub improve this... although see https://github.com/community/community/discussions/3021

tve, 2022-11-25:
I'm obviously late to the party here, but I'm sad about the move to GH discussions. I have yet to run into a GH discussion forum where I am compelled to engage on a persistent manner. There are a number of discourse forums where I am active on an ongoing basis. I find the Gh discussion UI pretty awful and the way it works not conducive to forming sub-communities (e.g. folks interested in esp32, or rp2040, etc).

In particular, I find it very difficult to get a picture of what's going on. In a good forum there are categories and sub-categories, so for example I can go to the esp32 sub-category and quickly scroll through to see what has happened in the last couple of days/weeks. I know GH has tags but my experience is that tags proliferate and posts become inconsistent  in their tag usage. So filtering for the esp32 tag is not the same as going to an esp32 sub-category.

Gh discussions is also not great for seeing what's new. I read about jimmo's method of using email, but my email is an even bigger disaster with way too many notifications from all kinds of stuff. I definitely do not want to add anything more to it!

In the end, a forum is about community and the tool needs to favor the creation of community and sub-community. I don't find that GH does that. IMHO Discourse is very good at that. If you have well-thought sub-categories you can get little communities of people who are interested in that category having lots of interesting discussions. I believe this happened for the various ports in the old forum too. It seems to be happening in discord too thanks to the hardware&ports channels. GH discussions is primarily a giant haystack of posts and finding gems in there is left as an exercise to the reader with some drop-downs for labels and filters. At least this is how it feels to me.

I also find the presentation on GH awful. I have a big monitor and the web browser full height yet, due to all the headers and side-bar I see only 11 posts, WFT? Makes it difficult to take a glance.

[image]

beyonlo, 2022-11-26:
I agree with @tve 

>I find the Gh discussion UI pretty awful and the way it works not conducive to forming sub-communities (e.g. folks interested in esp32, or rp2040, etc).

Exactly!

> In particular, I find it very difficult to get a picture of what's going on.

My feeling is the same.

 >In a good forum there are categories and sub-categories, so for example I can go to the esp32 sub-category and quickly scroll through to see what has happened in the last couple of days/weeks.

I agree 💯. That is the point. We simply lost that. That is what we need in my opinion.

> Gh discussions is also not great for seeing what's new. I read about jimmo's method of using email, but my email is an even bigger disaster with way too many notifications from all kinds of stuff.

I agree, forum can't use email as organization of categories method.

> If you have well-thought sub-categories you can get little communities of people who are interested in that category having lots of interesting discussions. 

Exactly. I think you told everything what mostly people missing on the GH discussion.

To have everything on GH, keeping all things in just one place is very good, integration with issues and PR and so on - excellent. But we can't to lost about categories and subcategories, where we can just go to that categories and/or subcategories and see what's going on, what's new. And to create a strong community sharing information for that same interest for that category and/or subcategory. I never could imagine that with the power of GH discussion, it can't to have that feature to have categories and subcategories like as already exist in the forum.micropython.org.  I'm really sad for that as well.

People that manage/admin the GH discussion, could you please to check if really is not possible do configure the GH discussion to works in that way?

Thank you very!

jimmo, 2022-11-26:
@tve @beyonlo  Thanks for the feedback.

When we first moved to Discussions, there was a limit of 10 categories so I didn't create board-specific categories. This has subsequently been increased to 25. If I understand both of your feedback, having more, focused categories (e.g. "ESP32") would help a lot. I'm more than happy to add more categories (a few people have mentioned this now) so I guess we'll start with one for each of the main ports.

I also understand that a "subscribe to category" feature is in the works, as well as a way to group categories into sub-categories. You can see on the "GitHub Community" (i.e. run by GitHub) discussions page, they already have the sub-category feature enabled -- https://github.com/community/community/discussions/categories/discussions

@tve I think you mentioned the "Welcome" banner also on Discord. This is actually supposed to be the pinned post (it makes more sense when there is more than one!). Hopefully soon there'll be a v1.20 release announcement to pin next to it :). I agree that it does take a lot of vertical space though. I've already sent feedback about the "organisation discussions" banner too.

If you have UX feedback for discussions, then the GitHub team are quite responsive to feedback on their discussions page linked above.

beyonlo, 2022-11-26:
@jimmo If you even no have the MicroPython ports as categories, to have specific boards as categories will me a mess. So, firstly, to create all MicroPython ports as categories already will be the amazing. 

First Ports Categories: [cc3200](https://micropython.org/download/?port=cc3200), [esp32](https://micropython.org/download/?port=esp32), [esp8266](https://micropython.org/download/?port=esp8266), [mimxrt](https://micropython.org/download/?port=mimxrt), [nrf](https://micropython.org/download/?port=nrf), [renesas-ra](https://micropython.org/download/?port=renesas-ra), [rp2](https://micropython.org/download/?port=rp2), [samd](https://micropython.org/download/?port=samd), [stm32](https://micropython.org/download/?port=stm32)

Inside each MicroPython ports above (that will be categories), as soon the sub-category works on GH discussions, you can to create the specific boards inside each MicroPython port.

But, for now (while subcategories still not working for all), more important than specific board, is to have firstly the all MicroPython ports as categories.

mattytrentini, 2022-11-27:
It's worth mentioning that this conversation [also continued on discord](https://discord.com/channels/574275045187125269/574275045611012097/1045771309210148956). Please try to read the material there before contributing here..

IhorNehrutsa, 2022-11-28:
-1 to Discord. They don't know Ukraine.
[image]
Empty!!!

jimmo, 2022-12-13:
@tve @beyonlo I have added categories for each of the major ports. See announcement https://github.com/orgs/micropython/discussions/10189

It's a long list now, but hopefully we can group them at some point in the future. Looks like people have already started using the ESP32 category: https://github.com/micropython/micropython/discussions/categories/ports-esp32

> @jimmo I think that If you even no have the MicroPython ports as categories, to have specific boards as categories will me a mess - because there are so many boards for each MicroPython port.

@beyonlo Yep, of course I meant port-specific categories not board-specific (!!!), this was a typo (I said ports later in the comment). This is exactly what is now implemented.

beyonlo, 2022-12-13:
@jimmo Excellent work. Thank you very much!