micropython/micropython #8804: Raspberry Pi PICO with Micropython and ESP01 gives socket Error for NTP

sksingh20, 2022-06-22:
**Setup:** Raspberry pi Pico is connected with ESP01 module. Thonny is used for connecting Pico and using micropython. 
Pico is connected with ESP01 module to connect with Wifi
20*4 LCD is connected via I2C interface to PICO

**Micropython:** MicroPython v1.19.1 on 2022-06-18; Raspberry Pi Pico with RP2040

**Objective:** Get time from internet and show on LCD

**Code:**
try:
    import usocket as socket
except:
    import socket
try:
    import ustruct as struct
except:
    import struct

# (date(2000, 1, 1) - date(1900, 1, 1)).days * 24*60*60
NTP_DELTA = 3155673600

# The NTP host can be configured at runtime by doing: ntptime.host = 'myhost.org'
host = "pool.ntp.org"

def time():
    NTP_QUERY = bytearray(48)
    NTP_QUERY[0] = 0x1B
    addr = socket.getaddrinfo(host, 123)[0][-1]
    s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
    try:
        s.settimeout(1)
        res = s.sendto(NTP_QUERY, addr)
        msg = s.recv(48)
    finally:
        s.close()
    val = struct.unpack("!I", msg[40:44])[0]
    return val - NTP_DELTA

# There's currently no timezone support in MicroPython, and the RTC is set in UTC time.
def settime():
    t = time()
    import machine
    import utime

    tm = utime.gmtime(t)
    machine.RTC().datetime((tm[0], tm[1], tm[2], tm[6] + 1, tm[3], tm[4], tm[5], 0))

**Error:**
Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
ImportError: no module named 'socket'

Tried downloading socket.py from various sources but all gives some or other error.

dpgeorge, 2022-06-23:
There is currently no support for PICO + ESP01.  That's why there is no `socket` module available for it.

closed 2022-06-27 by dpgeorge

sksingh20, 2023-05-20:
Do we have any update on this.. if ESP01 with PICO is now supported? Or in Roadmap? or any alternate method with have NTP time sync with PICO+ESP01?

robert-hh, 2023-05-20:
@sksingh20 Unless someone implements WiFi support for the PICO using an ESP01 module, **it will not exist**. You could try using the AT commands to set up a connection and get the date. You may search in the discussions section or the previous MicroPython forum at https://forum.micropython.org/ for PPP or PPPoS.

There are already two PICO boards with WiFi support, the `PICO W` and the `Arduino Nano RP2040 Connect`. So you could use one of these boards. Or just a plain ESP32.

sksingh20, 2023-05-21:
@robert-hh Thank you for response. While higher version of micro controllers have wifi support... PICO version is cost effective microcontroller. while I am able to have esp01 connected with Pico and webserver is running from PICO... Trying to have some cheap solution to have time from internet once during boot and update RTC of PICO... which is performed itself when PICO connected via USB.

any help will be highly appreciable..