micropython/micropython #8805: port/esp32 stops at objgenerator.c

Damon3499, 2022-06-22:
I try to build a Firmware with LV_micropython on Mac OS M1
/ports/esp32 make
I get an error. Do you have any idea? 

Bootloader binary size 0x5460 bytes. 0x1ba0 bytes (25%) free.
[3/416] Generating /Users/Hobby/esp/lv_micropython/mpy-cross/mpy-cross
FAILED: /Users/Hobby/esp/lv_micropython/mpy-cross/mpy-cross 
cd /Users/Hobby/esp/lv_micropython/ports/esp32/build-GENERIC/esp-idf/main && make -C /Users/Hobby/esp/lv_micropython/mpy-cross
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
LVGL-GEN build/lvgl/lv_mpy.c
cp ../lib/lv_bindings/driver/png/lodepng/lodepng.cpp build/lodepng/lodepng.c
LODEPNG-GEN build/lodepng/mp_lodepng.c
mkdir -p build/genhdr
GEN build/genhdr/mpversion.h
GEN build/genhdr/moduledefs.h
GEN build/genhdr/qstr.i.last
GEN build/genhdr/qstr.split
GEN build/genhdr/qstrdefs.collected.h
QSTR updated
GEN build/genhdr/qstrdefs.generated.h
mkdir -p build/py
mkdir -p build/shared/runtime
CC ../py/mpstate.c
CC ../py/nlr.c
CC ../py/nlrx86.c
CC ../py/nlrx64.c
CC ../py/nlrthumb.c
CC ../py/nlraarch64.c
CC ../py/nlrpowerpc.c
CC ../py/nlrxtensa.c
CC ../py/nlrsetjmp.c
CC ../py/malloc.c
CC ../py/gc.c
CC ../py/pystack.c
CC ../py/qstr.c
CC ../py/vstr.c
CC ../py/mpprint.c
CC ../py/unicode.c
CC ../py/mpz.c
CC ../py/reader.c
CC ../py/lexer.c
CC ../py/parse.c
CC ../py/scope.c
CC ../py/compile.c
CC ../py/emitcommon.c
CC ../py/emitbc.c
CC ../py/asmbase.c
CC ../py/asmx64.c
CC ../py/emitnx64.c
CC ../py/asmx86.c
CC ../py/emitnx86.c
CC ../py/asmthumb.c
CC ../py/emitnthumb.c
CC ../py/emitinlinethumb.c
CC ../py/asmarm.c
CC ../py/emitnarm.c
CC ../py/asmxtensa.c
CC ../py/emitnxtensa.c
CC ../py/emitinlinextensa.c
CC ../py/emitnxtensawin.c
CC ../py/formatfloat.c
CC ../py/parsenumbase.c
CC ../py/parsenum.c
CC ../py/emitglue.c
CC ../py/persistentcode.c
CC ../py/runtime.c
CC ../py/runtime_utils.c
CC ../py/scheduler.c
CC ../py/nativeglue.c
CC ../py/pairheap.c
CC ../py/ringbuf.c
CC ../py/stackctrl.c
CC ../py/argcheck.c
CC ../py/warning.c
CC ../py/profile.c
CC ../py/map.c
CC ../py/obj.c
CC ../py/objarray.c
CC ../py/objattrtuple.c
CC ../py/objbool.c
CC ../py/objboundmeth.c
CC ../py/objcell.c
CC ../py/objclosure.c
CC ../py/objcomplex.c
CC ../py/objdeque.c
CC ../py/objdict.c
CC ../py/objenumerate.c
CC ../py/objexcept.c
CC ../py/objfilter.c
CC ../py/objfloat.c
CC ../py/objfun.c
CC ../py/objgenerator.c
../py/objgenerator.c:102:21: error: variable 'n_exc_stack_unused' set but not used [-Werror,-Wunused-but-set-variable]
    size_t n_state, n_exc_stack_unused, scope_flags, n_pos_args, n_kwonly_args, n_def_args;
                    ^
../py/objgenerator.c:102:81: error: variable 'n_def_args' set but not used [-Werror,-Wunused-but-set-variable]
    size_t n_state, n_exc_stack_unused, scope_flags, n_pos_args, n_kwonly_args, n_def_args;
                                                                                ^
../py/objgenerator.c:102:66: error: variable 'n_kwonly_args' set but not used [-Werror,-Wunused-but-set-variable]
    size_t n_state, n_exc_stack_unused, scope_flags, n_pos_args, n_kwonly_args, n_def_args;
                                                                 ^
../py/objgenerator.c:102:54: error: variable 'n_pos_args' set but not used [-Werror,-Wunused-but-set-variable]
    size_t n_state, n_exc_stack_unused, scope_flags, n_pos_args, n_kwonly_args, n_def_args;
                                                     ^
../py/objgenerator.c:102:41: error: variable 'scope_flags' set but not used [-Werror,-Wunused-but-set-variable]
    size_t n_state, n_exc_stack_unused, scope_flags, n_pos_args, n_kwonly_args, n_def_args;
                                        ^
5 errors generated.
make[1]: *** [build/py/objgenerator.o] Error 1
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
make: *** [all] Error 2

dpgeorge, 2022-06-23:
What version of MicroPython are you using?  This was fixed a while ago in cfd08448a1cba0040ad3c62bce02b5f8bd76dd1d

Damon3499, 2022-06-23:
I got this repo lv_micropython downloaded yesterday. I use IDF 4.4 I don't switch the mircopython version.  I figured out that the just Micropython from the Micropython version 1.19 build with no errors. I checked the lines in the fix you sent me, that's okay. I don't need to change it. Maybe this could help, when I call make  V=1 -C mpy-cross. 

Hobby@Mac-mini-von-Dirk lv_micropython % make V=1 -C mpy-cross
LVGL-GEN build/lvgl/lv_mpy.c
mkdir -p build/lvgl/
clang -E   -Wno-unused-function -DLODEPNG_NO_COMPILE_ENCODER -DLODEPNG_NO_COMPILE_DISK -DLODEPNG_NO_COMPILE_ALLOCATORS -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DPYCPARSER -x c -I ../lib/lv_bindings/pycparser/utils/fake_libc_include -I../lib/lv_bindings -I../lib/lv_bindings/driver/png/lodepng -I. -Ibuild -I.. ../lib/lv_bindings/lvgl/lvgl.h > build/lvgl/lvgl.pp.c
python3 ../lib/lv_bindings/gen/gen_mpy.py -M lvgl -MP lv -MD build/lvgl/lv_mpy.json -E build/lvgl/lvgl.pp.c ../lib/lv_bindings/lvgl/lvgl.h > build/lvgl/lv_mpy.c
mkdir -p build/lodepng/
cp ../lib/lv_bindings/driver/png/lodepng/lodepng.cpp build/lodepng/lodepng.c
LODEPNG-GEN build/lodepng/mp_lodepng.c
mkdir -p build/lodepng/
clang -E -DLODEPNG_NO_COMPILE_ENCODER -DLODEPNG_NO_COMPILE_DISK -DLODEPNG_NO_COMPILE_ALLOCATORS -DPYCPARSER -x c -I../lib/lv_bindings -I../lib/lv_bindings/driver/png/lodepng -I. -Ibuild -I.. -I ../lib/lv_bindings/pycparser/utils/fake_libc_include ../lib/lv_bindings/driver/png/lodepng/lodepng.h > build/lodepng/lodepng.pp.c
python3 ../lib/lv_bindings/gen/gen_mpy.py -M lodepng -E build/lodepng/lodepng.pp.c ../lib/lv_bindings/driver/png/lodepng/lodepng.h > build/lodepng/mp_lodepng.c
mkdir -p build/genhdr
python3 ../py/makeversionhdr.py build/genhdr/mpversion.h
GEN build/genhdr/mpversion.h
GEN build/genhdr/moduledefs.h
python3 ../py/makemoduledefs.py --vpath="., .., " lib/lv_bindings/lvgl/src/misc/lv_utils.c lib/lv_bindings/lvgl/src/misc/lv_txt.c lib/lv_bindings/lvgl/src/misc/lv_anim_timeline.c lib/lv_bindings/lvgl/src/misc/lv_style.c lib/lv_bindings/lvgl/src/misc/lv_style_gen.c lib/lv_bindings/lvgl/src/misc/lv_mem.c lib/lv_bindings/lvgl/src/misc/lv_fs.c lib/lv_bindings/lvgl/src/misc/lv_txt_ap.c lib/lv_bindings/lvgl/src/misc/lv_color.c lib/lv_bindings/lvgl/src/misc/lv_ll.c lib/lv_bindings/lvgl/src/misc/lv_log.c lib/lv_bindings/lvgl/src/misc/lv_area.c lib/lv_bindings/lvgl/src/misc/lv_lru.c lib/lv_bindings/lvgl/src/misc/lv_timer.c lib/lv_bindings/lvgl/src/misc/lv_async.c lib/lv_bindings/lvgl/src/misc/lv_templ.c lib/lv_bindings/lvgl/src/misc/lv_printf.c lib/lv_bindings/lvgl/src/misc/lv_tlsf.c lib/lv_bindings/lvgl/src/misc/lv_math.c lib/lv_bindings/lvgl/src/misc/lv_anim.c lib/lv_bindings/lvgl/src/misc/lv_gc.c lib/lv_bindings/lvgl/src/misc/lv_bidi.c lib/lv_bindings/lvgl/src/core/lv_indev.c lib/lv_bindings/lvgl/src/core/lv_theme.c lib/lv_bindings/lvgl/src/core/lv_disp.c lib/lv_bindings/lvgl/src/core/lv_obj_class.c lib/lv_bindings/lvgl/src/core/lv_refr.c lib/lv_bindings/lvgl/src/core/lv_event.c lib/lv_bindings/lvgl/src/core/lv_obj_draw.c lib/lv_bindings/lvgl/src/core/lv_obj.c lib/lv_bindings/lvgl/src/core/lv_obj_style_gen.c lib/lv_bindings/lvgl/src/core/lv_obj_style.c lib/lv_bindings/lvgl/src/core/lv_group.c lib/lv_bindings/lvgl/src/core/lv_indev_scroll.c lib/lv_bindings/lvgl/src/core/lv_obj_pos.c lib/lv_bindings/lvgl/src/core/lv_obj_tree.c lib/lv_bindings/lvgl/src/core/lv_obj_scroll.c lib/lv_bindings/lvgl/src/hal/lv_hal_tick.c lib/lv_bindings/lvgl/src/hal/lv_hal_indev.c lib/lv_bindings/lvgl/src/hal/lv_hal_disp.c lib/lv_bindings/lvgl/src/extra/others/snapshot/lv_snapshot.c lib/lv_bindings/lvgl/src/extra/others/msg/lv_msg.c lib/lv_bindings/lvgl/src/extra/others/fragment/lv_fragment.c lib/lv_bindings/lvgl/src/extra/others/fragment/lv_fragment_manager.c lib/lv_bindings/lvgl/src/extra/others/gridnav/lv_gridnav.c lib/lv_bindings/lvgl/src/extra/others/monkey/lv_monkey.c lib/lv_bindings/lvgl/src/extra/others/imgfont/lv_imgfont.c lib/lv_bindings/lvgl/src/extra/libs/ffmpeg/lv_ffmpeg.c lib/lv_bindings/lvgl/src/extra/libs/sjpg/tjpgd.c lib/lv_bindings/lvgl/src/extra/libs/sjpg/lv_sjpg.c lib/lv_bindings/lvgl/src/extra/libs/freetype/lv_freetype.c lib/lv_bindings/lvgl/src/extra/libs/png/lv_png.c lib/lv_bindings/lvgl/src/extra/libs/png/lodepng.c lib/lv_bindings/lvgl/src/extra/libs/qrcode/lv_qrcode.c lib/lv_bindings/lvgl/src/extra/libs/qrcode/qrcodegen.c lib/lv_bindings/lvgl/src/extra/libs/rlottie/lv_rlottie.c lib/lv_bindings/lvgl/src/extra/libs/fsdrv/lv_fs_stdio.c lib/lv_bindings/lvgl/src/extra/libs/fsdrv/lv_fs_posix.c lib/lv_bindings/lvgl/src/extra/libs/fsdrv/lv_fs_fatfs.c lib/lv_bindings/lvgl/src/extra/libs/fsdrv/lv_fs_win32.c lib/lv_bindings/lvgl/src/extra/libs/bmp/lv_bmp.c lib/lv_bindings/lvgl/src/extra/libs/gif/lv_gif.c lib/lv_bindings/lvgl/src/extra/libs/gif/gifdec.c lib/lv_bindings/lvgl/src/extra/layouts/flex/lv_flex.c lib/lv_bindings/lvgl/src/extra/layouts/grid/lv_grid.c lib/lv_bindings/lvgl/src/extra/lv_extra.c lib/lv_bindings/lvgl/src/extra/themes/basic/lv_theme_basic.c lib/lv_bindings/lvgl/src/extra/themes/default/lv_theme_default.c lib/lv_bindings/lvgl/src/extra/themes/mono/lv_theme_mono.c lib/lv_bindings/lvgl/src/extra/widgets/calendar/lv_calendar_header_dropdown.c lib/lv_bindings/lvgl/src/extra/widgets/calendar/lv_calendar_header_arrow.c lib/lv_bindings/lvgl/src/extra/widgets/calendar/lv_calendar.c lib/lv_bindings/lvgl/src/extra/widgets/led/lv_led.c lib/lv_bindings/lvgl/src/extra/widgets/win/lv_win.c lib/lv_bindings/lvgl/src/extra/widgets/chart/lv_chart.c lib/lv_bindings/lvgl/src/extra/w [... cut ...]
../py/objgenerator.c:102:54: error: variable 'n_pos_args' set but not used [-Werror,-Wunused-but-set-variable]
    size_t n_state, n_exc_stack_unused, scope_flags, n_pos_args, n_kwonly_args, n_def_args;
                                                     ^
../py/objgenerator.c:102:41: error: variable 'scope_flags' set but not used [-Werror,-Wunused-but-set-variable]
    size_t n_state, n_exc_stack_unused, scope_flags, n_pos_args, n_kwonly_args, n_def_args;
                                        ^
5 errors generated.
make: *** [build/py/objgenerator.o] Error 1
Hobby@Mac-mini-von-Dirk lv_micropython %

dpgeorge, 2022-06-23:
It looks like this is a problem with lv_micropython so you'll need to report the error to that project.

closed 2022-06-23 by dpgeorge