micropython/micropython #8810: ESP32-S2 Guru Meditation when providing value argument to machine.Pin() constructor

ThinkTransit, 2022-06-24:
**ESP32-S2 SPI FLASH + SPI RAM idf v4.3.2** 

```
MicroPython v1.19.1-45-gba21f76f8 on 2022-06-24; THINK_TRANSIT with ESP32-S2
Type "help()" for more information.
```

**boot.py**
```
from machine import Pin

mypin = Pin(45, Pin.OUT, value=1)
```

**Decoded backtrace** 
```
PC: 0x400c9e95: gpio_od_disable at /home/patrick/esp/esp-idf/components/driver/gpio.c line 212
EXCVADDR: 0x00000000

Decoding stack results
0x400c9e92: gpio_od_disable at /home/patrick/esp/esp-idf/components/driver/gpio.c line 212
0x400ca6c4: gpio_set_direction at /home/patrick/esp/esp-idf/components/driver/gpio.c line 298
0x40086d95: machine_pin_obj_init_helper at /home/patrick/esp/micropython/ports/esp32/machine_pin.c line 321
0x40086e7d: mp_pin_make_new at /home/patrick/esp/micropython/ports/esp32/machine_pin.c line 373
0x4009219e: type_call at /home/patrick/esp/micropython/py/objtype.c line 997
0x40093da1: mp_call_function_n_kw at /home/patrick/esp/micropython/py/runtime.c line 684
0x4002835e: mp_execute_bytecode at /home/patrick/esp/micropython/py/vm.c line 941
0x4008d2b4: fun_bc_call at /home/patrick/esp/micropython/py/objfun.c line 288
0x40093da1: mp_call_function_n_kw at /home/patrick/esp/micropython/py/runtime.c line 684
0x40093eb5: mp_call_method_n_kw at /home/patrick/esp/micropython/py/runtime.c line 700
0x400283f1: mp_execute_bytecode at /home/patrick/esp/micropython/py/vm.c line 1026
0x4008d2b4: fun_bc_call at /home/patrick/esp/micropython/py/objfun.c line 288
0x40093da1: mp_call_function_n_kw at /home/patrick/esp/micropython/py/runtime.c line 684
0x40093dca: mp_call_function_0 at /home/patrick/esp/micropython/py/runtime.c line 658
0x4009ebca: parse_compile_execute at /home/patrick/esp/micropython/shared/runtime/pyexec.c line 120
0x4009ef71: pyexec_file_if_exists at /home/patrick/esp/micropython/shared/runtime/pyexec.c line 693
0x40086401: mp_task at /home/patrick/esp/micropython/ports/esp32/main.c line 165
```

**Working code**
```
from machine import Pin

mypin = Pin(45, Pin.OUT)
mypin.value(1)
```

dpgeorge, 2022-06-24:
It looks like you're using a custom board.  Maybe that doesn't allow pin 43 to be used as an output?

Also, IDF at v4.3.2 doesn't seem to match with line 212 being in `gpio_od_disable`... I'd suggest trying with the latest IDF v4.3.3.  Also double check that pin 45 on esp32-s2 can actually be used as an output.

ThinkTransit, 2022-06-25:
> It looks like you're using a custom board. Maybe that doesn't allow pin 45 to be used as an output?
> 
Yes that is correct it is my own board, GPIO45 is physically available and according to ESP32-S2 datasheet it is a digital I/O

> Also, IDF at v4.3.2 doesn't seem to match with line 212 being in `gpio_od_disable`... I'd suggest trying with the latest IDF v4.3.3. Also double check that pin 45 on esp32-s2 can actually be used as an output.

Same result in v4.3.3. The strange part is that it only happens after awaking from deepsleep or on a cold start. After the chip reboots from the first crash it will work ok

```
MPY: soft reboot
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40096aac: df5e36d0 dfa1d0f2 02106403
Core  0 register dump:
PC      : 0x40096ab3  PS      : 0x00060630  A0      : 0x8002808c  A1      : 0x3ffd6a20  
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x3ffd6af4  A5      : 0x3f7af3e3  
A6      : 0x3f7af3da  A7      : 0x3ffd6a10  A8      : 0x80096ab0  A9      : 0x3ffd6a00  
A10     : 0x00000006  A11     : 0x00000002  A12     : 0x00000000  A13     : 0x3ffd6af8  
A14     : 0x3f001ffc  A15     : 0x8002218c  SAR     : 0x00000013  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x3ffd6af8  LCOUNT  : 0x40026401  

Backtrace:0x40096ab0:0x3ffd6a20 0x40028089:0x3ffd6a40 0x4009003c:0x3ffd6ae0 0x40096999:0x3ffd6b30 0x400969c2:0x3ffd6b50 0x400a1666:0x3ffd6b70 0x400a1a0d:0x3ffd6c00 0x40089385:0x3ffd6c20

ELF file SHA256: 86160c207fb97166

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40026102
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x1030
load:0x4004c000,len:0xacc
load:0x40050000,len:0x2c80
entry 0x4004c1ac
MicroPython v1.19.1-45-gba21f76f8 on 2022-06-25; THINK_TRANSIT with ESP32-S2
Type "help()" for more information.
>>> mypin
Pin(45)
>>>
```

```
PC: 0x40096ab3: mp_call_method_n_kw at /home/patrick/esp/micropython/py/runtime.c line 700
EXCVADDR: 0x00000000

Decoding stack results
0x40096ab0: mp_call_method_n_kw at /home/patrick/esp/micropython/py/runtime.c line 700
0x40028089: mp_execute_bytecode at /home/patrick/esp/micropython/py/vm.c line 1026
0x4009003c: fun_bc_call at /home/patrick/esp/micropython/py/objfun.c line 288
0x40096999: mp_call_function_n_kw at /home/patrick/esp/micropython/py/runtime.c line 684
0x400969c2: mp_call_function_0 at /home/patrick/esp/micropython/py/runtime.c line 658
0x400a1666: parse_compile_execute at /home/patrick/esp/micropython/shared/runtime/pyexec.c line 120
0x400a1a0d: pyexec_file_if_exists at /home/patrick/esp/micropython/shared/runtime/pyexec.c line 693
0x40089385: mp_task at /home/patrick/esp/micropython/ports/esp32/main.c line 165
```

Different backtrace in v4.3.3

ThinkTransit, 2022-06-29:
This appears to be a hardware issue as I can't reproduce it on other ESP32-S2 boards.

Apologies for the false report.

closed 2022-06-29 by dpgeorge