micropython/micropython #8818: docs: Broken Link

mwerezak, 2022-06-25:
* Describe what was missing from the documentation and/or what was incorrect/incomplete.

In file:
https://github.com/micropython/micropython/blob/master/docs/reference/packages.rst#creating-distribution-packages

> The sdist_upip.py module as referenced above can be found in [micropython-lib](https://docs.micropython.org/en/latest/reference/glossary.html#term-micropython-lib): https://github.com/micropython/micropython-lib/blob/master/sdist_upip.py

The link is broken

* If possible, please link to the relevant page on https://docs.micropython.org/

https://docs.micropython.org/en/latest/reference/packages.html#creating-distribution-packages

andypiper, 2022-06-30:
Duplicate of #7908

jonnor, 2023-07-30:
Duplicate. Proposing to close

closed 2023-07-31 by mattytrentini