micropython/micropython #8819: ESP32 WPA2 Enterprise

MATTYGILO, 2022-06-25:
Micropython would feel more complete with WPA Enterprise for esp32. This is an essential function that is not yet met.

All help and suggestions would be great. If there are way around this for now.

MATTYGILO, 2022-06-25:
Been requested since 2019:

[https://forum.micropython.org/viewtopic.php?f=18&t=7219](url)
[https://forum.micropython.org/viewtopic.php?t=10543](url)

SL2021-Dev, 2023-09-18:
I am also in need of an enterprise login capabilities.
Arduino, Pycom (I know... GONE)  and CircuitPython have this ability.
v1.21 is about to come out, PLEASE INCLUDE
It has been 2 years 4 months since...
[image]

MATTYGILO, 2023-09-18:
@SL2021-Dev 

Step 1) 
Fork micropython repo
Add wpa_supplicant to set(IDF_COMPONENTS …) in /micropython/ports/esp32/main/CMakeLists.txt

Step 2)
Create a custom micropython module (Provided) and reference it with your build

enterprise.c:
```
#include <string.h>

#include "py/builtin.h"
#include "py/runtime.h"
#include "py/mpconfig.h"
#include "py/objstr.h"
#include "py/obj.h"
#include "py/stream.h"
#include "py/mperrno.h"

#include "esp_wifi.h"
#include "esp_wpa2.h"
#include "esp_log.h"
#include "esp_event.h"

#include "modnetwork.h"

// Set up EAP
STATIC mp_obj_t esp_seteap(mp_obj_t username, mp_obj_t password){

    // The length of username and password
    size_t Ilen;
    size_t Plen;

    // Converts from micropython form to C form
    const char *EAP_IDENTITY = mp_obj_str_get_data(username,&Ilen);
    const char *EAP_PASSWORD = mp_obj_str_get_data(password,&Plen);

    // Sets the identity and password
    esp_exceptions(esp_wifi_sta_wpa2_ent_set_identity((uint8_t *)EAP_IDENTITY, strlen(EAP_IDENTITY)));
    esp_exceptions(esp_wifi_sta_wpa2_ent_set_username((uint8_t *)EAP_IDENTITY, strlen(EAP_IDENTITY)));
    // esp_exceptions(esp_wifi_sta_wpa2_ent_set_new_password((uint8_t *)EAP_PASSWORD, strlen(EAP_PASSWORD)));
    esp_exceptions(esp_wifi_sta_wpa2_ent_set_password((uint8_t *)EAP_PASSWORD, strlen(EAP_PASSWORD)));

    // Enables wpa enterprise
    esp_wifi_sta_wpa2_ent_enable();

    return mp_const_none;
}

MP_DEFINE_CONST_FUN_OBJ_2(esp_seteap_obj, esp_seteap);

// Set channel
STATIC mp_obj_t esp_setchannel(mp_obj_t channel){

    // Converts from micropython form to C form
    int channel_num = mp_obj_get_int(channel);

    // Sets the channel
    esp_exceptions(esp_wifi_set_channel(channel_num, WIFI_SECOND_CHAN_NONE));

    return mp_const_none;
}

MP_DEFINE_CONST_FUN_OBJ_1(esp_setchannel_obj, esp_setchannel);

STATIC const mp_rom_map_elem_t mp_module_enterprise_globals_table[] = {
    { MP_ROM_QSTR(MP_QSTR___name__), MP_ROM_QSTR(MP_QSTR_enterprise) },
    { MP_ROM_QSTR(MP_QSTR_seteap), MP_ROM_PTR(&esp_seteap_obj) },
    { MP_ROM_QSTR(MP_QSTR_setchannel), MP_ROM_PTR(&esp_setchannel_obj) },
};

STATIC MP_DEFINE_CONST_DICT(mp_module_enterprise_globals, mp_module_enterprise_globals_table);

const mp_obj_module_t mp_module_enterprise = {
        .base = { &mp_type_module },
        .globals = (mp_obj_dict_t *)&mp_module_enterprise_globals,
};

MP_REGISTER_MODULE(MP_QSTR_enterprise, mp_module_enterprise);
```

micropython.cmake:
```
add_library(usermod_enterprise INTERFACE)

target_sources(usermod_enterprise INTERFACE
        ${CMAKE_CURRENT_LIST_DIR}/enterprise.c
)

target_include_directories(usermod_enterprise INTERFACE
    ${CMAKE_CURRENT_LIST_DIR}
)

target_link_libraries(usermod INTERFACE usermod_enterprise)
```

micropython.mk:
```
# this file autodetected by py/py.mk based on its name

MBEDTLS_MOD_DIR := $(USERMOD_DIR)

SRC_USERMOD += $(MBEDTLS_MOD_DIR)/enterprise.c
```

Place these files in an folder called enterprise
Add reference to you MicroPython firmware build using

```
include(${CMAKE_CURRENT_LIST_DIR}/enterprise/micropython.cmake)
```

closed 2023-09-18 by MATTYGILO

SL2021-Dev, 2023-09-18:
@MATTYGILO 

Thank you very much for your great instructions !!!
I am getting an error...
`CMake Error at enterprise/micropython.cmake:11 (target_link_libraries):
  Cannot specify link libraries for target "usermod" which is not built by
  this project.
`
I'm using micropythoh v 1.20 and IDF 4.4.5
Any ideas what I'm doing wrong?

MATTYGILO, 2023-09-18:
@SL2021-Dev Not fully sure, not incredibly well versed with CMAKE, but it will be due to the way you're referencing the cmake file

MATTYGILO, 2023-09-18:
You could be trying to reference it from a make file perhaps

SL2021-Dev, 2023-09-19:
Based on my interpretation of your instructions.

esp32
    dir> main  
        cmakelist.txt   wpa_supplicant to set(IDF_COMPONENTS …)

    dir> enterprise 
        enterprise.c
        micropython.mk
        micropython.cmake

    cmakelist.txt  include(${CMAKE_CURRENT_LIST_DIR}/enterprise/micropython.cmake)

I totally don't understand the **Make** system either. I'm hanging on by a thread with just building these binaries.

If micropython does not included a needed function, you literally would have to learn C, which defeats the whole purpose of using micropython. This is unfortunate because ESP IDF probably has many other capabilities that micropythoners can't use until the maintainers get around to including it.

Do you think this should be moved to the micropython group?

one0410, 2023-11-25:
Do you plan to merge this into master branch?

andrewleech, 2023-11-25:
It would be good to see the code above submitted as a pull request for proper review.

MATTYGILO, 2023-11-28:
@SL2021-Dev 
Path would be `micropython/ports/esp32/main/CMakeLists.txt` for the wpa_supplicant
[image]

Also have a look into making a custom C module:
[https://docs.micropython.org/en/latest/develop/cmodules.html](https://docs.micropython.org/en/latest/develop/cmodules.html)

one0410, 2024-03-08:
I still don't know how to make a custom image.
It seems different from above.

[image]

Could anybody help to write a tutorial or make image for us??

one0410, 2024-06-19:
Hi, I really need this wpa2-enterprise feature.
Could anybody teach me how to build a custom image??
Thank you very much.